wills106/homeassistant-solax-modbus #681: [Bug]: Unknown solax_run_mode on newest firmware SOLAX X1-Hybrid-G4

ivankarpov, 2023-12-28:
### Describe the bug

I recently updated the firmware of the inverter to the latest version. 
Reported by the intergration:
ARM - 1.38
DSP - 1.41 

After the update the running mode of the inverter while using remote power control is reported as unknown. On the screen of the inverter is Normal(R) displayed 

[image]
[image]

### Integration Version

2023.12.5

### Homeassistant core version

2023.12.3

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

H475

### Inverter firmware versions

ARM - 1.38 , DSP - 1.41 

### Connection Method

SolaX Poket WiFi 3.0

### Dongle firmware

12.01

wills106, 2024-01-02:
If you either comment out or delete Lines 3528 - 35538

https://github.com/wills106/homeassistant-solax-modbus/blob/7aeb55b54dde4bd29849fd9a30c9b518183b335b/custom_components/solax_modbus/plugin_solax.py#L3524-L3542

So you just have:
```
    SolaXModbusSensorEntityDescription(
        name = "Run Mode",
        key = "run_mode",
        register = 0x9,
        register_type = REG_INPUT,
        allowedtypes = GEN4,
        icon = "mdi:run",
    ),
```
What does it return when in `Normal(R)` mode? As the docs don't mention that mode, so I don't know what to map it to.

matomatusov, 2024-01-02:
Hi @Wills106
normal (R) is when it is running modbus remote control.

wills106, 2024-01-02:
I meant what does the Integration return with the modified "Run Mode" with the scaling removed. It will return a number.

ivankarpov, 2024-01-02:
Hi @wills106 

The "Normal (R)" Mode is 20 :), do you need a marge request or you will add it ? 

Regads

wills106, 2024-01-02:
I'll add it in thanks.

wills106, 2024-01-03:
This should be corrected in [2024.01.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.01.2)

wills106, 2024-01-07:
@ivankarpov can this issue be closed down now?

ivankarpov, 2024-01-08:
Yes, it is displayed as expected with the new version of the integration. Thank you1

closed 2024-01-08 by ivankarpov