wills106/homeassistant-solax-modbus #684: [Bug]: Error loading on latest version

dorianwoolger, 2024-01-02:
### Describe the bug

Hi,  many thanks for releaseing the serial number fix.
After updating to latest version the module will now not load.

Error log:

> Logger: homeassistant.config_entries
Source: config_entries.py:402
First occurred: 18:17:00 (3 occurrences)
Last logged: 18:23:59

> Error setting up entry solissolar for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 402, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 117, in async_setup_entry
    plugin = importlib.import_module(f".plugin_{plugin_name}", 'custom_components.solax_modbus')
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/plugin_solis.py", line 577, in <module>
    SolisModbusNumberEntityDescription(
TypeError: SolisModbusNumberEntityDescription.__init__() got an unexpected keyword argument 'state_class'

### Integration Version

2023.01.1

### Homeassistant core version

2023.12.4

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### Serial prefix

160F32

### Connection Method

TCP via waveshare

### Detailed Error Log

Logger: homeassistant.config_entries
Source: config_entries.py:402
First occurred: 18:17:00 (3 occurrences)
Last logged: 18:23:59

Error setting up entry solissolar for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 402, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 117, in async_setup_entry
    plugin = importlib.import_module(f".plugin_{plugin_name}", 'custom_components.solax_modbus')
            ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.11/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1204, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/plugin_solis.py", line 577, in <module>
    SolisModbusNumberEntityDescription(
TypeError: SolisModbusNumberEntityDescription.__init__() got an unexpected keyword argument 'state_class'

wills106, 2024-01-02:
Does this Beta load correctly? [2024.01.2b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.01.2b1)

dorianwoolger, 2024-01-03:
Yes that seems to load fine.

wills106, 2024-01-03:
I'll close this down then as I have released a normal release [2024.01.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.01.2) with the same change.

closed 2024-01-03 by wills106

dorianwoolger, 2024-01-03:
Many thanks.