wills106/homeassistant-solax-modbus #686: [FR] Add support for Growatt SPF

wills106, 2024-01-03:
Have same problem: ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : 067.01068.01 , with Growatt spf 5000es inverter , spf inverter is not listed, how to setup?

_Originally posted by @Asterix229 in https://github.com/wills106/homeassistant-solax-modbus/issues/673#issuecomment-1863296639_

Asterix229, 2024-01-03:
My inverter hibrid Growatt SPF 5000 ES 1x solar PV, main, load, battery, software and hardware 067 01 1.04 and 068 01 1.04, DTC device type code 034xx OffGrid SPF 3-5K, where to change register? Try plugin_growatt.py change HIBRID number but no work, think Modbus protocol v0.14 is my inverter protocol. Where to change? Firmware version recognize.
Error log:

2023-12-17 21:51:02.439 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 344: return self._client.read_holding_registers(address, count, **kwargs), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
2023-12-17 21:51:02.504 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 344: return self._client.read_holding_registers(address, count, **kwargs), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
2023-12-17 21:51:02.556 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : 067.01068.01

wills106, 2024-01-03:
Just released a Beta with a couple of Solar an Battery sensors. [2024.01.3b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.01.3b1)
Do they work ok?

Asterix229, 2024-01-03:
just instal and try. no work. before recognize firmware, now not. error log
2024-01-03 16:57:53.796 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 344: return self._client.read_holding_registers(address, count, **kwargs), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
2024-01-03 16:57:56.808 WARNING (MainThread) [custom_components.solax_modbus.plugin_growatt] growatt: reading firmware number from address 0xbb9 failed; other address may succeed
2024-01-03 16:57:56.809 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] growatt: cannot find firmware version, even not for other Inverter
2024-01-03 16:57:56.809 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : unknown

Asterix229, 2024-01-03:
My mistake recognize firmware. error log: 
2024-01-03 17:27:17.450 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 344: return self._client.read_holding_registers(address, count, **kwargs), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
2024-01-03 17:27:17.468 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 344: return self._client.read_holding_registers(address, count, **kwargs), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
2024-01-03 17:27:17.474 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : 067.01068.01

wills106, 2024-01-03:
After installing [2024.01.3b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.01.3b1) did you restart HA?

Asterix229, 2024-01-03:
yes, many times reinstal restarc change rs485 adapter the resul is same.
2024-01-03 18:46:11.203 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 344: return self._client.read_holding_registers(address, count, **kwargs), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
2024-01-03 18:46:11.218 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 344: return self._client.read_holding_registers(address, count, **kwargs), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
2024-01-03 18:46:11.234 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 344: return self._client.read_holding_registers(address, count, **kwargs), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
2024-01-03 18:46:11.249 WARNING (MainThread) [homeassistant.util.async_] Detected blocking call to sleep inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 344: return self._client.read_holding_registers(address, count, **kwargs), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues
2024-01-03 18:46:11.255 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : 067.01068.01

choenig, 2024-01-07:
Hi,

the beta broke my non SPF-Inverter. I guess the bug is 

```
GEN            = 0x0001 # base generation for MIC, PV, AC
GEN2           = 0x0002
GEN3           = 0x0004
GEN4           = 0x0008
SPF            = 0x0016             <-- should be 0x0010 :)
```

/br
christian

wills106, 2024-01-07:
> the beta broke my non SPF-Inverter.

What exactly broke? Do you have any errors in your log?

wills106, 2024-01-07:
@Asterix229 are you able to try [2024.01.3b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.01.3b2) does it start to return a couple of registers?

choenig, 2024-01-08:
Hi @wills106, 

Thanks for the fix :)

I did not have time to test it again, but the problem was, that it tried to use the SPF-Registers for my SPH, too.

> I am getting my Hex and Decimal mixed up.

That's why I'm always using the following syntax in C++, don't know about python, though:

```
enum MyType {
    Value0 = 0,       // 0
    Value1 = 1 << 0,  // 2^0 = 1
    Value2 = 1 << 1,  // 2^1 = 2
    Value3 = 1 << 2,  // 2^2 = 4
    ...
};
```

/christian

choenig, 2024-01-08:
Hi,

works fine now, thank you so much :)

/christian

wills106, 2024-01-11:
@Asterix229 Have you been able to test any of the later releases?
I want to check the couple of registers I have implemented return something, before I work on adding in more.

Asterix229, 2024-01-11:
2 days ago when install 2023.12.5 (show SW and FW in log file) gradual one update version  not work even SW and FW, then install 2024.01.3b2 it work and show some data for 5 min when reload data or restart reinstall nothing work. Today finally find time and reinstall from 23.12.5 to 24.1.3 and work all ok. Reinstall 24.1.5 work too. Thank for patience, thank so much.
error log:  __init__.py, line 344 - 12 times, line 350 - 9 times

wills106, 2024-01-11:
You can hide the blocking calls by [looking here](https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#detected-blocking-call) It's a know issue when using an RS485 - USB adaptor.

Now I know your Inverter is returning values I'll add in some more.

wills106, 2024-01-12:
I have added more support for your Inverter with [2024.01.6](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.01.6) I still need to do the Write registers

Asterix229, 2024-01-12:
Work well, chenged register 27, Battery Load, and register 21 and 23, scale=0.001.
Thank's

wills106, 2024-01-12:
Thanks for the feedback.

I'll try and make some changes over the weekend for you to try.

wills106, 2024-02-15:
I made those changes in [2024.02.5b5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.5b5)
Which missing features do you want me to focus on?

PrimusNZ, 2024-03-15:
I have the SPF 5000TL and get the following, so desnt yet work with my generation:

```
Logger: custom_components.solax_modbus.plugin_growatt
Source: custom_components/solax_modbus/plugin_growatt.py:4761
integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 2:25:28 PM (1 occurrences)
Last logged: 2:25:28 PM

unrecognized SPF5000TL inverter type - firmware version : 500.10002.05
```

wills106, 2024-08-03:
Closing down, if anyone wants to contribute more for the SPF Inverters please submit a PR.

closed 2024-08-03 by wills106