wills106/homeassistant-solax-modbus #688: [Bug]: cannot find serial number, even not for MIC

HeinzDanny, 2024-01-04:
### Describe the bug

Hallo,
seit dem Update auf Version 2024.01.2 habe ich ein Problem mit meinem SolarX-Wechselrichter. Die Entitäten verschwinden und lassen sich nur durch einen Neustart des Wechselrichters temporär wiederherstellen. Nach ein bis zwei Tagen sind wieder nur 2 Entitäten statt der sonst 270 Entitäten sichtbar.
Im Log habe ich folgende Einträge gefunden:

`2024-01-04 19:28:53.491 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] Solar: reading serial number from address 0x0 failed; other address may succeed
2024-01-04 19:28:53.591 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] Solar: reading serial number from address 0x300 failed; other address may succeed
2024-01-04 19:28:53.591 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] Solar: cannot find serial number, even not for MIC
2024-01-04 19:28:53.591 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown
2024-01-04 19:31:23.692 INFO (MainThread) [custom_components.solax_modbus] no local data file found after 5 tries - is this a first time run? or didnt you modify any DATA_LOCAL entity?`

Die Konfiguration ist richtig, ich habe sie sowohl überprüft, als auch komplett neu eingefügt. Verbindung besteht über TCP / Ethernet und über die eigegebene IP-Adresse bekomme ich direkten Kontakt zum Gerät.

### Integration Version

2024.01.02

### Homeassistant core version

2024.1.0

### Inverter brand

SolaX Power

### Plugin used

plugin_solax.py

### Serial prefix

H34C06

### Inverter firmware versions

3.001.02

### Connection Method

SolaX PocketWiFi 3.0

### Dongle firmware

./.

### Detailed Error Log

```
2024-01-04 19:28:53.491 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] Solar: reading serial number from address 0x0 failed; other address may succeed
2024-01-04 19:28:53.591 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] Solar: reading serial number from address 0x300 failed; other address may succeed
2024-01-04 19:28:53.591 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] Solar: cannot find serial number, even not for MIC
2024-01-04 19:28:53.591 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown
2024-01-04 19:31:23.692 INFO (MainThread) [custom_components.solax_modbus] no local data file found after 5 tries - is this a first time run? or didnt you modify any DATA_LOCAL entity?
2024-01-04 19:31:23.692 DEBUG (MainThread) [custom_components.solax_modbus] assuming sleep mode - slowing down by factor 10
```

wills106, 2024-01-07:
> ### Inverter firmware versions
> 3.001.02

That's your Pocket WiFi 3.0 serial.
Which is over 3 years old now.

SolaX docs mention to use a newer version than that [see here](https://github.com/wills106/homeassistant-solax-modbus?tab=readme-ov-file#pocket-wifi-30-with-firmware-v300403-and-above-is-only-officially-supported-solax-only-mentions-gen4-hybrid-other-inverters-may-work)
Even that version is around two years old and they are now on at least [V3.010.16](https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq/)

I would contact SolaX and ask for them to update the Firmware for you as it provides a more reliable connection.

wills106, 2024-01-10:
Closing this down as it's a connectivity issue and not an Integration issue.

closed 2024-01-10 by wills106