wills106/homeassistant-solax-modbus #689: Repeated entries for "number:Solax_grid_export_limit" changes in Logbook

pickeringar, 2024-01-07:
### Describe the bug

Multiple entries in logbook (which need not be there?) with potential repeated & unnecessary writes to Inverter

[image]

### Integration Version

2024.01.2

### Homeassistant core version

2024.1.2

### Inverter brand

Solax X1-Hybrid-G3

### Plugin used

plugin_solax.py

### Serial prefix

H1E5

### Inverter firmware versions

not known

### Connection Method

Modbus over Ethernet into Ethernet port on inverter

### Dongle firmware

WiFi 2 

### Additional context

Still have Solax API running whilst I change over to modbus

wills106, 2024-01-07:
It's not the Integration writing to the Inverter, but the Inverter letting you know what the Inverter is doing.
It's basically a duplicate of sensor.solax_inverter_power

If you don't want to use [button.solax_grid_export](https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-G23-operation-modes/#grid-export-gen-3-only) you could just disable it.

pickeringar, 2024-01-08:
Having tried to follow this, I now relaise how much there is in this INtegration, for which I thank you

wills106, 2024-01-08:
If you want to export the details are [here](https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-G23-operation-modes/#grid-export-gen-3-only)

I have a basic [automation example](https://github.com/wills106/homeassistant-solax-modbus/discussions/110#discussioncomment-7738020)

pickeringar, 2024-01-08:
Will work my way through those documents again having read them last night. 

In the interim I have added "number.solax_grid_export_limit" to the exclusions for "recorder" in the "configuration.yaml" file to stop the entreis in the Logbook

closed 2024-01-08 by pickeringar