wills106/homeassistant-solax-modbus #690: [Bug]: cant Change values on q cells inverter

HubsiKU, 2024-01-07:
### Describe the bug

Hello,
I worked with the solax modbus integration well. But actual i cant cange soc or so on. I cant press unlock inverter and the dropdown state ist unlocked or unlocked - advanced. What can i do?

### Integration Version

Newest

### Homeassistant core version

2024.1.2

### Inverter brand

Qcells

### Plugin used

plugin_solax.py

### Serial prefix

Srkclr

### Inverter firmware versions

?

### Connection Method

Wifi dongle

### Dongle firmware

?

wills106, 2024-01-08:
The button's were removed a couple of release ago as they are duplicated in the drop down now.

Is the lock state showing as Locked? If so change the drop down to Unlocked.

HubsiKU, 2024-01-08:
Hey, the dropdown is also on unlocked - advanced

wills106, 2024-01-08:
So when you try and move the slider on the Battery SOC is just reverts back?

[image]

HubsiKU, 2024-01-08:
Exaxtly

image.png (view on web)<https://github.com/wills106/homeassistant-solax-modbus/assets/18155231/8b7718da-42a4-4a98-942f-6b22e20cd4f0>

wills106, 2024-01-10:
Were you trying to change the entity when the Inverter was [idle](https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#unable-to-control-inverter-when-pv-0-battery-soc-is-at-minimum)?

wills106, 2024-01-23:
Closing down as no further response.
This might be the issue [FAQ](https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#unable-to-control-inverter-when-pv-0-battery-soc-is-at-minimum)

closed 2024-01-23 by wills106