wills106/homeassistant-solax-modbus #691: [Bug]: Solax Retrofit

hwvvse, 2024-01-09:
### Describe the bug

Dosent show any entities after upgrade 

### Integration Version

 2024.01.3

### Homeassistant core version

2024.1.2

### Inverter brand

Solax Retrofit

### Plugin used

plugin_solax.py

### Serial prefix

F3E103F

### Inverter firmware versions

re

### Connection Method

Lan

### Dongle firmware

lan

### Detailed Error Log

SolaX: cannot find serial number, even not for MIC
2:56:39 PM – (ERROR) SolaX Inverter Modbus (custom integration)
File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 176, in execute
raise ConnectionException(f"Failed to connect[{self!s}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.25.58:502)]
2024-01-09 14:56:39.230 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2024-01-09 14:56:39.237 ERROR (MainThread) [pymodbus.logging] Connection to (192.168.25.58, 502) failed: [Errno 104] Connection reset by peer
2024-01-09 14:56:39.237 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: attempt to read serialnumber failed at 0x300 data: None
Traceback (most recent call last):
File "/config/custom_components/solax_modbus/plugin_solax.py", line 60, in _read_serialnr
inverter_data = hub.read_holding_registers(unit=hub._modbus_addr, address=address, count=7)

^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/config/custom_components/solax_modbus/__init__.py", line 344, in read_holding_registers
return self._client.read_holding_registers(address, count, **kwargs)
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/pymodbus/client/mixin.py", line 102, in read_holding_registers
return self.execute(
^^^^^^^^^^^^^
File "/usr/local/lib/python3.11/site-packages/pymodbus/client/base.py", line 176, in execute
raise ConnectionException(f"Failed to connect[{self!s}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusTcpClient(192.168.25.58:502)]
2024-01-09 14:56:39.249 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2024-01-09 14:56:39.249 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find serial number, even not for MIC
2024-01-09 14:56:39.249 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown
2024-01-09 15:55:01.724 ERROR (MainThread) [frontend.js.latest.202401040] Uncaught error from Chrome 120.0.0.0 on Windows 10

IP is correct.....

wills106, 2024-01-09:
Are you using the PocketLan?
If so it doesn't provide a [Modbus connection](https://github.com/wills106/homeassistant-solax-modbus/blob/main/README.md#pocket-lan-does-not-provide-a-modbus-connection-at-all-and-the-pocket-wifi-v1--v2-does-not-provide-a-reliable-modbus-connection-trouble-shooting-for-pocket-wifi-v1-or-v2-will-not-be-provided)

wills106, 2024-01-09:
I wasn't aware the Retro Fit's had built in ethernet?

I thought they either connected over RS485 or PocketWiFi

hwvvse, 2024-01-09:
the realy old ones has Lan build in like X3 G2 , more or less the same i think.

hwvvse, 2024-01-10:
found the error... :) it was a network error, both Lan and dongel took the same ip... :)