wills106/homeassistant-solax-modbus #692: [Bug]: Modbus timeout

victorwm7, 2024-01-10:
### Describe the bug

The inverter goes offline during the night and here in Finland the inverter can be offline for few weeks with snow on the panels. It seems as this integration goes to sleep/ timeout when tge inverter has not responded to requests in 24 hours and is noy capable to to turn itself on again whithout restarting HA. Can the timeout be changed to atleast 48 hours?

### Integration Version

2024.01.2

### Homeassistant core version

2024.1.2

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

123456

### Connection Method

Wifi-modbus

flopp999, 2025-04-02:
Still problem?

wills106, 2025-08-28:
Closing down as no response.

closed 2025-08-28 by wills106