wills106/homeassistant-solax-modbus #693: [Bug]: After update to 2024.01.4 all Solis entities show as unavailable

J14111, 2024-01-10:
### Describe the bug

Just a heads up, I don't have much time to troubleshoot but going from 2024.01.3 to 2024.01.4 all my entities for the Solis in HA show as unavailable going back to 2024.01.3 gets it working again, I have tried it twice so there is something breaking in that update

### Integration Version

2024.01.4

### Homeassistant core version

2023.12.4

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### Serial prefix

?

### Connection Method

TCP/IP Adapter

wills106, 2024-01-10:
Does it give any error in the Log or just `unrecognized XYZ inverter type - serial number : unknown`

wills106, 2024-01-10:
Just released [2024.01.5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.01.5) It was trying to define a Bitmask that doesn't exist.

J14111, 2024-01-10:
Great thanks seems to be all ok now.

closed 2024-01-10 by J14111

J14111, 2024-01-10:
Love the new calculated string wattage BTW thanks so much :)