wills106/homeassistant-solax-modbus #696: [Bug]: Battery capacity, remaining capacity do not shown

shaarkys, 2024-01-11:
### Describe the bug

Hi @wills106 , I wonder - I saw some users does see remaining capacity in kWh for batteries but for some reason it's 0 for every possible entity I tried. However new cloud version does show it - maybe there some way to add it into HA for my specific inverter as well ? I thought that ARM/DSP would help but unfortunately not.

[image]

[image]

[image]

From Solax Cloud
[image]

### Integration Version

2024.01.3

### Homeassistant core version

2024.1.2

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

ARM 1.35 and DSP 1.36

### Connection Method

Solax Pocket Wifi 3.0

### Dongle firmware

10.16

shaarkys, 2024-01-11:
Debug as required : 
[debug_solax.txt](https://github.com/wills106/homeassistant-solax-modbus/files/13904894/debug_solax.txt)

wills106, 2024-01-11:
There is nothing obvious in the logs. It's adding in the Registers.

Like I mention in the release notes BMS registers don't always return values. This could be based on Inverter Firmware Version, Battery BMS especially third party BMS units.

The cloud portal and Modbus communicate by two different methods, so what is reported on one might not be present on the other.

Also on the X3 Gen4 there are what appears to be 4 different Hardware Revisions and I don't know what the difference is.
You have:
H34A
H34B
H34C
H34T

In the past we have had to blacklist Inverters from reading certain Registers as they fault when trying to be read

https://github.com/wills106/homeassistant-solax-modbus/blob/4fd201846054a3b21117b7a9d8cb0c9541c27a3b/custom_components/solax_modbus/plugin_solax.py#L2655

The 'XRE' is one of the 5 variations of the Gen3 Hybrid X1

shaarkys, 2024-01-11:
Thank you for your reply. That's a pity, I have Solax X3 G4 btw, I just know that Solax added that very recently so I was hoping it's just some new register ;-( I have also different firmware versions and none of them activated this value in the HA itself, including the dongle firmware. The H34A I do have is from aprox. 9/2022... 

I believe I can't investigate or help to investigate this anyhow ? Eg. maybe I can contact Solax and ask them which register is this coming from ?

Looking in the https://downloads.vodnici.net/uploads/wpforo/attachments/906/6650-SolaxHybrid-X1X3-G4-ModbusTCPRTU-V3-24-English230317.pdf - I see that BMS value in kWh is based on register 0x26 but what means 0x0027 ?

[image]

I guess you tried that already some time ago, right - https://github.com/wills106/homeassistant-solax-modbus/issues/245#issuecomment-1353150912 ?

Stepsilein, 2024-01-24:
Hi,
newest Modbus Register pdf I got from SolaX is V3.29, which I enclose here.
May be it will help will106 too.
[(Solax)Hybrid X1&X3-G4 ModbusTCP&RTU V3.29-English_231031 .pdf](https://github.com/wills106/homeassistant-solax-modbus/files/14044521/Solax.Hybrid.X1.X3-G4.ModbusTCP.RTU.V3.29-English_231031.pdf)

Stepsilein, 2024-01-24:
Just a hint and question too:
The Entinity:
number.solax_battery_charge_upper_soc
does't get saved if changed.

Using the App, its changed after hitting the "save" button.
In HA, all Entities (number settings) are usually saved imediately after been changed.
But not so by number.solax_battery_charge_upper_soc
After a few seconds, its always jumping back to 100% (or what's being saved bevore at using the App.

Is there any reason for this behavior?
I've seen 2 other Registers using too the 0x00E0
May this cause the
number.solax_battery_charge_upper_soc
not been saved by using the Integration in HA?

Its not important for me, but irritates me a little by its unusual failing.
Any info welcome.

Stepsilein, 2024-01-27:
Deeep sleep here...👎
Seems even not mentioned to say thanks for offering up to date modbus.pdf 🙄.
Auf deutsch:
Wird ja immer schlimmer mit der Github Blase, keine Antwort wie leider oft, nicht mal danke fürs pdf.. oh Mann.
Suboptimal, vorsichtig gesagt.

shaarkys, 2024-01-27:
Well @Stepsilein , thanks for the latest TCP&RTU Communication protocol, however based on the changelog, it's not really bringing anything new for this topic...

For number.solax_battery_charge_upper_soc the best would be to open separate issue.

Stepsilein, 2024-01-27:
Okay, thanks 👍

shaarkys, 2024-02-18:
@wills106 , I got finally respond from Solax : 

>
This is the installed capacity information of the battery. The data is 32 bits, 0x0026 is low 16 bits, 0x0027 is high 16 bits.
If you want to use the general remaining capacity of the battery, take the following data.
[image]

> If you want to use the general remaining capacity of the battery, take the following data.
[image]

Does it provide any help ?

wills106, 2024-02-18:
Those are the registers I am using.
https://github.com/wills106/homeassistant-solax-modbus/blob/3e6f08c9763c206777cd8131d1187b3ef56ec7c9/custom_components/solax_modbus/plugin_solax.py#L3907-L3917

SolaX uses Big endian for the 16bit registers and Little for the 32bit. Where some of the other Inverter brands use Big Endian encoding for both. 
https://github.com/wills106/homeassistant-solax-modbus/blob/3e6f08c9763c206777cd8131d1187b3ef56ec7c9/custom_components/solax_modbus/plugin_solax.py#L6658-L6659
Like I have previously mentioned BMS registers don't always work on all Inverters and that's why I generally leave them out.
If the Inverter doesn't return anything there isn't anything I can do about it unfortunately.

shaarkys, 2024-02-18:
Thank you @wills106  - in my case, in the cloud version of Solax app, I see battery capacity... and that was what I asked  them - why I see it in the cloud and which registers they use.  So on my case, Solax X4 G4 is able to return battery capacity and remaining battery capacity... it would be deal breaker, to have it in your integration ;-) 

Any final words ? ;-)

wills106, 2024-02-18:
The cloud uses a different communication method (MQTT) compared to this Integration.
Things tend to get added into one before the other as mentioned https://github.com/wills106/homeassistant-solax-modbus/issues/696#issuecomment-1887782277 or they are completely missing on some Inverters.
Only SolaX can answer why some Inverters return zero or are missing certain registers completely.

shaarkys, 2024-02-18:
Hmm, that's what I asked them as well and this is what they replied above. Was hoping it's just about the way how to read those registry correctly but I assume you are doing it right.

Stepsilein, 2024-02-18:
Its all available by the integration already, please see my screen attached.
[image]

shaarkys, 2024-02-18:
Yeah but we don't know, why in case of my Solax X3 Hybrid G4 (H34A1...)  (and some others) those values are empty ;-(
While, despite of different methods, as Wills mentioned, those data are available in the native integration.

shaarkys, 2024-02-18:
Unfortunately I don't understand fully the meaning of : 

```
This is the installed capacity information of the battery. The data is 32 bits, 0x0026 is low 16 bits, 0x0027 is high 16 bits.
If you want to use the general remaining capacity of the battery, take the following data.
```

eg. I have 3 batteries in total capacity 17.4kWh - do those registries needs to be combined somehow to get full capacity ? Eg. 16bit vs 32bits...  while integration reads it as 32bit ?      unit = REGISTER_U32,

On the other side, that would not explain 0Wh for 0x1C

shaarkys, 2024-02-18:
ChatGPT suggestions (**don't kill me**)

`BMS Battery Capacity:

    The vendor mentions that the installed capacity information of the battery is a 32-bit data, where 0x0026 is the low 16 bits and 0x0027 is the high 16 bits.
    In the integration code, only register 0x26 is read for the BMS Battery Capacity. This means it's only reading the low 16 bits of a 32-bit data. To correctly read the full 32-bit value, the integration should also read from register 0x27 and combine the two 16-bit values appropriately.
    The unit = REGISTER_U32 suggests that it expects a 32-bit value, but without reading from both 0x26 and 0x27, it's not getting the complete data.`

```
Combining two 16-bit registers into a single 32-bit value in a Modbus system involves reading both registers and then correctly combining them, taking into account the endianness. In your case, the endianness is specified as Endian.LITTLE, which affects how the registers should be combined.

Here's a general approach to combine the 16-bit values from registers 0x26 (low 16 bits) and 0x27 (high 16 bits) into a single 32-bit value:

    Read Both Registers:
        Read the value from register 0x26. Let's call this low_value.
        Read the value from register 0x27. Let's call this high_value.

    Combine the Values:
        Since the system uses little-endian format, the lower address (0x26) contains the least significant part of the data.
        Combine the two 16-bit values into one 32-bit value. In little-endian format, the low_value should be shifted left by 16 bits and then added to the high_value.
```

Stepsilein, 2024-02-18:
Asking KI... okay... but, I would ask, how I got those registers uskng HA and wills106's integration.
Nothing from cloud server, just pure modbus connection by wills106 geniouse integration.

I supose, Many are still at very old firmware at their SolaX System (Inverter, Batteries and ! Wlan Dongle) causing missing stuff and poor working system at all.

If anybody is interested, I will have a look into my used entities.

shaarkys, 2024-02-18:
> I supose, Many are still at very old firmware at their SolaX System (Inverter, Batteries and ! Wlan Dongle) causing missing stuff and poor working system at all.

I have the newest, for inverter, batteries and dongle.

shaarkys, 2024-02-18:
@wills106 if you confirm above combination of two 16bit registries is a nonsense, I will close this "notaBug" issue.
Thank you.

Stepsilein, 2024-02-18:
That's the entities I use:
sensor.solax_chargeable_battery_capacity
and
sensor.solax_remaining_battery_capacity

So... 🤷🏼‍♂️

shaarkys, 2024-02-18:
Thank you. Those are slightly different registers, register = 0x118, and register = 0x116, in my case : 
[image]
[image]

🤷🏼‍♂️

wills106, 2024-02-18:
It might depend on what batteries are you using?
T3.0, T5.8, BMS PARALLEL BOX-II, BYD Batteries could all have a different effect on what the BMS provides to the Inverter and what the Inverter can give to the Integration.

There isn't anything else I can do and the Integration is working correctly as far as I can tell otherwise Stepsilen wouldn't be getting any values.

For whatever unknown reason your Inverter can't provide the information.

shaarkys, 2024-02-18:
Thank you

> It might depend on what batteries are you using

REP-T58 - so I guess it's T5.8 - which is standard

> Integration is working correctly

Yes, it is, the rest of values are fine, it's just about those two entities, which I found interesting to have complete view. 

 I was just wondering about the comments above and the reply from Solax about 16bit and combining those values. So even you are reading one register and not combining it, it's OKay, correct ? 

Anyway, closing this as promised, will be experimenting probably on my own then.

closed 2024-02-18 by shaarkys

wills106, 2024-02-18:
`unit = REGISTER_U32,` is what reads the two 16bit registers and combines them.
`order32 = Endian.LITTLE,` is what determines the endianness to decode them in.
I wouldn't recommend trying to change `order32 = Endian.LITTLE,` to `order32 = Endian.BIG,` it will really mess up you long term stats as any other sensor using either `unit = REGISTER_U32,` or `unit = REGISTER_S32,` will get messed up and return values like 
[image]

matomatusov, 2024-02-22:
Hi @shaarkys,
I have a similar problem. Also, the given values are not shown to me. Thanks for the information.

shaarkys, 2024-02-22:
Hi @matomatusov , can you share on which inverter, first few digits and type of batteries ?

I contacted Solax again with statement that we (Wills) read the registry in proper way, format etc. but it do not work for my specific inverter, so if they can explain. I doubt will get answer but who knows....maybe

matomatusov, 2024-02-22:
Ahoj @shaarkys, inverter serial number H34A10 (first 6 numbers). Battery Trimple power 4xT58.

Stepsilein, 2024-02-22:
T58 may be the problem 🤷🏼‍♂️.
I got T30 (as far as I know).

But I don't understand still not the problem completely 🤔.
Is it "just" wrong , (or .) at the value... os is that value wrong or missing at all?

What about those values showing at the App or Portal?

Hard tl understand, as always all entities showing correct values ever at my setup.

shaarkys, 2024-02-22:
it's simply Zero for us

Stepsilein, 2024-02-22:
Okay, I understand.
Some additional points:
This 2 entities are "just" estimates values from the BMS (which is always just estimating about battery capacity).
DolaX stated f. i. a 6 kWh battery system an total capacity of 6,144 kWh.
The 2 entities (remaining and chargable) are in sum that rated total capacity and just divided in chargable and remaining part.
So you shound have the "total capacity" entity too, isn't it.
If its zero too, your T58 system will not give those values at all by may be used entities, or their settings at solax.by or ?

Do you see any capacity and charge curves at the SolaX/Qcells App?

shaarkys, 2024-02-22:
> values from the BMS
Yes, I guess even it's estimated and by BMS, it's best we can get, to really monitor battery, right ?

> So you should have the "total capacity" entity too,

Maybe, somewhere, I don't see where....

> Do you see any capacity and charge curves at the SolaX/Qcells App?

This is what I see : 
SOC is percentage, if it's not artificial, system needs to know I have 3 batteries of certain total capacity (eg. in my case 17.4kWh) but I don't see the total capacity nor where electricity value take from.

[image]

matomatusov, 2024-02-22:
I managed to find out through the api

[image]

Stepsilein, 2024-02-22:
> > values from the BMS
> > Yes, I guess even it's estimated and by BMS, it's best we can get, to really monitor battery, right ?
> 
> > So you should have the "total capacity" entity too,
> 
> Maybe, somewhere, I don't see where....
> 
> > Do you see any capacity and charge curves at the SolaX/Qcells App?
> 
> This is what I see : SOC is percentage, if it's not artificial, system needs to know I have 3 batteries of certain total capacity (eg. in my case 17.4kWh) but I don't see the total capacity nor where electricity value take from.
> 
> ![obrazek](https://private-user-images.githubusercontent.com/14859895/306988605-940c21da-4dc9-475f-9746-fa23b963ee77.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg2MDYwNzYsIm5iZiI6MTcwODYwNTc3NiwicGF0aCI6Ii8xNDg1OTg5NS8zMDY5ODg2MDUtOTQwYzIxZGEtNGRjOS00NzVmLTk3NDYtZmEyM2I5NjNlZTc3LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAyMjIlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMjIyVDEyNDI1NlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTBmZGY2MWMwYjRlNzhmNWY1YzE1MDMyYWI1N2Y1OTliYWIxNGM2ODI2YTk3ZDY4OTQxMWFlNGIwY2U0ZTY0NjkmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.-DwkRTnN7sC8t5LhGh9iFQenEmXuLRkIts5kyePbVys)
> 
> ...no graphs/curves or haven't found them (yet).
> 
> What do you mean by QCells?

Qcells = SolaX rebranded (Google would have..) 🤣.

So about your Entities... how much do you have shown at the wills106 untegration?
Should be about min. 260 even if all additional addon hooks are not set.

So, if I and now matumatosov too got those values, hmm... you should get it too.

Stepsilein, 2024-02-22:
> SolaX cloud [...]

Wills106 integration is showung all those cloud values and even some important more, which SolaX / Qcells is hiding since some time sadly.

matomatusov, 2024-02-22:
Wills106 integration
 
[image]

[image]

Stepsilein, 2024-02-22:
[image]