wills106/homeassistant-solax-modbus #699: [Bug]: Run mode : Normal (R) - will not appear

matomatusov, 2024-01-13:
### Describe the bug

Hi,
I don't know why I don't see run mode - normal (R)
Modbus Power Control (remotecontrol power) - activated

[image]

[image]

Thanks

### Integration Version

2024.01.6

### Homeassistant core version

2024.1.3

### Inverter brand

SolaX power

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

ARM: 1.35 DSP: 1.36

### Connection Method

Waveshare adapter

### Dongle firmware

3.010

wills106, 2024-01-13:
If you remove the Brackets () so you just have "Normal R"

https://github.com/wills106/homeassistant-solax-modbus/blob/bf9df3ecef2a60db1e64a69f43f4aa56086836be/custom_components/solax_modbus/plugin_solax.py#L3620

Does it display?

matomatusov, 2024-01-13:
I removed the () and it's the same.
I have Normal mode.

[image]

wills106, 2024-01-13:
You are possibly on an older Firmware version then. Where the Modbus side doesn't distinguish between Normal Mode and Normal Mode (R)

Normal Mode (R) isn't in any of the know docs yet.

wills106, 2024-01-13:
I'm going to close this down as it isn't an issue with the Integration.

The Integration can only report what the Inverter is telling it.

closed 2024-01-13 by wills106

matomatusov, 2024-01-13:
think fimware ARM, DSP?
It is current.

wills106, 2024-01-13:
Yes it looks like your Inverter is on the latest?
Perhaps "Normal Mode (R)" got added by mistake and then removed again?

It's hard to know as it's not documented and there are no proper release notes.

matomatusov, 2024-01-13:
I understand that the check log is not much.
Thank you for the effort and the info.