wills106/homeassistant-solax-modbus #700: [Bug]: Detected blocking call to sleep inside the event loop by custom integration 'solax_modbus'

ruimmsimoes, 2024-01-16:
### Describe the bug

I'm getting many warnings in the home assistant log.

In 15 minutes, more than 2100 log entries.

I already tried the workaround and restarted the Home Assistant:
```
logger:
  default: warning
  logs:
    homeassistant.util.async_: error
```

But even after restart the log entries remain.

### Integration Version

2024.01.7

### Homeassistant core version

2024.1.3

### Inverter brand

SolaX

### Plugin used

plugin_solax.py

### Serial prefix

H34A15

### Inverter firmware versions

ARM 1.35 and DSP 1.33

### Connection Method

Generic USB RS-485 adapter

### Dongle firmware

Serial connection

### Detailed Error Log

Detected blocking call to sleep inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 344: return self._client.read_holding_registers(address, count, **kwargs), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues

Detected blocking call to sleep inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 350: return self._client.read_input_registers(address, count, **kwargs), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues

wills106, 2024-01-16:
Have you tried to filter on "Detected blocking call"
Following: https://www.home-assistant.io/integrations/logger/#log-filters

The method might have changed since a user fed back on the current documented method.

ruimmsimoes, 2024-01-16:
Just tried the following config:
```
logger:
  filters:
    custom_component.solax_modbus:
      - "Detected blocking call"
```

Restarted and the messages are still appearing in the log.

Should I configure it differently?

wills106, 2024-01-16:
Perhaps try:

```
logger:
  filters:
    custom_component.solax_modbus:
      - "*blocking call*"
```

That should allow it to ignore anything before or after blocking call so it's just looking for that phrase.

wills106, 2024-01-16:
Was it one of the above examples? If so which?
I can then update the docs. Thanks