wills106/homeassistant-solax-modbus #701: [Bug]: After updating to version: 2024.01.7

JAFC-BandO, 2024-01-16:
### Describe the bug

After updating SolaX inverter Modbus integration, everything stopped working. Sensors just shows unavailable. Inveter is online and I can read data in the SolaX app and Cloud. Please help...

### Integration Version

2024.01.7

### Homeassistant core version

2024.1.3

### Inverter brand

SolaX Power

### Plugin used

plugin_solax.py

### Serial prefix

H34B06

### Inverter firmware versions

NA

### Connection Method

Solax WIFI pocket 3.0

### Dongle firmware

N/A

### Detailed Error Log

After updating SolaX inverter Modbus integration, everything stopped working. Sensors just shows unavailable. Inveter is online and I can read data in the SolaX app and Cloud. Please help...

JAFC-BandO, 2024-01-16:
I tried to roll back to working version, but didn't fix anything...

wills106, 2024-01-16:
Have you tried the suggestion's [here](https://github.com/wills106/homeassistant-solax-modbus?tab=readme-ov-file#pocket-wifi-30-with-firmware-v300403-and-above-is-only-officially-supported-solax-only-mentions-gen4-hybrid-other-inverters-may-work)?

JAFC-BandO, 2024-01-16:
What suggestion's? FAQ ?

wills106, 2024-01-16:
The link isn't working correctly, I'll move it over to the new docs.

> Pocket WiFi 3.0 with Firmware V3.004.03 and above is only [officially supported](https://kb.solaxpower.com/solution/detail/ff8080818407e2a701840a22dec20032). SolaX only mentions Gen4 Hybrid, other inverters may work?
> Contact SolaX for latest version.
> (⚠I still don't recomend the PocketWiFi. If you loose all entites after normal operation, try power cycling your Inverter.
> 
> Another approach to fixing previously working PocketWiFi installs is to restart your rooter and then reload the integration in Home Assistant. If that doesn't work you can unplug PocketWifi usb for 30 seconds and plug it in again, then reload the integration.
> 
> Updating / downgrading the integration or Home Assitant won't help, you have lost the internal Modbus connection between the Inverter and PocketWiFi and I am unable to assist with this issue.⚠)

JAFC-BandO, 2024-01-16:
No I didn't try this : Another approach to fixing previously working PocketWiFi installs is to restart your rooter and then reload the integration in Home Assistant. If that doesn't work you can unplug PocketWifi usb for 30 seconds and plug it in again, then reload the integration.

Just werid It worked yesterday, and then I updated and its just broken now, showing unavailable.. :(

wills106, 2024-01-16:
They are unreliable unfortunately especially if it's still on an early firmware version.

JAFC-BandO, 2024-01-16:
What would be the best way to set it up if not PocketWIFI  ?

wills106, 2024-01-16:
[Waveshare](https://homeassistant-solax-modbus.readthedocs.io/en/latest/compatible-adaptors/#waveshare-rs485-to-rj45-ethernet-din-rail-mount-rs485-to-eth-b-rs485-to-poe-eth-b) is the most common method.

Check your [Firmware Version](https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq/) of the Pocket WiFi though. Might be worth asking SolaX to update it first.

JAFC-BandO, 2024-01-16:
Thanks! I now see i have PocketWIFI v. 3.001.02 

I just don't understand this worked this worked the past long time untill yesterday ?

JAFC-BandO, 2024-01-16:
After updating my PocketWiFI to V3.010.16 and rebooting my router it works again. 

You can find the files to update the PocketWIFI 3.0 here : https://app.box.com/s/3nvo7ic523fhojf8uuto105q9a8dgk9n/folder/145866811000

Thanks for the quick replies wills106 :)

nitronix17, 2024-01-16:
had the same thing.
rebooted my machine and waited for like a half an hour.
suddenly it was working again (with the latest version)

UKMedia, 2024-01-22:
I've got the same issue.  I have two G4 hybrid inverters, each with 11.5kWh's of battery storage in a Master - Slave configuration.  Until very recently all was working great but then I got the sensor unavailable message.

I tries the suggestions above Including router reboot and power cycling the inverter and battery but no improvement.  So I removed all of the integrations for SolaX and restarted.  I can add the Master Inverter but when the Slave Inverter is added, no device is created.  I've tried adding the Slave Inverter first but it still won't create a device.

wills106, 2024-01-23:
Closing down following Firmware update on PocketWiFi

closed 2024-01-23 by wills106