wills106/homeassistant-solax-modbus #703: [Bug]: Battery Minimum Capacity not applied

Matz88, 2024-01-17:
### Describe the bug

With an HYD6000 (sofar but rebranded) whenever I try to set the Battery Minimum Capacity or  Battery Minimum Capacity OffGrid, the setting is not applied, and the previous value reverted back in few seconds.
I see no specific errors in the log, do you know if some button need to pressed in order to save the value like for reflux?

### Integration Version

 2024.01.7

### Homeassistant core version

2024.1.2

### Inverter brand

Sofar

### Plugin used

plugin_growatt.py

### Serial prefix

ZM2ES0

### Connection Method

Lan adapter

kai-seipp, 2024-02-14:
This hier might be related:
Setting the value of "Battery Minimum SOC" also can be set remotely. Once set via UI, it will jump back to the previous setting once the inverters values are read again via modbus.

BUT:
Setting "Force Charge SOC" actually works.
These are the two values you can set within a solis inverter on battery management, so I believe there is a good chance the first one can be set as well.

Both values are on Solis Inverters.

cschlipf, 2024-02-19:
That's right. This is not working as this is one of the examples, where Sofar requires a lot of registers to be written at once. Obviously it never worked.

However when looking at it I found out that in order just to set the depth of discharge we have to set 23(!) registers containing all the battery parameters. And a lot of these battery parameters are mission critical. Setting one of these parameters to the wrong value can be harmful.

See the documentation here:
[image]

Register we want to change are 0x104D and 0x104E. I intend to work on this issue in stages: Stage 1: Read all the parameters and get a good feeling on how reliably these are read. However given the criticality of these parameters I have a very bad feeling about this and I would rather propose to remove both parameters from the integration.

What is your use case that you want to achieve by changing these values? Maybe we can find another solution?

Matz88, 2024-02-20:
I see the point.

Was to decrease the DoD in winter to not leave the battery with a low state of charge when there are several days of bad weather. I tried also to limit the discharge staying in passive mode, but this has the drawback that in case the grid has a glitch, the inverter resets and as effect I have a small gap in the house power supply, few sec but enough to power cycle some devices that should not.

cschlipf, 2024-02-21:
@kai-seipp - this defect is about Sofar. Please don't mix up other brands.

cschlipf, 2024-02-21:
I know this is not what you are hoping for, but the PR will remove these non-functional controls for now. Closing it as it is very unlikely that these controls will be added back

closed 2024-02-21 by Matz88

cschlipf, 2024-02-21:
Yes, I added that and tested it with my inverter. However I have an HYD 10 KTL-3PH.

cschlipf, 2024-03-10:
@Matz88 with PR #799 we can start reading the battery parameters that are required in order to set depth of discharge, end of discharge, and EPS buffer.

Once released please activate the entities starting with "BatConfig:" and send me a screenshot of the values. This will be very useful in evaluating whether a write operation ever can be implemented successfully. On my inverter it doesn't look good unfortunately :(

Matz88, 2024-03-17:
Sorry I was not able to take time for this before, I cannot see any batconfig in the diagnostic, have been they already removed?

cschlipf, 2024-03-17:
@Matz88 - the values are hidden by default and need to be enabled.

Matz88, 2024-03-20:
For me too the battery type is wrong:
[image]
rest seems ok, settings are ok

cschlipf, 2024-03-21:
> For me too the battery type is wrong: [...]

Thank you. Your protocol is also wrong.

PR #764 (2024-02-22): Removed non-functional Battery Control and Capacity controls
Closes #703 

Removing the following fields:
- Battery Active Control: Was not working and should never be changed. Disabling does not make sense as a battery in standby would no longer wake up
- Battery Minimum Capacity: Not working
- Battery Maximum Capacity: Not working

Later two points  have to be set with a huge set of other extremely critical registers. Will need to evaluate further, but it's very unlikely that the risk is worth adding these two capacity controls back. Therefore closing #703 for now.