wills106/homeassistant-solax-modbus #704: [Bug]: solis error AttributeError: 'builtin_function on _or_method' object has no attribute 'startswith' and modbus compatability with 4g inverters

holly1889, 2024-01-17:
### Describe the bug

Integration won’t set up as there is the following error

AttributeError: 'builtin_function_or_method' object has no attribute 'startswith'

also solis old appears to contain incorrect modbus registers, running mbpoll and reference to the linked document indicate Modbus Addresses are for some of the integrations of set minus 1 and confirm working modbus connection with 4g pv inverter 

see this document section 6 

https://diysolarforum.com/resources/solis-grid-tied-inverters-2018-rs485-modbus-communication-protocol.272/download

### Integration Version

2024.1.7

### Homeassistant core version

2024.1.3

### Inverter brand

Solis

### Plugin used

plugin_growatt.py

### Serial prefix

160A80

### Connection Method

S2-WL-ST

### Detailed Error Log

Logger: homeassistant.config_entries
Source: config_entries.py:406 
First occurred: 10:53:23 (1 occurrences) 
Last logged: 10:53:23

Error setting up entry Solis pv for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 406, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 138, in async_setup_entry
    hub = SolaXModbusHub(hass, name, host, port, tcp_type, modbus_addr, interface, serial_port, baudrate, scan_interval, plugin, config)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 233, in __init__
    self._invertertype = self.plugin.determineInverterType(self, config)
                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_solis_old.py", line 437, in determineInverterType
    if   seriesnumber.startswith('303105'):  invertertype = HYBRID | X1 # Hybrid Gen5 3kW
         ^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'builtin_function_or_method' object has no attribute 'startswith'

wills106, 2024-01-22:
Is this with 2024.01.7 or 2024.01.8b1?
Not seen this error before. I made a change in 2024.01.8b1 to try and bypass the serial check, so I'm not sure if this is what triggered that error?

holly1889, 2024-01-24:
> Is this with 2024.01.7 or 2024.01.8b1?
> Not seen this error before. I made a change in 2024.01.8b1 to try and bypass the serial check, so I'm not sure if this is what triggered that error?

Hi this is with both the beta and 2024.01.7 there is a missing bracket in the code the decode() is missing two brackets. 

I have my 4g solis pv none hybrid inverter being polled using the solis data logging stick and the Modbus intergration but it’s not an ideal situation but I can get data. Therefore I would like to get this intergration working. The problem seems to be the serial number read it fails, fails to read the serial at all I have tried placing my serial number into your solis old script without  luck. I get inverter not found, and it does not read the serial number imply there is an issue with the connection. However this is not correct I have managed a basic connection. My serial number starts with 160A80, it’s not hybrid pv only 4g 3kw inverter could you add it to the code and I will see if it works. 

The beta still tried to check my inverter serial number I did not see any difference in behaviour.

wills106, 2024-02-09:
I have added in the missing serial decode bit in [2024.02.3b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.3b1) does it start decoding the serial number now?

wills106, 2024-05-18:
Closing down as no further response in past couple of months.

Feel free to reopen if needed.

closed 2024-05-18 by wills106