wills106/homeassistant-solax-modbus #708: [Bug]: ERROR (MainThread) [homeassistant.components.sensor] Error adding entities for domain sensor with platform solax_modbus

chronvas, 2024-01-24:
### Describe the bug

When I first installed the integration and successfully connected to the inverter, I noticed some unexpected logs in the integration logbook associated with some error logs in the HA logger.
The errors repeat every time I restart HA.

[image]

### Integration Version

2024.1

### Homeassistant core version

2024.1.5

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### Serial prefix

603101

### Connection Method

RS485-WiFi (Elfin-EW11)

### Detailed Error Log

```
2024-01-24 18:55:00.840 ERROR (MainThread) [homeassistant.components.sensor] Error adding entities for domain sensor with platform solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 752, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1280, in add_to_platform_finish
    await self.async_internal_added_to_hass()
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 303, in async_internal_added_to_hass
    raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Entity sensor.solisinverter_backflow_power cannot be added as the entity category is set to config
2024-01-24 18:55:00.844 ERROR (MainThread) [homeassistant.components.sensor] Error while setting up solax_modbus platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 368, in _async_setup_platform
    await asyncio.gather(*pending)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 507, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 752, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1280, in add_to_platform_finish
    await self.async_internal_added_to_hass()
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 303, in async_internal_added_to_hass
    raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Entity sensor.solisinverter_backflow_power cannot be added as the entity category is set to config
```

### Additional context

Inverter is Solis S5-EH1P3K-L

chronvas, 2024-01-24:
I've also noticed that HA front end micro-freezes every 15 seconds (assumingly every time an update is requested?). 
What I mean is e.g.: when I use the terminal addon and I type characters, every 15 seconds there are a few seconds that the terminal doesn't show me what I type. No characters are lost, they eventually appear with a delay of 3-4 seconds.
No associated logs, this might be a separate issue?

wills106, 2024-02-20:
There has been a couple of backflow changes recently. https://github.com/wills106/homeassistant-solax-modbus/releases?q=backflow&expanded=true

Have these changes helped?

chronvas, 2024-02-21:
I've just updated to the latest but the error looks the same, different code line

```
Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:533
Integration: Sensor (documentation, issues)
First occurred: 6:00:49 PM (1 occurrences)
Last logged: 6:00:49 PM

Error adding entities for domain sensor with platform solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 533, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 782, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1323, in add_to_platform_finish
    await self.async_internal_added_to_hass()
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 304, in async_internal_added_to_hass
    raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Entity sensor.solisinverter_backflow_power cannot be added as the entity category is set to config
```

```
Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:352
Integration: Sensor
First occurred: 6:00:49 PM (1 occurrences)
Last logged: 6:00:49 PM

Error while setting up solax_modbus platform for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 352, in _async_setup_platform
    await asyncio.gather(*pending)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 533, in async_add_entities
    await asyncio.gather(*tasks)
  File "/usr/src/homeassistant/homeassistant/helpers/entity_platform.py", line 782, in _async_add_entity
    await entity.add_to_platform_finish()
  File "/usr/src/homeassistant/homeassistant/helpers/entity.py", line 1323, in add_to_platform_finish
    await self.async_internal_added_to_hass()
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 304, in async_internal_added_to_hass
    raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: Entity sensor.solisinverter_backflow_power cannot be added as the entity category is set to config
```

jone9618, 2024-03-02:
I have been getting the same backflow error with my Solis inverter for the past few months.

chronvas, 2024-03-16:
I just updated to the latest and the error is now gone, after I followed your instructions in the release notes

```
Bump pyModbus to v3.6.6 Hopefully resolves communication problems people are experiencing. 
I would also recommend downloading the update through HACS removing the integration from the 
Integration page, restarting HA and adding it in again. 
To ensure there isn't any left over configuration prior to 2024.03.1 stored in the cache.

```
@jone9618 give this a try too.

I will close that next week if I don't notice the error coming up again, thanks in advance

wills106, 2024-05-18:
Closing down as no further response in past couple of months.

Feel free to reopen if needed.

closed 2024-05-18 by wills106