wills106/homeassistant-solax-modbus #710: [Bug]: Growatt SPH 10000TL3 BH-UP wrong data

Hellraiser8, 2024-01-26:
### Describe the bug

installed and something is working but the data seems very odd. Looks like registers are read wrongly. 
See pic attached, esp if you compare Phase 1,2,3
[image]

### Integration Version

2024.01.8

### Homeassistant core version

2023.11.3

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

TPJ4CH

### Connection Method

Waveshare Adapter

the-sim87, 2024-02-12:
I have the same issue with my Growatt SPH 6000TL3 BH-UP

wills106, 2024-02-12:
If you edit:
"AC Power to Grid L1"
"AC Power to Grid L2"
"AC Power to Grid L3"
"House Load L1"
"House Load L2"
"House Load L3"

Replacing `unit = REGISTER_S32` with `unit = REGISTER_U32`

So
https://github.com/wills106/homeassistant-solax-modbus/blob/e4c4f436780009eeb175134f7a19c6905293b68c/custom_components/solax_modbus/plugin_growatt.py#L2667-L2680

Becomes:
```
    GrowattModbusSensorEntityDescription(
        name = "AC Power to Grid L1",
        key = "ac_power_to_grid_l1",
        native_unit_of_measurement = UnitOfPower.WATT,
        device_class = SensorDeviceClass.POWER,
        state_class = SensorStateClass.MEASUREMENT,
        register = 1023,
        register_type = REG_INPUT,
        unit = REGISTER_U32,
        scale = 0.1,
        rounding = 1,
        allowedtypes = GEN | GEN2 | X3,
        icon = "mdi:home",
    ),
```

Do they return correct values?

MichaelNRW, 2024-03-07:
Hi,
the same issue with SPH 4000TL3 BH-UP

a9e78rauijasdf, 2024-03-26:
I would like to see if I can replicate your problems - but at the moment I am unable to get data into the system. Would you mind sharing your setup parameters (modbus parameters, solax integration parameters) ?

the-sim87, 2024-03-27:
Hi! My setup:
Raspberry-Pi with evcc and RS485 adapter running with these options: baudrate: 9600 and comset: "8N1"
evcc is making a modbus-proxy, I am accessing with homeassistent-solax-modbus with TCP, no other options active, everything standard.

a9e78rauijasdf, 2024-03-27:
Thank you for replying :-) Im not sure your default parameters/options are the same as mine - would it be possible to post a screenshot of all settings?
Can you confirm using pin 4/5 (blue/blue-white) and RS485-3?

the-sim87, 2024-03-27:
Hi!
RS485-3 is correct. I use Pin 1 (orange-white) to B and Pin 5 (blue-white) to A.
EVCC settings is as here: 
meters:
- type: template
  template: growatt-hybrid
  id: 1
  device: /dev/ttyUSB0
  baudrate: 9600
  comset: 8N1
  usage: grid
  modbus: rs485serial
  name: grid1
- type: template
  template: growatt-hybrid
  id: 1
  device: /dev/ttyUSB0
  baudrate: 9600
  comset: 8N1
  usage: pv
  modbus: rs485serial
  name: pv2

modbusproxy:
  - port: 5021
    device: /dev/ttyUSB0
    baudrate: 9600
    comset: "8N1"
    readonly: true

homeassistant-solax-modus settings:
[image]

sccssf, 2024-03-30:
Same problem. Same modbusproxy Evcc

[image]
[image]

a9e78rauijasdf, 2024-04-03:
Hi,
I have just got it working. I dont see any weird values, sorry. But it does seem that i cannot modify any values - that is probably not related to this weird bug though.

asiudoisjdfo, 2024-04-08:
> Hi, I have just got it working. I dont see any weird values (yet?), sorry.

Sorry, i stand corrected - I also have weird values for some entities:

[image]

There are a few other entities with errors as well.

blinderix, 2024-08-19:
Sph 8000 TL3 some values are wrong. Any update?

TheEragon, 2024-09-16:
Same problem with SPH 10000 TL3 that just got installed

rebay-viktor, 2024-10-11:
Same story with the SPH5000TL3BH-UP

wills106, 2024-10-12:
What does "Firmware Version" & "Firmware Control Version" return for people with this issue?
Does everyone have a Hybrid Inverter?

@choenig What do those values return on your Inverter as you helped to add in support for this Inverter with #473 & #697 I'm wondering if your on a slightly different Inverter to those with issues.

sccssf, 2024-10-12:
> What does "Firmware Version" & "Firmware Control Version" return for people with this issue? Does everyone have a Hybrid Inverter?
>
> @choenig What do those values return on your Inverter as you [...]

They all have the same family. They are all hybrid inverters. The question is do they deliver wrong values or does the plugin do it wrong.

My SPH8000 TL3 BHUP has the firmware.

YA1.0 / YBAA050510 / ZDAA-0010

How can I help?

wills106, 2024-10-12:
Main thing for now is to see what choenig's Inverter returns. Also to try and find any users of this Inverter who isn't getting wrong data. Then we can look at splitting the entities out if there is a difference in Firmware.

badgercho, 2024-10-12:
I have the same Firmware Version and Control Version like @rebay-viktor, I also see the wrong values on L2 and L3. 

Still I'm hoping that any future firmwareversion will fix this :)

choenig, 2024-10-12:
[Now from the correct one ;) ]

I have the same Firmware Version and Control Version like @rebay-viktor, I also see the wrong values on L2 and L3.

Still I'm hoping that any future firmwareversion will fix this :)

wills106, 2024-10-12:
Thanks @choenig so it's possibly a Firmware Issue?

Has anyone reported this to Growatt?

choenig, 2024-10-12:
I have not. I tried to contact Growatt once to ask for a firmware-update for my inverter. Never heard anything from them.

Their firmware-politics seem to be very confusing. Over at the photovoltaik-forum is a thread about bad low-light performance of the Growatt SPHs and there they discuss a ton of different firmwares. 050510 seems to be the "new" stable one, but there are various firmwares they install on your inverter.

There is a .zip of the 050510 version over there but I don't really dare to install that on my inverter ;)

TheEragon, 2024-10-24:
I tried contacting Growatt, unfortunately they shared only a part of new PDF specs. It seems that either the definition of some registers have changed or were always wrong in the old documentation. Here is a snipet I got from them

[image]
[image]

As you can see, Load is not in registers 1031-1036 like the v1.24 documentation says but it's in registers 1034-1036 and has signed 16bit type instead as well. Also this documentation has a bug in it as well and the scale is not 0.1W but 1W instead as they confirmed in follow up mail.

I suspect this also applies for AC Power to User and AC Power to Grid, etc.

choenig, 2024-10-24:
This looks really interesting, I really tried to understand the values from the registers but a layout like this (having 1033 reserved and R, S and T only single registers) was never in my mind. Let's see what this might give ...

[image]

choenig, 2024-10-24:
But for AC Power to User (which starts at 1015 afaik) this does not seem to fit:

[image]

TheEragon, 2024-10-25:
It could follow the same scheme 1015-1016 AC power to user, 1017 AC Power to user R (in your case 465), 1018 AC Power to user S (-247), 1019 AC Power to user T (-203)

choenig, 2024-10-25:
You are absolutely right.

With all positive values, this looks like:
[image]

So 
- 1015+1016 is AC Power to User (0.1 W)
- 1017 is AC Power to User R (1 W)
- 1018 is AC Power to User S (1 W)
- 1019 is AC Power to User T (1 W)
- 1020 seems reserved
- 1021+102 is AC Power to User (0.1 W) <-- again?

TheEragon, 2024-10-25:
Yes, the total value repeats for some reason.

I think this pattern also applies to Power to grid

choenig, 2024-10-25:
Yep, looks really good.

If I find the time, I'll submit a patch today or at the weekend :)

choenig, 2024-10-26:
Patch us submitted.

This now means, that my SPH is fully supported and I don't have any strange values anymore, thank you @TheEragon  and @wills106 for this :)

TheEragon, 2024-10-26:
> Patch us submitted.
> 
> This now means, that my SPH is fully supported and I don't have any strange values anymore, thank you @TheEragon  and @wills106 for this :)

Thank you for the patch! Now I don't need to do it :))

TheEragon, 2024-10-26:
By the way, @choenig, what values do you see in under Load? Mine seemed really weird, for example all three phases were always same or of by 1-5W and when you add them together they don't match with Total load.

sccssf, 2024-11-01:
works Thanks @choenig @TheEragon

wills106, 2024-11-02:
Just released [2024.11.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.11.1) with these fixes

wills106, 2024-11-03:
Going to close this issue down as I believe it's resolved.

closed 2024-11-03 by wills106

dacCR, 2025-05-03:
Does HA work with the SPH10000TL-HU-US? I cannot find this specific model and connectors for the others are not the same.