wills106/homeassistant-solax-modbus #713: [Bug]: number.solax_battery_charge_upper_soc does't get saved if changed.

Stepsilein, 2024-01-27:
### Describe the bug

Just a hint and question too:
The Entinity:
number.solax_battery_charge_upper_soc
does't get saved if changed.

Using the App, its changed after hitting the "save" button.
In HA, all Entities (number settings) are usually saved imediately after been changed.
But not so by number.solax_battery_charge_upper_soc
After a few seconds, its always jumping back to 100% (or what's being saved bevor at using the App.

Is there any reason for this behavior?
I've seen 2 other Registers using too the 0x00E0
May this cause the
number.solax_battery_charge_upper_soc
not been saved by using the Integration in HA?

Its not important for me, but irritates me a little by its unusual failing.
Any info welcome.

### Integration Version

latest always

### Homeassistant core version

latest always

### Inverter brand

SolaX Qcells

### Plugin used

plugin_solax.py

### Serial prefix

-----------

### Inverter firmware versions

Latest 36 09 35

### Connection Method

Wifi Dongle

### Dongle firmware

----

### Detailed Error Log

no error

kai-seipp, 2024-02-14:
This hier might be related: 
Setting the value of "Battery Minimum SOC" also can be set remotely. Once set via UI, it will jump back to the previous setting once the inverters values are read again via modbus. 

BUT: 
Setting "Force Charge SOC" actually works. 
These are the two values you can set within a solis inverter on battery management, so I believe there is a good chance the first one can be set as well. 

Both values are on Solis Inverters.

nosaltomas, 2024-04-10:
Hi there, 
did you find some solution or workaround how to change entity number.solax_battery_charge_upper_soc?

I have the same behaviour as [Stepsilein](https://github.com/Stepsilein).

It should be for Solax this code in Plugin_Solax.py row 924:

    SolaxModbusNumberEntityDescription(
        name = "Battery Charge Upper SOC",
        key = "battery_charge_upper_soc",
        register = 0xE0,
        fmt = "i",
        native_min_value = 10,
        native_max_value = 100,
        native_step = 1,
        native_unit_of_measurement = PERCENTAGE,
        allowedtypes = AC | HYBRID | GEN4 | GEN5,
        icon = "mdi:battery-charging-high",
    ),

There is also similar entity for readings (0x010E). So Im aware to use for writing entity with prefix number.

Thanks for any news

Xoffroad, 2024-05-01:
Same issue here. Set battery_charge_upper_soc dosn‘t work.

nadlabak, 2024-05-12:
Same here. In a few moments, the battery_charge_upper_soc reverts itself back to whatever value has been set via the solax app.

mrxps, 2024-07-28:
same here.
any news on that?
Solax X3-Hybrid-G4-15.0 (G4.2) via WIFI Dongle (v3)

johny-mnemonic, 2024-09-01:
Same issue here.

Solax X3-Hybrid-G4-10.0 via WIFI Dongle (v3)

wills106, 2024-09-21:
I'm closing this report down and keep one of the one's open on the first page of the Issues as more visible.

Use #1017 as the main discussion.

closed 2024-09-21 by wills106