wills106/homeassistant-solax-modbus #714: [Bug]:  connection with SOLAX and HA is lost

matomatusov, 2024-01-29:
### Describe the bug

Hi,
I started having the following problem. I am attaching the given logs (ERROR_1 and ERROR_2).
I replaced Packet wifi with Packet wifi + LAN which supports Modbus TCP over LAN.
I don't know if it can be related. When I had communication via the waveshare adapter, everything was OK.

Thanks

---------------------------------------------------------------------------
 Log Details( ERROR_1)

This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:73
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 7:03:03 PM (6 occurrences)
Last logged: 7:05:05 PM

SolaX: reading serial number from address 0x0 failed; other address may succeed
SolaX: reading serial number from address 0x300 failed; other address may succeed

-------------------------------------------------------------------------------------
 Log Details( ERROR_2)

This error originated from a custom integration.

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:6451
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 7:03:03 PM (3 occurrences)
Last logged: 7:05:05 PM

SolaX: cannot find serial number, even not for MIC

### Integration Version

latest always

### Homeassistant core version

latest always

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

ARM1.36 DSP1.36

### Connection Method

Packet Wifi + LAN 3.0

### Dongle firmware

01.20

Rosentorp, 2024-02-01:
Same issue. Serial prefix H34A15J...

2024-02-01 17:49:50.544 INFO (MainThread) [custom_components.solax_modbus.config_flow] detected HA core version 2024 1
2024-02-01 17:49:50.544 INFO (MainThread) [custom_components.solax_modbus.config_flow] starting configflow - domain = solax_modbus
2024-02-01 17:49:50.544 INFO (MainThread) [custom_components.solax_modbus] trying to load plugin - plugin_name: solax
2024-02-01 17:49:50.559 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: trying to determine inverter type
2024-02-01 17:49:50.658 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x0 failed; other address may succeed
2024-02-01 17:49:50.658 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] Read SolaX 0x0 serial number before potential swap: None
2024-02-01 17:49:50.758 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: reading serial number from address 0x300 failed; other address may succeed
2024-02-01 17:49:50.758 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] Read SolaX 0x300 serial number before potential swap: None
2024-02-01 17:49:50.758 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: cannot find serial number, even not for MIC
2024-02-01 17:49:50.758 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown

syca87, 2024-02-04:
had same issue in logs:

unrecognized inverter type - serial number : unknown
00:14:04 – (FEHLER) SolaX Inverter Modbus (benutzerdefinierte Integration)
SolaX: cannot find serial number, even not for MIC
00:14:04 – (FEHLER) SolaX Inverter Modbus (benutzerdefinierte Integration)
SolaX: reading serial number from address 0x0 failed; other address may succeed
00:14:04 – (WARNUNG) SolaX Inverter Modbus (benutzerdefinierte Integration) - Die Nachricht ist zum ersten Mal am 00:14:04 aufgetreten und erscheint 2 mal

I guess I fixed it by unplugging the pocket-wifi stick from the inverter and plugged it in after about 1 minute. After that I did reload the integration and the data is now coming back into HA.

syca87, 2024-02-05:
> had same issue in logs:
>
> unrecognized inverter type - serial number : unknown 00:14:04 – (FEHLER) SolaX Inverter Modbus (benutzerdefinierte Integration) SolaX: cannot find serial number, even not [...]

problem was fixed for about an hour, then it came back and no data again...

wills106, 2024-02-07:
@matomatusov There really isn't any information on the new WiFi+Lan dongle. I don't know if it provides a Modbus connection or not? Are you using it in WiFi mode of Lan mode?

@Rosentorp You don't have a working Modbus connection https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#unrecognized-xyz-inverter-type-serial-number-unknown

@syca87 You might be on an old Firmware version? https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq/#i-have-lost-my-entities-on-restart-update-using-pocketwifi-30

matomatusov, 2024-02-07:
@wills106 
It just provides modbus TCP over wifi and Lan. I needed to release the COM port for the EV charger. The older type did not allow this via LAN. I thought that the given configuration might work on this device. Sometimes modbus TCP stops working for me.
Writes the given problem. It works without problems through the WaveShare converter.

wills106, 2024-02-10:
The early PocketWiFi 3.0 needed firmware updates to provide a more stable Modbus connection. Perhaps this newer one is having similar issues? Might be worth asking SolaX if there is an update for it?

syca87, 2024-02-10:
Thank you for the support!
I have the most recent Firmware installed now on my Pocket-Wifi Stick and replugged it and also restarted the networking components - but at some random time the error comes again. I have installed the newest Integration version today. Maybe this fixes it? At the moment it works.

Another thing to mention maybe: Before using HA, I used iobroker with its Solax adapter (https://github.com/simatec/ioBroker.solax)
This worked forever without any interruption ever.

wills106, 2024-02-10:
Unfortunately the Pocket WiFi isn't reliable and that's why I don't recommend them.
I don't own one, so I can't do any testing.

That ioBroker thing it using the Cloud api by the looks. Completely different communication method to how this integration works.

syca87, 2024-02-10:
with the iobroker adapter, you have the option to use it via Cloud or local IP (of the pocket WIFI). I always used it with local and Polling of 5seconds which is described over here: https://github.com/simatec/ioBroker.solax/blob/master/docs/en/solax.md

wills106, 2024-02-11:
That's not using a Modbus connection though.
It's using a local API which this Integration doesn't support.

I'm closing this down as I can't provide any further support on PocketWiFi or the newer combined one as I don't own any of the Pocket series.

Hopefully SolaX can provide further Firmware updates to make the Modbus connection on them more reliable and not shutting down communications within the PocketWiFi.

closed 2024-02-11 by wills106

matomatusov, 2024-02-11:
Thanks for your time. I communicate with support and apparently they are having problems there. Regarding modbus connections.