wills106/homeassistant-solax-modbus #718: [Bug]:

iobrokerneu, 2024-02-02:
### Describe the bug

Hello, first of all I wanted to thank you for your work. I use your software in the Home Assistant and I have a question or it may be a bug. I use a Growatt Mod 5000TL3-XH with an ARK HV battery. Unfortunately, no data from the battery is displayed in your software. I miss the charge in %, discharge and charge in w. Am I doing something wrong or doesn't it exist yet? Unfortunately I didn't know any other way to ask. Thanks and greetings Sascha

### Integration Version

2024.01.8

### Homeassistant core version

2024.1.5

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

000000

### Connection Method

USR TCP232-410S

iobrokerneu, 2024-02-02:
Hello again. Attached is a screenshot of an entity. This is always at 0 or is not updated. Greetings Sascha
[image]

sensor.growattmodbus_battery_combined_power

wills106, 2024-02-18:
Just released [2024.02.5b8](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.5b8) does it fix missing"Battery Charge Power" & "Battery Discharge Power" and the combined one?

iobrokerneu, 2024-02-18:
Hello, there are now more entities, but unfortunately no data is displayed. Greetings Sascha
[image]

iobrokerneu, 2024-02-18:
[image]

iobrokerneu, 2024-02-18:
[image]

wills106, 2024-02-18:
I have removed those extra entities that return zero and I have move the battery sensors to a different register set in [2024.02.5b9](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.5b9) to see if they work.

iobrokerneu, 2024-02-19:
Hello, the data is now displayed. Thank you for your work!!!

Entity:
growattmodbus battery charge power when charging the battery

Entity: growattmodbus battery discharge power when discharging.

Entity: growattmodbus battery combinend power is also displayed when loading and unloading.

Greetings Sascha
[image]
[image]

wills106, 2024-02-19:
Glad it's working.
Can this be closed down now?

closed 2024-02-19 by iobrokerneu