wills106/homeassistant-solax-modbus #727: [Bug]: Wrong Values multiplier

Saentist, 2024-02-07:
### Describe the bug

Some of values have wrong interpretation
```sh
Grid Current: 0.8A >> 0.08A
Grid Power: 93W >> 9.3W
Grid Power VA: 118 VA >> 11.8 VA
```
Multiplied by 0.1 need to be 0.01

### Integration Version

2024.02.1

### Homeassistant core version

2024.01.1

### Inverter brand

Growatt MIC 600TL-X

### Plugin used

plugin_growatt.py

### Serial prefix

GH1.0ZAAa

### Connection Method

Elfin EW11

### Additional context

```js
Device Serial Number：FRJxxxxxxx
Version: GHAA1050
Communication Version Number: ZAAa-0094
Build Number: GH1.0
Device Model: MIC 600TL-X
mppt: 1
```
[image]

Saentist, 2024-02-09:
Inverter product line is
**MIC 600~3300TL-X**
https://growatt.tech/wp-content/uploads/2023/02/Datasheet-Growatt-MIC-6003300TL-X-1-1.pdf

in comment is mentioned other product line
**MIN 2500~6000TL-X**
https://growatt.tech/wp-content/uploads/2023/02/MIN_2500-6000TL-X_Datasheet.pdf

MIC and MIN are different products.
Maby from there is confusion?
Guess some modbus register can extend detection of inverter

wills106, 2024-02-09:
Your Inverter is using the same register set as the MIN range.
I'll wait and see if another MIC user is having the same scaling issue or not.

Saentist, 2024-02-09:
Hope someone other wit MIC inverter post.

until then I see this when read registers
[image]

wills106, 2024-02-09:
I'm not sure what Register set you are using. I am only aware of the Growatt SPF returning values on input register 36.

Power VA for example is using:

https://github.com/wills106/homeassistant-solax-modbus/blob/9fbab81a5628c0d4bf710152fb5bd59ed225d15f/custom_components/solax_modbus/plugin_growatt.py#L3435-L3445

Do you have any links to Modbus documentation specific to the MIC TL-X?

Saentist, 2024-02-09:
Standard RTU documentation revision 1.24 should cover it
Inverter is not RTU2 

Actually there is a API documentation with cover some specific inverter TYPE parameters 
but is more reverse engineering if converted to ModBus request
https://www.showdoc.com.cn/262556420217021/6118559963559236

Saentist, 2024-02-10:
RTU
[Growatt-PV-Inverter-Modbus-RS485-RTU-Protocol-V3-14.pdf](https://github.com/wills106/homeassistant-solax-modbus/files/14230117/Growatt-PV-Inverter-Modbus-RS485-RTU-Protocol-V3-14.pdf)

RTU 2
[Growatt-Inverter-Modbus-RTU-Protocol_II-V1_24-English.pdf](https://github.com/wills106/homeassistant-solax-modbus/files/14230118/Growatt-Inverter-Modbus-RTU-Protocol_II-V1_24-English.pdf)

Its very strange to call protocol by some document revision instead as real names
RTU and RTU2

```
TL-X/TL-XH/TL-XH US（MIN Type)：
03 register range：0~124,3000~3124， 3125~3249(TL-XHUS)； 
04 register range ： 3000~3124,3125~3249 3250~3374（TL-XH）

TL3-X(MAX、MID、MAC Type)：
03 register range：0~124,125~249； 04 register range：0~124,125~249

MAX 1500V、MAX-X LV: 
03 register range：0~124,125~249；04 register range：0~124,125~249,875~999

MOD TL3-XH:  
register range：0~124,3000~3124；04 register range： 3000~3124,3125~3249

Storage(MIX Type)：
03 register range：0~124,1000~1124；04 register range：0~124,1000~1124

Storage(SPA Type)：
03 register range：0~124,1000~1124；04 register range：1000~1124,2000~2124，1125~1249

Storage(SPH Type)：
03 register range：0~124,1000~1124；04 register range：0~124,1000~1124，1125~1249
```

Documentation is not complete there is no UA series for example

wills106, 2024-02-11:
Just had a look at the 1.24 and by 1.20 docs and I have missed out the scaling of 0.1 for `Grid Power` & `Grid Power VA` on the following  Inverters TL-X/TL-XH/TL-XH US

I'll release an update for them two and a couple more corrections.

wills106, 2024-02-11:
Just released [2024.02.3b3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.3b3) with a couple more entities for Single Phase Inverters.

Saentist, 2024-02-11:
> Just released [2024.02.3b3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.3b3) with a couple more entities for Single Phase Inverters.

Any plan when it will be in main release?

wills106, 2024-02-12:
Just released it [2024.02.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.3)

Saentist, 2024-02-17:
`Grid Current` still have wrong multiplier
[image]

wills106, 2024-05-18:
Is this still an issue? Or has it been corrected?

Saentist, 2024-05-19:
Will post ASAP values when there is sun on panels.

Saentist, 2024-05-20:
@wills106 
not fixed yet
[image]

V*A=W (Grid)
234*0.3 = 72≠12.5 (as value show)

flopp999, 2025-04-02:
still problem?

wills106, 2025-08-28:
Closing down as no response.