wills106/homeassistant-solax-modbus #729: Only one Solax inverter possible?[Bug]:

RSX2613, 2024-02-08:
### Describe the bug

Hello, I operate 3 Solax inverters: 1 x X3 Hybrid and 2 x X1 mini.
I can add these and everything works fine. But after restarting HA, out of 2 inverters the entity is no longer available? How could I fix this problem?

### Integration Version

2024.01.1

### Homeassistant core version

2024.2.0

### Inverter brand

Solax 

### Plugin used

plugin_solax.py

### Serial prefix

H34B08

### Inverter firmware versions

ARM 2.03 DSP 2.07

### Connection Method

3 x Solax WiFi 3.0 

### Dongle firmware

3.001.02

wills106, 2024-02-09:
The Integration hasn't been tested a whole lot with multiple Inverters.
The main multi inverter setup I know had two identical Inverters.

I have exposed the Serial Number in the Device Info section with [2024.02.3b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.3b1) does that help the Integration to differentiate between the Inverters correctly?

If it doesn't correct the Inverters following a restart of HA you might need to remove all 3 Instances of the Integration, restart HA and add them back in.

RSX2613, 2024-02-09:
> If it doesn't correct the Inverters following a restart of HA you might need to remove all 3 Instances of the Integration, restart HA and add them back in.
Yes, this is the only way to get all three inverters working again at the same time. But of course after restarting HA the whole thing has to be repeated. However, this is not a really usable method for users with more than one inverter!

wills106, 2024-02-09:
I meant to get them working again after trying out [2024.02.3b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.3b1) then hopefully they'll carry on working correctly and survive restarts etc.

RSX2613, 2024-02-09:
Oh my mistake, I'll test it!

RSX2613, 2024-02-09:
Can you tell me how I can Install the Beta Version?

wills106, 2024-02-09:
Did you use HACS? You can redownload it and select the Beta options.
Then select 2024.02.3b1 from the list.

https://hacs.xyz/docs/faq/select_version

RSX2613, 2024-02-09:
Ahh I find it. Thank You!

wills106, 2024-02-09:
If the Beta is installed correctly each Inverter should display it's serial, under Device info.
Hopefully that will allow HA to know which Inverter is which correctly.

[image]

RSX2613, 2024-02-10:
I tried it today and all three inverters are working. The serial numbers are also displayed correctly.
But unfortunately only 2 of 4 strings are displayed correctly. As either PV1 and PV2 from the Solax X3 or twice PV 1 from my 2 Solax X1 mini.

wills106, 2024-02-11:
Could you try [2024.02.3b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.3b2) I have added the serial number into device_info identifiers.
Hopefully this will allow HA to keep all of the Inverter entities separate from each other.

RSX2613, 2024-02-11:
I tested versions 2024.02.3b2 and 2024.02.3b3.
But unfortunately the X3 inverter doesn't work after restarting HomeAssitant.
Most entities of the X3 are not available after restarting HA.
The two X1s run without problems.

[image]

wills106, 2024-02-11:
> ### Dongle firmware
> 3.001.02

Are you sure all 3 are on that version? As it's over 3 years old now.

You can check by following the [FAQ](https://homeassistant-solax-modbus.readthedocs.io/en/latest/solax-faq/)
The Internal code is the Firmware Version, the one that say Firmware always reads 3.001.02

I'm just wondering if the Pocket WiFi on the X3 is earlier than the X1's and that's what's causing you to keep loosing it. As early Firmwares on the PocketWiFi weren't stable.

wills106, 2024-02-12:
If you add them in a different order do you lose the first type on Inverter added on restart?

1st X3 then the two X1 = Losing the X3 on restart
1st X1, X1 and then X3 = Losing the X1's on restart

RSX2613, 2024-02-12:
No it is not working in a different order.
And all three working on Firmware version 3.001.02.
But I made a mistake on the X3 is a Wifi Dongle V2 workin (the white on) and on the X1 Wifi Dongle V3.

wills106, 2024-02-12:
If you use a PocketWiFi 3.0 on the X3 does it survive a restart and you start getting issues with which ever X1 that then has the PocketWiFi2?

wills106, 2024-02-12:
If you are still getting the same issue where the Hybrid fails try [2024.02.4b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.4b1) I have merged the two SensorEntityDescription's into a single one.

Other than that I'm not sure what else to try as I only have a single Inverter to test on.

RSX2613, 2024-02-12:
Now I can install only version 2024.02.3.
I test it tomorrow!
I noticed that when I switched off the two X1s (no more sun) the X3 worked again?

wills106, 2024-02-12:
I'm thinking https://github.com/wills106/homeassistant-solax-modbus/blob/a558ca3999ec6518d09e2cae932400e1f05003d7/custom_components/solax_modbus/plugin_solax.py#L6526 is stopping your Hybrid from working, while it can read a "MIC" Your X1 Mini's use the same registers as MIC.

Which I have removed in 2024.02.4b1 and merged them into one.

Let me know how you get on tomorrow.

RSX2613, 2024-02-13:
In the new version the X3 works but the X1 no longer works at all. In addition, when adding inverters, they are always added twice.
[image]
[image]
[image]

wills106, 2024-02-13:
The two devices per Inverter was fixed in 2024.02.5 Beta branch [2024.02.5b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.5b2) I forgot to mention it.

RSX2613, 2024-02-13:
I updated to the latest version and added everything new.
The X3 is running but X1 is added but no data is read out. In addition, they now have 77 entities.
When they were still functioning there were 25 entities

[image]

RSX2613, 2024-02-13:
Yes

wills106, 2024-02-13:
Are you able to add the following to your configuration.yaml
```
logger:
  default: info
```
Then in the logs search for solax and "Load Full Logs"
Then download the file and then send it to me. I want to see how it's trying to configure itself.
You can hide any personal info like serials. But leave enough that I can tell which of the 3 Inverters is which.

RSX2613, 2024-02-13:
[home-assistant_2024-02-13T18-01-56.612Z.log](https://github.com/wills106/homeassistant-solax-modbus/files/14270549/home-assistant_2024-02-13T18-01-56.612Z.log)

wills106, 2024-02-13:
Are your two X1's actually in Run mode or have they both gone to sleep now? Due to lack of Solar PV?

RSX2613, 2024-02-13:
sleep now

wills106, 2024-02-13:
That'll be why they won't work at the moment then.

Can you try again tomorrow and send me the log of them trying to communicate.

RSX2613, 2024-02-13:
When Im back from work I will send you a new log.

wills106, 2024-02-14:
I have released [2024.02.5b3](https://github.com/wills106/homeassistant-solax-modbus/tree/2024.02.5b3) I think a number of the 77 entities for each X1 is picking up AC/Hybrid registers. I have been over the AC/Hybrid list and properly assigned them.

I think there are only about 30 or 30 registers for the X1 PV Inverters.

RSX2613, 2024-02-14:
Unfortunately I can't try out the version 2024.02.5b3 because it isn't available for selection in HA.
Otherwise everything is like yesterday, I attached the log file.

[home-assistant_2024-02-14T14-45-51.139Z.log](https://github.com/wills106/homeassistant-solax-modbus/files/14281116/home-assistant_2024-02-14T14-45-51.139Z.log)

wills106, 2024-02-14:
Not sure what was up with 2024.02.5b3? I have just pushed the same out as 2024.02.5b4 which shows up.

[image]

wills106, 2024-02-14:
It looks like I was correct and it's trying to mix Hybrid and PV only registers together.
2024.02.5b4 should get rid of that issue.

RSX2613, 2024-02-14:
Now it works!
A big thank you.
Hopefully it will still work when I put an X3 Mic into operation soon.

[image]

wills106, 2024-02-16:
Did you manage to test with all 3 Inverters?

RSX2613, 2024-02-16:
a short feedback on the current status.
The X3 runs very well, the X1 with 78 entities also runs wonderfully.
Only the X1 with 27 entities has to be added again and again so that it runs again.
I had to add this again after the last update and therefore only has 27 entities, which is probably leading to the problems.
[image]

wills106, 2024-02-16:
I don't think they are setup correctly.
The X1 Mini's shouldn't have 78 entities. 27 Sounds correct.

Personally I would remove all 3 Inverters. Restart and add them back in. Then you know for sure they are all using the corrected code. You still have 2024.02.5b3 code loaded on one of the X1's

RSX2613, 2024-02-17:
All inverters added again.
And again problems with the one X1, first it freezes and the data stops and after restarting it is not available.

[home-assistant_2024-02-17T08-29-06.219Z.log](https://github.com/wills106/homeassistant-solax-modbus/files/14318196/home-assistant_2024-02-17T08-29-06.219Z.log)

wills106, 2024-02-17:
Is that log after restarting HA with the 3 Inverters connected?

wills106, 2024-02-17:
Is IP .84 the Garage Inverter?

RSX2613, 2024-02-17:
Yes IP.84 is the Garage Inverter.

RSX2613, 2024-02-17:
log after restarting HA.

[home-assistant_2024-02-17T14-27-32.983Z.log](https://github.com/wills106/homeassistant-solax-modbus/files/14319135/home-assistant_2024-02-17T14-27-32.983Z.log)

wills106, 2024-02-18:
From what I can see multiple Inverters are working together ok on the latest Beta releases #743 the only difference I can see if the use of Waveshare's on the X1's but I wouldn't expect that to make a difference?

RSX2613, 2024-02-20:
I've tried several times in the last few days but always have the same problem.

[image]
[image]
[home-assistant_2024-02-20T11-38-32.131Z.log](https://github.com/wills106/homeassistant-solax-modbus/files/14343710/home-assistant_2024-02-20T11-38-32.131Z.log)

wills106, 2024-09-15:
Did you get all Inverters working? Can this be closed now, as other users have managed to get multiple SolaX Inverters working together.

RSX2613, 2024-09-18:
Yes all works good. Can be closed.

closed 2024-09-18 by wills106