wills106/homeassistant-solax-modbus #731: [Bug]: Integration causes user interface hangs when reading Modbus data

Zwer2k, 2024-02-08:
### Describe the bug

Every time the Modbus data is read out, the Homeassistant UI hangs for a few seconds, which is very annoying. Events from devices are also delayed. If the integration is deactivated, the error disappears. The problem occurs both in the live system and in the HA development environment. It looks like the method self._client.read_holding_registers() is responsible.

I suspect that the library must be used as async to fix this.
from pymodbus.client import AsyncModbusTcpClient, AsyncModbusSerialClient

### Integration Version

2024.02.1

### Homeassistant core version

2024.2.0

### Inverter brand

Sofar Solar HYD 10KTL-3ph

### Plugin used

plugin_sofar.py

### Serial prefix

SP2ES110

### Connection Method

RS485-TCP with Waveshare TCP Adapter

wills106, 2024-02-09:
I have just updated from HA 2024.1.6 to HA 2024.2.1 without issue. Apart from a warning for commented out code in config_flow.py which has been addressed in [2024.02.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.2)

Does the Integration develop any errors in the logs?
How quick are you trying to poll the Inverter?

Zwer2k, 2024-02-10:
Apart from the warning about custom integration, I don't get any errors or warnings from Solax Integration.
I have already tried different speeds 9600, 57000 baud and others. 
The problem also existed with older versions of HA and the integration. 
My HA runs as a VM under Proxmox with an i5-8250U CPU. Should actually be better than a suitable Raspberry Pi 4.

wills106, 2024-02-10:
The baud rate needs to match whatever is on the Inverter LCD.

I was referring to the polling rate, when you setup the Integration.

[image]

Zwer2k, 2024-02-12:
Die Baudrate wurde selbstverständlich sowohl in der Integration als auch im Wechselrichter eingestellt.
Der pulling-Interval steht bei mir auf 15 sek. Hab aber auch bereits mit 30sek. getestet gehabt. Dann tritt das problem etsprechend seltener auf.
Modbus Gateway Type steht bei mir auf "Auto query storage type"

cschlipf, 2024-02-20:
No solution, but I can just add that I cannot reproduce this with a HYD 10 KTL 3PH, which is connected using ModBus TCP to a LSE-3 logger stick on a Rasperry PI4 at 15s polling interval.

fische11, 2024-02-20:
I sldo have this issue while using an  LSE-3 LAN Stick Logger with an HYD 10 KTL 3PH. I also get these error logs. Do I need to change the ba on the inverter for the  LSE-3 LAN Stick Logger?

`2024-02-20 23:53:38.859 WARNING (MainThread) [homeassistant.helpers.entity] Update of binary_sensor.rpi_power_status is taking over 10 seconds
2024-02-20 23:56:10.904 WARNING (MainThread) [homeassistant.helpers.entity] Update of binary_sensor.rpi_power_status is taking over 10 seconds
2024-02-20 23:56:41.064 WARNING (MainThread) [homeassistant.helpers.entity] Update of binary_sensor.rpi_power_status is taking over 10 seconds
2024-02-21 00:20:30.493 INFO (MainThread) [custom_components.solax_modbus] read_error : Sofar01 cannot read holding registers at device 1 position 0x584
NoneType: None
2024-02-21 00:20:30.494 DEBUG (MainThread) [custom_components.solax_modbus] assuming sleep mode - slowing down by factor 10
2024-02-21 00:22:58.420 INFO (MainThread) [custom_components.solax_modbus] read_error : Sofar01 cannot read holding registers at device 1 position 0x404
NoneType: None
2024-02-21 00:22:58.420 DEBUG (MainThread) [custom_components.solax_modbus] assuming sleep mode - slowing down by factor 10
2024-02-21 00:25:05.549 ERROR (MainThread) [homeassistant.components.hassio.handler] Timeout on /store request
2024-02-21 00:25:05.550 ERROR (MainThread) [homeassistant.components.hassio] Failed to to call /store - 
2024-02-21 00:28:14.499 INFO (MainThread) [custom_components.solax_modbus] read_error : Sofar01 cannot read holding registers at device 1 position 0x445
NoneType: None
2024-02-21 00:28:14.499 DEBUG (MainThread) [custom_components.solax_modbus] assuming sleep mode - slowing down by factor 10
2024-02-21 00:30:58.637 INFO (MainThread) [custom_components.solax_modbus] read_error : Sofar01 cannot read holding registers at device 1 position 0x605
NoneType: None
2024-02-21 00:30:58.638 DEBUG (MainThread) [custom_components.solax_modbus] assuming sleep mode - slowing down by factor 10
2024-02-21 00:32:18.659 DEBUG (MainThread) [custom_components.solax_modbus] assuming sleep mode - slowing down by factor 10`

Zwer2k, 2024-02-22:
Is fixed. @wills106 thanks for the merge and the great integration.

closed 2024-02-22 by Zwer2k