wills106/homeassistant-solax-modbus #735: [Bug]: Missing all houseload, import, export data.

aratel88, 2024-02-12:
### Describe the bug

I've got the entitys but there is no data.
Using a MOD10KTL3-XH inverter, serial number starts with DFK.
I do not think this integration can read my serialnumber.

### Integration Version

2024.02.2

### Homeassistant core version

2024.02

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

DN1.0

### Connection Method

RS485 to USB adapter

wills106, 2024-02-12:
Do you get either of the two described errors in your log?
https://github.com/wills106/homeassistant-solax-modbus/discussions/523

aratel88, 2024-02-12:
No i only get:4
Detected blocking call to sleep inside the event loop by custom integration 'solax_modbus' at custom_components/solax_modbus/__init__.py, line 344: return self._client.read_holding_registers(address, count, **kwargs), please create a bug report at https://github.com/wills106/homsassistant-solax-modbus/issues

aratel88, 2024-02-12:
[image]

aratel88, 2024-02-12:
[image]

aratel88, 2024-02-12:
Yes i tryed to change the inverter to another model but changed back.
Did reload the integration now and now I just got 45 entities.
Im using a Chint smart meter.
[image]

wills106, 2024-02-12:
Looks like I tried to add this inverter #374 but I didn't get any further feedback.
Looking at the later docs the MOD shares the same registers as the MIN.
Try changing it to a GEN4

aratel88, 2024-02-12:
Yes that did work, but the gen4 is only one phase, mine is 3 phase so dont get the correct values.

wills106, 2024-02-12:
What values are wrong?

There should be support for 3Phase.
https://github.com/wills106/homeassistant-solax-modbus/blob/e4c4f436780009eeb175134f7a19c6905293b68c/custom_components/solax_modbus/plugin_growatt.py#L3402-L3412

aratel88, 2024-02-12:
Heres the entities with gen4: Not correct.
[image]
Heres gen3: correct
[image]

wills106, 2024-02-12:
Could you get a list of all entities that 
work on Gen3
don't work on Gen3
work on Gen4
don't work on Gen4
I might need to use a mixture of both types.

I'm guessing the docs aren't correct for this particular Inverter type.

aratel88, 2024-02-12:
All entities works on gen 3, just missing import/export, house load. All entities that include smart meter.
Gen 4 everything is wrong except: Total Pv Power, PV Voltage 1, PV Voltage 2, PV Power 1, PV Power 2, PV Current 1, PV Current 2, Grid Voltage L1, Grid Voltage L2, Grid Voltage L3, Grid Frequency

wills106, 2024-02-12:
When you had these https://github.com/wills106/homeassistant-solax-modbus/issues/735#issuecomment-1938342300 did you setup your Inverter as a Gen or Gen2?

aratel88, 2024-02-12:
Gen2

wills106, 2024-02-12:
I have just had another look at the docs and the only "Today's Grid Import" is either
https://github.com/wills106/homeassistant-solax-modbus/blob/e4c4f436780009eeb175134f7a19c6905293b68c/custom_components/solax_modbus/plugin_growatt.py#L2806-L2819

Or
https://github.com/wills106/homeassistant-solax-modbus/blob/e4c4f436780009eeb175134f7a19c6905293b68c/custom_components/solax_modbus/plugin_growatt.py#L3705-L3718

So I'm not sure what else to try.

wills106, 2024-02-12:
Are you able to view "Today's Grid Import" or equivalent naming on either the Inverter it's self or the clod app?

aratel88, 2024-02-12:
Yes i can see it on my inverter app.

aratel88, 2024-02-12:
Still not right.

wills106, 2024-02-12:
Ok I was wondering if the endian is different for 32bit registers like on SolaX Inverters.

With it set as Gen4 and `order32 = Endian.LITTLE,` are the values closer to what's in the app? Perhaps the scale needs adjusting?

aratel88, 2024-02-12:
Dont think the problem is scaling,
[image]
todays grid export is 0, for example.
heres a screenshot from growatt server.
[image]

wills106, 2024-02-12:
I'm not sure what else to suggest at the moment then?

You could try and ask Growatt for more specific Modbus documentation for your model of Inverter?

aratel88, 2024-02-13:
Yes I will contact my contractor who installed it.

wills106, 2024-02-16:
I have just had a look at the docs and 3041, 3043 & 3045 should have been 0.1 like you mention.

Are you able to submit a PR with these changes?
The only thing I ask is that you swap Gen4 & Gen3 round the other way. So the style is the same as the other multigeneration entries. `GEN3 | GEN4`

aratel88, 2024-02-16:
Not sure how to do it. I will just upload the file here.
[plugin_growatt.zip](https://github.com/wills106/homeassistant-solax-modbus/files/14313142/plugin_growatt.zip)

wills106, 2024-02-17:
I have just released [2024.02.5b7](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.5b7) with the changes you have made.

wills106, 2024-02-17:
Are you also missing Battery Charge / Discharge Power and the "Battery Combined Power" entities as per #718 and #749 
If so I think I have spotted what's wrong with it.

iobrokerneu, 2024-02-17:
yes, I'm missing the data #718 

Greetings and thanks

wills106, 2024-02-18:
Just released [2024.02.5b8](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.5b8) with those changes as I don't think they went in 2024.02.5b7 and the ones in #718

wills106, 2024-02-18:
@aratel88 Does your Inverter return AC Power to User L1-3 and AC Power to User Total?
Just noticed on #718 they just return 0. I was wondering if they got added by mistake?

iobrokerneu, 2024-02-18:
Hello . I'm not sure if I should answer for #718. When updating from 7 to 8 the entities were installed, but unfortunately without data. Greetings Sascha

wills106, 2024-02-18:
That's why I'm asking aratel88 if they have AC Power to User L1-3 and AC Power to User Total returning zero like you are getting.
I am wondering if they got added by mistake if the return zero for both of you.

If you both are getting zero I'll remove them.

aratel88, 2024-02-19:
> @aratel88 Does your Inverter return AC Power to User L1-3 and AC Power to User Total? Just noticed on #718 they just return 0. I was wondering if they got added by mistake?

Yes it was by mistake. I will check whats working on my inverter and update it here. But i dont have battery so i dont know if thoese entities will work.

aratel88, 2024-02-19:
Now its complete.
[plugin_growatt.zip](https://github.com/wills106/homeassistant-solax-modbus/files/14326968/plugin_growatt.zip)

wills106, 2024-02-19:
Can this be closed down then?

aratel88, 2024-02-19:
Yes I am happy

closed 2024-02-19 by wills106

Zvolli, 2024-02-19:
@wills106 Been following this issue since i am also missing same import/load and export data on my MOD. The latest 2024.05.2  shows in the changelog that this should be solved but after updating it it's not working. Maybe i am misunderstanding something here but please clarify. :)

wills106, 2024-02-19:
Once installing 2024.02.5 have you tried removing the Integration from the Integration Page, restarting HA and adding the Inverter again?
That way you are sure the Integration is using new code and not old cahced.

Zvolli, 2024-02-19:
> Once installing 2024.02.5 have you tried removing the Integration from the Integration Page, restarting HA and adding the Inverter again? That way you are sure the Integration is using new code and not old cahced.

Yes, removed solax modbus and redownloaded and added the integration but with same results. Not sure if it's relevant but i'm running HA in docker container.

Zvolli, 2024-02-19:
yes, complete restarting but I can try via docker also

Zvolli, 2024-02-19:
same results, all import, export, load metrics are zeroed

[image]

[image]

wills106, 2024-02-19:
The AC Power shouldn't be there. I forgot to remove them.

Zvolli, 2024-02-19:
Both are found as entities but zeroed.

[image]

wills106, 2024-02-19:
Maybe they should be removed as well if they don't return anything for aratel88
Unfortunately this Inverter isn't properly documented, so it's a bit trial and error.

Zvolli, 2024-02-19:
> Maybe they should be removed as well if they don't return anything for aratel88 Unfortunately this Inverter isn't properly documented, so it's a bit trial and error.

Fully understand. Appreciate your efforts

wills106, 2024-02-19:
@iobrokerneu Does your Inverter return anything for "Total Forward Power", "Total Reverse Power" & "Total Load Power"
Or is yours zero as well?

iobrokerneu, 2024-02-19:
hi, total forward power and total reverse power is zero. Data is displayed at total load power.

wills106, 2024-02-19:
I have just released [2024.02.6b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.6b1) with "AC Power to x" removed. I have left "Total Forward Power", "Total Reverse Power" & "Total Load Power" in for now until we have better understanding of them.

There are some new entities for battery. They are best being discussed in #757 to keep the conversation in one place.

Zvolli, 2024-02-19:
ok, just updated and they seem to still be zeroed. no worries about battery the MOD is not a Hybrid.
Let's see what it collects....

wills106, 2024-02-19:
O, so is there a Hybrid and Non Hybrid version of the MOD then?

Zvolli, 2024-02-19:
Absolutely. the "H" in the model number indicates Hybrid. Mine is MOD 10KTL3-X whereas MOD 10KTL3-XH indicates a Hybrid.

wills106, 2024-02-19:
We might need to expand the Firmware version detection.

On the Integration Page under Device what's displayed in Device info for the Serial on everyone's Inverter?
It says it's the Serial but it's actually the Firmware Version on Growatt.

Zvolli, 2024-02-19:
The non hybrid says DL1.0ZBAA

Michael Svoldgaard
Hulvejen 190
DK-9530 Støvring

Zvolli, 2024-02-20:
@wills106 is DL1.0ZBAA the correct serial or a transcoded version of Inverter serial? I'm asking since they don't match according to Growatt inverter label: RKH9CE4054 and DL1.0 corresponds to the build ID.

Zvolli, 2024-02-20:
@wills106 maybe this can be helpful with registry ID's
[Growatt-Inverter-Modbus-RTU-Protocol-II-V1-24-English-new.pdf](https://github.com/wills106/homeassistant-solax-modbus/files/14342073/Growatt-Inverter-Modbus-RTU-Protocol-II-V1-24-English-new.pdf)

wills106, 2024-02-20:
That's what I'm using thanks.

On Growatt I'm using what I understand to be the Firmware version. As the serial number is truly unique unlike other brands.
On other brands the first 4-6 Characters of the serial indicate the model type.

I'm using 9-10, perhaps we might need to use 12-14 as well? Or 28 & 29 after the serial.
[image]

Zvolli, 2024-02-20:
> That's what I'm using thanks.
>
> On Growatt I'm using what I understand to be the Firmware version. As the serial number is truly unique unlike other brands. On other brands the first 4-6 Characters [...]

ah ok got it. Yes that should indicate a model and as there is many different models on the MOD series, you might want to include 12-14.

wills106, 2024-02-22:
I need to add them into the plugin_growatt.py but need to wait finish merging #762 otherwise there will be too much code difference between different releases.

Zvolli, 2024-02-22:
> I need to add them into the plugin_growatt.py but need to wait finish merging #762 otherwise there will be too much code difference between different releases.

No problem, fully understand

wills106, 2024-02-22:
Just realised Register 12-14 are already added as "Firmware Control Version" but disabled by default.

I'll be adding register 28 & 29 soon as "Inverter Module"

Zvolli, 2024-02-22:
> Just realised Register 12-14 are already added as "Firmware Control Version" but disabled by default.
> 
> I'll be adding register 28 & 29 soon as "Inverter Module"

Got it

aratel88, 2024-02-22:
> @aratel88 does your Inverter return "Total Forward Power" & "Total Load Power"

Yes my invertver returns forward power, load power and revers power.

[22 more replies]