wills106/homeassistant-solax-modbus #737: [Bug]: big delay in reading data from inverter

josefkrtil, 2024-02-12:
### Describe the bug

From version 2024.2.1 I have problem with long delay between reading data. Before 2024.2.1. it takes few second or minutes to read changes from inverter and from 2024.2.1 it take in some cases 60 minutes. You can see it in history logs from battery capacity.
[history.csv](https://github.com/wills106/homeassistant-solax-modbus/files/14240546/history.csv)

### Integration Version

2024.2.1

### Homeassistant core version

2024.2.1

### Inverter brand

Sofar

### Plugin used

plugin_sofar.py

### Serial prefix

SP2ES1

### Connection Method

RS485 - USB

wills106, 2024-02-12:
What version of the Integration were you on when it was working ok?

The only thing that changed in 2024.02.1 for Sofar is the following.
https://github.com/wills106/homeassistant-solax-modbus/commit/6750846cdd45fb64f2d10b455b65d9dbc4da3d0b

josefkrtil, 2024-02-12:
2024.1.8

josefkrtil, 2024-02-14:
I upload backup before changes and still the same. I had to reconfigure serial communication, set higher baud rate and it works. So I close this ticket.

closed 2024-02-14 by josefkrtil