wills106/homeassistant-solax-modbus #742: [Bug]:  After the update, the inverter was split into two items

vitjurasek, 2024-02-14:
### Describe the bug

After a normal update within Home Assistant, a new version of MODBUS integration for Solax G4 was installed. Originally one instance of the inverter has split into two. The original one is working. The other is completely dead, but cannot be deleted.
After a reboot of HA it now takes several minutes, sometimes tens of minutes, to establish communication with the inverter.

[image]
[image]

### Integration Version

2024.02.4

### Homeassistant core version

2024.2.1

### Inverter brand

SolaX Power G4

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

-

### Connection Method

Solax pocket wifi

### Dongle firmware

-

wills106, 2024-02-14:
This was introduced in 2024.02.3 and corrected in 2024.02.4
If it hasn't corrected it's self press the 3 dots to delete it. Then add it back in again.

If you name it the same all the history will still be there.

matomatusov, 2024-02-14:
Hi, I have a similar problem.
It also appeared to me when I updated to 2024.02.4.

matomatusov, 2024-02-14:
@wills106 your delete and add procedure didn't help.

wills106, 2024-02-14:
I just downgraded and it is indeed still there.

[2024.02.5b4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.5b4) is working fine.
You will possibly need to delete and read it. As 2024.2.5 is changing the way multiple Inverters are handled.

matomatusov, 2024-02-14:
@wills106
solax shadow fix function level split into 2 entities?

SolaX Shadow Fix Function Level PV1 (GMPPT)
SolaX Shadow Fix Function Level PV2 (GMPPT)

"You will possibly need to delete and read it."
I do not understand the meaning to read?

Thanks

wills106, 2024-02-14:
> solax shadow fix function level split into 2 entities?
> 
> SolaX Shadow Fix Function Level PV1 (GMPPT) SolaX Shadow Fix Function Level PV2 (GMPPT)

That got added in with #599 and released in [2024.01.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.01.3)

> "You will possibly need to delete and read it." I do not understand the meaning to read?

It's a typo should be "readd" (to add again)

> If it hasn't corrected itself press the 3 dots to delete it. Then add it back in again.
> 
> If you name it the same all the history will still be there.

vitjurasek, 2024-02-14:
> That got added in with #599 and released in [2024.01.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.01.3)
>
>
> It's a typo should be "readd" (to add again)

Not work for me.

Stepsilein, 2024-02-14:
No way, ist a terrible bug still.
I used all posdible versions, even beta ...2.5b.
But the Fake inverter still dhown (but dead as always)
I stopped and deleted the integration, HA restart snd installed the 2024.2.2 (last bugfree working version) again.
Now clean 1 inverter dhown working nicd as before.

I will definitily wait for a future bugfree version... or never ever update to that buggy ones.
Keep on the good work, but try to workbout that bug please.
🙋🏼‍♂️

wills106, 2024-02-14:
You can go from 2024.02.2 to 2024.02.5b4 without an issue.

Although if you own an SolaX Air, Boost, Mini or MIC / MIC Pro you will possibly need to delete and readd the Inverter, as part of the change for correctly supporting multiple Inverter installations.

cschlipf, 2024-02-14:
I have the same problem with a Sofar Inverter after updating today.

Device 1 shows most of the sensors and some controls
[image]

Device 2 shows options:
[image]

wills106, 2024-02-14:
Either go back to 2024.02.2 or use 2024.02.5b4

> If it hasn't corrected it's self press the 3 dots to delete it. Then add it back in again.
> 
> If you name it the same all the history will still be there.

You may need to do the above.

wills106, 2024-02-14:
2024.02.3 and 2024.02.4 have been deleted from the Releases page.

wills106, 2024-02-14:
Hopefully 2023.02.5b4 will be pushed out soon as a full release and fix for #729 which will benefit anyone with multiple Inverter installs.
I appreciate it's a pain at times when this Integration breaks, but it's difficult to test every scenario with only a single Inverter out of the vast many supported so far.

cschlipf, 2024-02-14:
Totally understood. Thanks so much for your awesome work. Enjoy your Ko-Fi ;-)

Stepsilein, 2024-02-14:
Thanks a lot for your work with this veeery helpful Integration 😍👍👍👍👍

sahomm, 2024-02-14:
I observed the same issue, search about that and I read it's observed.

202!!!4!!!.02.5b4 tested, done! Grate job, well done! 👍

wills106, 2024-02-19:
I'm closing this down as I have released [2024.02.5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.5)

closed 2024-02-19 by wills106