wills106/homeassistant-solax-modbus #743: No Serial Number for SolaX X1 Mini 3.0 and missing configuration entity for manipulate "ActivePowerLimit"

edli83, 2024-02-15:
### Describe the bug

I never get a Serial Number for my 3 SolaX X1 Mini 3.0 inverters.
Might be, that therefore I also do not get some entitys. Currently I urgently miss configuration for "ActivePowerLimit" out of the Super-Configration menu from SolaX App.
My goal is, to turn down these 3 X1 inverters in OffGrid-Mode of the central X3 Hyb. G4 inverter in case battery is getting full charged to avoid any damage. These 3 X1 inverters have in total 6kW peak, which I want be part to charge battery also in OffGrid-Mode. But unfortunately they can not be managed directly by the X3 inverter. So I have to use HA for this job.
Hopefully you could help.
Many Thanks in advice

### Integration Version

2024.02.5b5

### Homeassistant core version

2024.2.1

### Inverter brand

SolaX

### Plugin used

plugin_solax.py

### Serial prefix

H34A10I

### Inverter firmware versions

Arm 1.35 DSP 1.36

### Connection Method

SolaX Pocket WiFi 3.0

### Dongle firmware

3.001.02

### Detailed Error Log

2024-02-15 07:49:15.653 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX_04: reading serial number from address 0x0 failed; other address may succeed
2024-02-15 07:49:15.719 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX_02: reading serial number from address 0x0 failed; other address may succeed
2024-02-15 07:49:15.787 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX_03: reading serial number from address 0x0 failed; other address may succeed
2024-02-15 07:49:15.843 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_04_grid_frequency already exists - ignoring sensor.solax_04_inverter_frequency
2024-02-15 07:49:15.844 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_04_run_mode already exists - ignoring sensor.solax_04_run_mode
2024-02-15 07:49:15.848 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_04_grid_export already exists - ignoring sensor.solax_04_grid_export
2024-02-15 07:49:15.848 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_04_grid_import already exists - ignoring sensor.solax_04_grid_import
2024-02-15 07:49:15.856 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_02_grid_frequency already exists - ignoring sensor.solax_02_inverter_frequency
2024-02-15 07:49:15.856 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_02_run_mode already exists - ignoring sensor.solax_02_run_mode
2024-02-15 07:49:15.857 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_02_grid_export already exists - ignoring sensor.solax_02_grid_export
2024-02-15 07:49:15.857 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_02_grid_import already exists - ignoring sensor.solax_02_grid_import
2024-02-15 07:49:15.865 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_03_grid_frequency already exists - ignoring sensor.solax_03_inverter_frequency
2024-02-15 07:49:15.865 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_03_run_mode already exists - ignoring sensor.solax_03_run_mode
2024-02-15 07:49:15.869 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_03_grid_export already exists - ignoring sensor.solax_03_grid_export
2024-02-15 07:49:15.869 ERROR (MainThread) [homeassistant.components.sensor] Platform solax_modbus does not generate unique IDs. ID SolaX_03_grid_import already exists - ignoring sensor.solax_03_grid_import

### Additional context

The SolaX X1 have Serial prefix XM3302F and are connected via Waveshare Eth
With the central X3 Hyb connected are 2 T58 Batteries and a SolaX Wallbox EV Charge 11 kW, connected over COM Port.
This unfortunately means I am not able to connect also the X3 via Waveshare caus the COM port is blocked by the Wallbox Modbus.

wills106, 2024-02-17:
When you setup the Integration for each Inverter did you give them a Unique name?
[image]
Ie
SolaX 1
SolaX 2
SolaX 3

or
SolaX Garage
SolaX East
SolaX West
etc

edli83, 2024-02-17:
Hi,
thank you for your quick response.

Yes, I called them SolaX_01_Haus_Ost, SolaX_02, SolaX_03, SolaX_04.
[image]

Yesterday I updated the SolaX integration to the Beta 2024.02.4.5.
Now I get, after activating the entities luckily the serial numbers of all inverters. --> Great success :-) !
Unfortunately from the X1´s I still do not get the ARM/ DSP versions - not really important, but it was good to know the current firmware also via HA.
Do you know or have a tipp how I can control the Energy output of the X1´s via a HA automization? It seems, that the control of "ActivePowerLimit" does not work. I have tested it manually via the SolaX App. But set to 0% the inverter still provides the full PV power.
It would be very helpful if you have an idea.

You wrote in your description it could be possible to connect more than one inverter to one waveshare. I tried this with parallel Modbus-connection to the waveshare in 2 different ways - see picture.
[image]
Also I gave each inverter modbus a separate channel. But in integration I do not get the entities filled. Maybe you also have an idea how I can manage this.

Very much thanks to your help
Greetings from Bavaria

edli83, 2024-02-17:
Currently I use 3 waveshare´s - for each inverter one.
This works fine, but needs 3 Eth pins at the switch.

wills106, 2024-02-17:
> Yesterday I updated the SolaX integration to the Beta 2024.02.4.5.
> Now I get, after activating the entities luckily the serial numbers of all inverters. --> Great success :-) !

So you are no longer getting:
```
unique IDs. ID SolaX_04_grid_frequency already exists - ignoring sensor.solax_04_inverter_frequency
unique IDs. ID SolaX_03_grid_frequency already exists - ignoring sensor.solax_03_inverter_frequency
```

edli83, 2024-02-17:
I get entity "Inverter Frequency" for all 4 inverters, but no "grid frequency" (for X3 Hyb inverter I get this entity, but without value)

Maybe current log could help - here the entries of all SolaX_xx lines:

2024-02-17 07:45:56.461 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX_02: reading serial number from address 0x0 failed; other address may succeed
2024-02-17 07:46:01.468 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX_02: reading serial number from address 0x300 failed; other address may succeed
2024-02-17 07:46:01.468 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX_02: cannot find serial number, even not for MIC
2024-02-17 07:46:01.468 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown
2024-02-17 07:46:05.172 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX_04: reading serial number from address 0x0 failed; other address may succeed
2024-02-17 07:46:09.422 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX_04: reading serial number from address 0x300 failed; other address may succeed
2024-02-17 07:46:09.422 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX_04: cannot find serial number, even not for MIC
2024-02-17 07:46:09.422 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown
2024-02-17 07:46:14.430 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX_03: reading serial number from address 0x0 failed; other address may succeed
2024-02-17 07:46:19.437 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX_03: reading serial number from address 0x300 failed; other address may succeed
2024-02-17 07:46:19.437 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX_03: cannot find serial number, even not for MIC
2024-02-17 07:46:19.437 ERROR (MainThread) [custom_components.solax_modbus.plugin_solax] unrecognized inverter type - serial number : unknown
2024-02-17 10:03:49.983 ERROR (MainThread) [pymodbus.logging] Connection to (10.21.5.110, 502) failed: [Errno 111] Connection refused
2024-02-17 10:06:19.996 ERROR (MainThread) [pymodbus.logging] Connection to (10.21.5.110, 502) failed: [Errno 111] Connection refused
2024-02-17 11:04:23.366 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX_02: reading serial number from address 0x0 failed; other address may succeed
2024-02-17 11:04:23.432 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX_04: reading serial number from address 0x0 failed; other address may succeed
2024-02-17 11:04:23.498 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX_03: reading serial number from address 0x0 failed; other address may succeed

wills106, 2024-02-17:
I'm not really sure what the difference between Inverter Frequency and Grid Frequency is.
The Grid one is disabled by default. I don't have a Gen4 Hybrid so I can't test that out.

edli83, 2024-02-17:
Maybe you have ideas to the other 2 questions?
Thank you in advice

edli83, 2024-02-18:
Hi,
which resistor I would need - 120 Ohm? Do I need only 1 resistor after last inverter?
And yes I gave each inverter a different modbus adress (1 / 2 / 3). Is it relevant which adress have to be at the 1st inverter and the last inverter? I understand, this should not have any influence.
Maybe it is only that I do not use a terminating resistor - hopefully.
I will test this, but I need first this resistor - may last a few days ...

Tomorrow I will check the new update, today it´s already dark.

wills106, 2024-02-18:
According to https://know.innon.com/bias-termination-rs485-network you will need them on the Waveshare and last Inverter.
I don't use RS485 so I don't know how critical they are on SolaX Inverters.

I wouldn't expect the order you assign or add the Inverter matters?

edli83, 2024-02-25:
Hi,
today I installed the 2 resistors and what should I say - it works fine with 1 waveshare.
Very good - I´m happy.
Thank you for your tipp and your good work with the integration.

So now my open topics are:
1. Firmware ARM and DSP also from the X1´s
2. Manage the output of the X1´s with a configurable entity

If you have some idea it would be very helpful.

Greetings from Bavaria

edli83, 2024-02-25:
Might be helpful - this is the Modbus description I get from my SolaX distributor.

[SolaxPower_Single_Phase_External_Communication_Pro_231110_121454.pdf](https://github.com/wills106/homeassistant-solax-modbus/files/14397064/SolaxPower_Single_Phase_External_Communication_Pro_231110_121454.pdf)

edli83, 2024-02-25:
And this I got for also X1 and X3 inverters

[Hybrid-X1X3-G4-ModbusTCPRTU-V3.21-English_0622-public-version.pdf](https://github.com/wills106/homeassistant-solax-modbus/files/14397074/Hybrid-X1X3-G4-ModbusTCPRTU-V3.21-English_0622-public-version.pdf)

edli83, 2024-02-27:
Hi,
The new update works great, all firmware versions are now visible.

I tested the “active power limit” manually with the Solax Cloud app – set to 0%.
But unfortunately the “Grid Export” and “Measured Power” entities remain with power – no change. It seems that this setting value does not work.
I suspect I'll have to disconnect the X1s with a Shelly or a contactor.

Thanks a lot for your help and your expertise.
Greetings from Bavaria

wills106, 2024-02-29:
I'm closing this down as I emailed SolaX to ask if there were any updated docs.
They claimed there isn't.

If any become available in the future I can relook into this issue.

closed 2024-02-29 by wills106