wills106/homeassistant-solax-modbus #744: [Bug]: Swatten inverter : no devices or entities

lgisler72, 2024-02-15:
### Describe the bug

Swatten : no devices or entities
[home-assistant_solax_modbus_2024-02-15T16-18-22.093Z.log](https://github.com/wills106/homeassistant-solax-modbus/files/14299520/home-assistant_solax_modbus_2024-02-15T16-18-22.093Z.log)

### Integration Version

2024.02.5b5

### Homeassistant core version

2024.2.1

### Inverter brand

Swatten

### Plugin used

plugin_growatt.py

### Serial prefix

S0067692380000910000 

### Connection Method

Waveshare RS485 to ETH

wills106, 2024-02-15:
We could have carried on the discussion / development in the discussion page.
But I'll stick to using the issue going forward, then the conversation will be in one place.

Good news is that the Inverter is responding and I can tell if it's Single or 3Phase from what it returns. I was worried I would need to probe another register to workout if it was Single or 3Phase.

wills106, 2024-02-15:
I have just released [2024.02.5b6](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.5b6) hopefully it will return a few entities now.

lgisler72, 2024-02-15:
Now it is showing something; It returns : 1device, 19 entities.
Here is the log file: 
[home-assistant_solax_modbus_2024-02-15T20-55-09.131Z.log](https://github.com/wills106/homeassistant-solax-modbus/files/14302361/home-assistant_solax_modbus_2024-02-15T20-55-09.131Z.log)
Thank you.
(I will be away for 2 days but I'm looking forward to continuing the tests as soon as possible)

wills106, 2024-02-15:
Can you take a screenshot of the values returned?

From the "error log" it's looking promising though!

lgisler72, 2024-02-15:
Here you have
[image]
[image]

wills106, 2024-02-15:
Ok further work is required on PV current, and PV Total.
Also "Phase" isn't returning what I hoped for. Luckily "Model Type" can be used to determine if the Inverter is Single or 3Phase.
So "Phase" can be removed.

Would be good to see a screenshot during the day when the PV is fully active.

wills106, 2024-02-17:
I have just released [2024.02.5b7](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.5b7)
I have changed endianness on order32 to LITTLE so `PV Power Total` & `Total Output Power` should return better looking values.
RTC has been changed to an input register, to see if it returns the time (Don't use `Sync RTC` yet!)
Removed `Phase` as didn't provide what's expected and no longer required.

wills106, 2024-02-18:
I just wanted to check a couple of registers were working correctly before adding the rest in with the previous docs.
Have you been able to check  `PV Power Total` & `Total Output Power` since I changed the endianness?

wills106, 2024-02-19:
I have released [2024.02.5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.5) containing more entities.
Could you let me know if they look ok or if any scaling needs changing?

lgisler72, 2024-02-20:
I have solar panels for ~6 year and rcently had the oportuniti to buy the Swatten inverter and the battery. As the hibrid inverter is consuming ~120W permanently so I keep my solars panel on the SolarEdge inverter and do an automation to turn ON/OFF the Swatten inverter dependign of the Battery SOC and solar production. So PV voltage at 0V on the swatten is normal.
(Today it is raining; Heat-pump is running)
[image]
[image]

lgisler72, 2024-02-20:
[image]

wills106, 2024-10-13:
Do you think this can be closed down now? Or are they any missing sensors?
I know you can't test the PV side of things.

lgisler72, 2024-10-14:
Hello, Yes you can close the case now. Thank you fro your work.

closed 2024-10-14 by wills106