wills106/homeassistant-solax-modbus #753: [Bug]: Solax x3 G4 Parallel setup, Remotecontrol won't work

Evarra, 2024-02-17:
### Describe the bug

I have been using the Solax X3 Gen4 inverter with Remotecontrol for a long time, and everything has been working great.

Now I decided to buy an additional inverter and connect them in parallel.

The inverters are working together and have been configured.

The problem is that after connecting them in parallel, the Modbus remote control no longer works.

If I switch from Parallel Setting - Master to Free, everything starts working again.
However, if I switch from Free to Master while Remote Control is active, both inverters start showing Normal (R) status, and after a while, they switch to Normal mode.

Sometimes, the solution seems to work again, but it reacts very slowly. For example, when you change the power or mode, it takes 5-8 minutes, and sometimes you have to press the trigger button again.

Also, the Remotecontrol Autorepeat Remaining timer does not advance after pressing the trigger button; the time appears, but it doesn't count down as usual.

After switching back from parallel mode: Master to Free, everything starts working again automatically.

### Integration Version

2024.02.5b7

### Homeassistant core version

2024.2.2

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

H34A15

### Inverter firmware versions

ARM 1.36 DSP 1.36

### Connection Method

USR-IOT LAN-RS485

### Dongle firmware

no dongle

### Detailed Error Log

2024-02-17 18:52:05.701 INFO (MainThread) [custom_components.solax_modbus] saved modified persistent date: {'_version': 1, 'remotecontrol_active_power': 5000.0, 'remotecontrol_reactive_power': 0.0, 'remotecontrol_duration': 20.0, 'remotecontrol_autorepeat_duration': 28800.0, 'remotecontrol_import_limit': 16000.0, 'config_export_control_limit_readscale': 1.1, 'config_max_export': 36400.0, 'remotecontrol_power_control': 'Enabled Battery Control', 'remotecontrol_set_type': 'Set'}
2024-02-17 18:52:05.701 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] local data update callback
2024-02-17 18:52:05.702 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] local data update callback for read_scale: 1.1 enabled: True
2024-02-17 18:52:05.702 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] local data update callback for entity: remotecontrol_active_power new limit: 36400.0
2024-02-17 18:52:05.703 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] local data update callback for entity: remotecontrol_import_limit new limit: 36400.0
2024-02-17 18:52:05.703 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] local data update callback for entity: export_control_user_limit new limit: 36400.0
2024-02-17 18:52:05.703 DEBUG (MainThread) [custom_components.solax_modbus.plugin_solax] Evaluated remotecontrol_trigger: corrected/clamped values: [('remotecontrol_power_control', 'Enabled Power Control'), ('remotecontrol_set_type', 'Set'), ('remotecontrol_active_power', 5000.0), ('remotecontrol_reactive_power', 0.0), ('remotecontrol_duration', 20.0)]
2024-02-17 18:52:05.704 DEBUG (MainThread) [custom_components.solax_modbus] ready to repeat button remotecontrol_trigger data: [('remotecontrol_power_control', 'Enabled Power Control'), ('remotecontrol_set_type', 'Set'), ('remotecontrol_active_power', 5000.0), ('remotecontrol_reactive_power', 0.0), ('remotecontrol_duration', 20.0)]
2024-02-17 18:52:05.704 DEBUG (MainThread) [custom_components.solax_modbus] Ready to write multiple registers at 0x7c: [1, 1, 5000, 0, 0, 0, 20]
2024-02-17 18:52:20.038 INFO (MainThread) [custom_components.solax_modbus] read_error : SolaX cannot read holding registers at device 1 position 0x0
NoneType: None
2024-02-17 18:52:20.038 DEBUG (MainThread) [custom_components.solax_modbus.plugin_solax] Evaluated remotecontrol_trigger: corrected/clamped values: [('remotecontrol_power_control', 'Enabled Power Control'), ('remotecontrol_set_type', 'Set'), ('remotecontrol_active_power', 5000.0), ('remotecontrol_reactive_power', 0.0), ('remotecontrol_duration', 20.0)]
2024-02-17 18:52:20.039 DEBUG (MainThread) [custom_components.solax_modbus] ready to repeat button remotecontrol_trigger data: [('remotecontrol_power_control', 'Enabled Power Control'), ('remotecontrol_set_type', 'Set'), ('remotecontrol_active_power', 5000.0), ('remotecontrol_reactive_power', 0.0), ('remotecontrol_duration', 20.0)]
2024-02-17 18:52:20.039 DEBUG (MainThread) [custom_components.solax_modbus] Ready to write multiple registers at 0x7c: [1, 1, 5000, 0, 0, 0, 20]
2024-02-17 18:52:20.063 DEBUG (MainThread) [custom_components.solax_modbus] assuming sleep mode - slowing down by factor 10

wills106, 2024-02-18:
[Remote Control](https://homeassistant-solax-modbus.readthedocs.io/en/latest/modbus-power-control/) was written for a single Inverter as the author of that code only has a single Inverter.

The code would need rewriting for multiple Inverters and have the ability to enter the name of the other Inverter, so it know what other values to look at.

Evarra, 2024-02-18:
Thank you for responding.

Got both inverters woring well if I use direct modbu-write in Node Red:

// Example input data
var inputData = [1, 1, 10500, 0, 0, 0, 20];

msg.payload = {
    'value': inputData,
    'fc': 16, // Function code for Write Multiple Registers
    'unitid': 1, // Slave address
    'address': 124, // Starting address
    'quantity': 7 // Quantity of registers to write
};

return msg;

I really like your integration in home assistant and wouldn't want to redo everything in node red.

I would be willing to test and givce you remote access if necessary.
Unfortunately, I myself am not very good at home assistant programming.

It seems like the home assistant cannot get some parameters from Inverter and thinks that the inverter is in sleep mode.

Evarra, 2024-02-20:
It seems that the remote control actually works and activates as soon as the trigger button is pressed. The problem is instead that the sensor data is updated with a very large delay, It seems as if the code reads them over an unspecified number of seconds.

When reading the modbus directly, the sensor information comes immediately, so it cannot be a problem with the inverter.

What could be the reason for such slow reading? 

Thank you.

Evarra, 2024-02-21:
At the moment the solution works 100% and fast, I just changed the "slowing down by factor 10" value from 10 to 1.

Since it's a Hybrid inverter, they don't sleep and I shouldn't have a problem with it.

`else: 
                _LOGGER.debug(f"assuming sleep mode - slowing down by factor 10")
                self.slowdown = 1
                for i in self.sleepnone: self.data.pop(i, None)
                for i in self.sleepzero: self.data[i] = 0
                # self.data = {} # invalidate data - do we want this ??`

closed 2024-02-21 by Evarra