wills106/homeassistant-solax-modbus #755: [Bug]: Modbus Remotepower Control

manuelmrmorgado, 2024-02-18:
### Describe the bug

Since last week, when triggering the remote control, I get an error and it doesn't work.
I've tried updating to newer versions of this Integration but it doesn't work anyways.

### Integration Version

2024.02.5b8

### Homeassistant core version

2024.2.2

### Inverter brand

SolaX Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A08

### Inverter firmware versions

ARM 1.31 DSP 1.33

### Connection Method

SolaX Pocket Wifi 3.0

### Dongle firmware

10.16

### Detailed Error Log

This error originated from a custom integration.

Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:436
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 4:09:50 PM (1 occurrences)
Last logged: 4:09:50 PM

Something went wrong reading from modbus
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/__init__.py", line 436, in read_modbus_data
    res = self.read_modbus_registers_all()
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 558, in read_modbus_registers_all
    payload = buttondescr.value_function(0, buttondescr, self.data)
              ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 156, in value_function_remotecontrol_recompute
    ('remotecontrol_reactive_power', max(min(reap_up, reactive_power), reap_lo), ),
                                         ^^^^^^^^^^^^^^^^^^^^^^^^^^^^
TypeError: '<' not supported between instances of 'NoneType' and 'int'

wills106, 2024-09-11:
HAs this been resolved in later Integration versions?

manuelmrmorgado, 2024-11-22:
No, still the same behavior.