wills106/homeassistant-solax-modbus #757: [Bug]: battery capacity in %

iobrokerneu, 2024-02-19:
### Describe the bug

hi, would it be possible to create an entity from the battery capacity in %.

I use a Growatt Mod 5000TL3-XH with an ARK HV battery. 

Thanks and greetings Sascha

### Integration Version

2024.01.1

### Homeassistant core version

2024.2.1

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

JDM0DB

### Connection Method

USR TCP232-410S

wills106, 2024-02-19:
I have just released [2024.02.6b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.6b1) with some more battery entities set to Gen4 as the MOD seems to use a mix of Gen2 and Gen4. But it's not documented which are used.

iobrokerneu, 2024-02-19:
I cannot install the new beta version. It's displayed, but you can't install it.

iobrokerneu, 2024-02-19:
[image]

wills106, 2024-02-19:
Can't you select 2024.02.6b1 from that drop down? Or won't it let you?

iobrokerneu, 2024-02-19:
Whenever I select 6b1, it always jumps to 2.5. Could it perhaps be because it is in beta below 2.5? I've already tested it on my iphone and PC.

wills106, 2024-02-19:
If you go to the HACS page, Integrations, click on "SolaX Inverter Modbus" then the 3 dots top right.
Then Redownload. Can you select 2024.02.6b1 that way?

Just updated for me.
[image]

iobrokerneu, 2024-02-19:
Unfortunately does not work . ha was also restarted.

https://github.com/wills106/homeassistant-solax-modbus/assets/55629670/176ba68e-1b03-448b-a1ba-9fca0cb8b111

wills106, 2024-02-20:
Just pushed out 2024.02.6b2 as there was an update for Sofar. Does that show up now?

[image]

iobrokerneu, 2024-02-20:
hi, I have just installed the new beta and the battery display is displayed. Very good work. Thanks and greetings Sascha
[image]

wills106, 2024-02-20:
Do the other Battery sensors work or display correctly?

iobrokerneu, 2024-02-20:
generally all data is displayed correctly. The above data is currently not available. But you set it that way. The process can be closed again immediately. Thanks
[image]

wills106, 2024-02-20:
AC Power to x can be deleted if you click on each one that says Unknown.

Are you able to control the battery from the Integration? Or does that need work still?

iobrokerneu, 2024-02-20:
I don't control the battery directly. I just read the data via modbus. Unfortunately I can't test that.

wills106, 2024-02-20:
I'm going to leave this open for now incase anyone else wants' to feedback on extra battery entities for the MOD.

Emilthn, 2024-03-29:
@wills106 
I've got a Hybrid MOD inverter with a ARX battery and I can see all values but I can't change "Priority" from "Load First", when trying it always jumps back to "Load First". 
So from what I've been reading you must write start and stop time to holding registers 3038 and 3039 because Growatt have chosen to merge mode selection with time slots. Changing only 3144 doesn't work. 

Information found here:
https://community.home-assistant.io/t/esphome-modbus-growatt-shinewifi-s/369171/323

wills106, 2024-04-05:
> Growatt have chosen to merge mode selection with time slots.

It hurts my head just looking at it!
Which bits do you want to control?

Emilthn, 2024-04-06:
Haha yes, it's a "bit" messy..! 
For my MOD 10KTL3-XH changing only 3144 doesn't work but I don't really understand if register 3144 also should be set at the same time as registers 3038 and 3039.. I guess it's best to just do a test and see if it works by just setting 3038/3039.

So he is using these values for setting the different modes.
"Load First" 
Register 3038 = 32768
Register 3039 = 5947

" Battery First"
 Register 3038 = 40960
 Register 3039 = 5947

"Grid First"
 Register 3038 = 49152
 Register 3039 = 5947

Does it make sense for you?

wills106, 2024-04-06:
Ahh that's ok then. I just wanted to make sure you didn't want to adjust times as well, as that would have been a nightmare!

JohNan, 2024-06-10:
This would be great if implemented. The proposed value does however implicitly also set the start and end time to 00:00 - 23:59. That means that it will fail to write if any of the other 6 time slots are enabled and have an overlapping time.

But it a good start and will have the desired effect of choosing mode.

abspoel99, 2024-07-16:
> Haha yes, it's a "bit" messy..! For my MOD 10KTL3-XH changing only 3144 doesn't work but I don't really understand if register 3144 also should be set at the same time as registers 3038 and 3039.. I [...]

Hello, would love to edit the existing code to make this work on my inverter. But as im not familiar with coding / yaml I’m struggling with making the correct changes to test this.

Would any of you be so kind to give me some pointers 
Do i need to add a extra entry or modify entry 3144?

TonyMathiesen, 2024-10-21:
With version 2024.10.4 it is now supported for TL3-XH to set the time slot 1, 2 & 3 for priority settings. See comment https://github.com/wills106/homeassistant-solax-modbus/issues/940#issuecomment-2426266714 for more details.

flopp999, 2025-04-02:
@iobrokerneu still problem?

iobrokerneu, 2025-04-02:
@flopp999 no

flopp999, 2025-04-02:
> @flopp999 no

Great. You may then close this issue.

closed 2025-04-11 by iobrokerneu