wills106/homeassistant-solax-modbus #760: [Bug]: GROWATT SPF5000es no enteties no data

marbeja, 2024-02-20:
### Describe the bug

I try to connect my Nuc based HA via this Solax modbus but can't get anything out of it.
The serial to Ethernet converter blinks Rx and tx led, IP address correct, tried Port 23 25 26 502 506 but never got anything, tried TCP server modbus  also with RTU and without, tried modbus address 1 2 3 4 30001 30.000
always the same, no data. Will try to upload the log files here. 
I also tried two USB to Rs485 sticks no data, but the wires are having the typical 0.5V from Rs485 I worked before (orange pair)
Logfile says unable to read serial number %9

### Integration Version

2024 02 05

### Homeassistant core version

114 c01 05

### Inverter brand

GROWATT spf5000es

### Plugin used

plugin_growatt.py

### Serial prefix

JZM2FAD

### Connection Method

Use iot just tcp232-410S

### Detailed Error Log

Will add from pc

marbeja, 2024-02-20:
Core 2024.2.1
Supervisor 2024.01.1
Operating System 11.5
Frontend 20240207.1

marbeja, 2024-02-20:
[home-assistant_solax_modbus_2024-02-20T09-54-30.592Z.log](https://github.com/wills106/homeassistant-solax-modbus/files/14342583/home-assistant_solax_modbus_2024-02-20T09-54-30.592Z.log)

marbeja, 2024-02-20:
sorry as newbie I cannot edit the issue with adding the error log, it is too big!

marbeja, 2024-02-20:
the inverter and  is set to 9600bd corretctly, no idea where these 19200 baud come from!
Sorry my logfile is 4,5 MB but there is no possibility to delete it in HA under logs

marbeja, 2024-02-20:
what does SPF -WID mean ?  I know that there are several SPF Inverters but never heard of  WID, google neither :(  

Growatt:

    AC Battery Storage:
        SPA

    Hybrid:
        SPF - WIP
        SPH
        TL-XH (MIN & MOD)

    PV Only:
        MAC
        MAX
        MID
        TL-X

wills106, 2024-02-20:
WIP = Work In Progress You can follow the progress #686 

As you are using an RS485-USB Adaptor you need to select `Serial` on the first window (1st Image) and then it will ask you for the `Serial port name` on the next page (3rd Image)
https://homeassistant-solax-modbus.readthedocs.io/en/latest/installation/#pair-your-inverter

marbeja, 2024-02-20:
thanks for the speedy reply, I tried **both, the USB and the  RS485 to Ethernet**  but both don't work!

marbeja, 2024-02-20:
[image]

I tried modbus adress 1 to 3 and without parallel even I have 2 inverters 
[image]

do I have to mark `Modbus  RTU over  TOP here for the rather old SPF ??

wills106, 2024-02-20:
EPS, Dry Contact and Parallel Mode wont do anything on your Inverter.

Can't you see the modbus address on the Inverter LCD?

I would put Misc Config back to default. Or factory reset the URS.

Also it looks like you are pointing the Integration at 192.168.1.7 but you are configuring the URS on 192.168.1.10 though.

Also are you connecting A to A and B to B between the URS and Inverter? They don't go A-B and B-A like on a RS232 setup.