wills106/homeassistant-solax-modbus #770: [Bug]: after the update to 2024.02.6, the sensors are unavailable

matomatusov, 2024-02-23:
### Describe the bug

Hi,
after the update to 2024.02.6, the sensors are unavailable.

[image]

### Integration Version

latest always

### Homeassistant core version

latest always

### Inverter brand

Solax Power

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

ARM1.35 DSP1.34

### Connection Method

Waveshare adapter

### Dongle firmware

02.09

wills106, 2024-02-26:
Does [2024.02.7b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.7b1) help with the issue?

matomatusov, 2024-02-26:
@wills106  - I updated to 2024.02.7b1 and it helped.

Thanks for help.

matomatusov, 2024-02-26:
@wills106 - it didn't help. I had to go back on the version 2024.02.5.

matomatusov, 2024-02-26:
@jikuci  but after reloading the sensors were unavailable.

jikuci, 2024-02-26:
@matomatusov  After rebor or reload, sensors are avaliable.

wills106, 2024-02-27:
Just released [2024.02.7b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.7b2) which waits for HA to fully load before establishing the Modbus connection.

matomatusov, 2024-02-27:
@wills106  - [2024.02.7b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.7b2) entities were loaded but Modbus communication stopped. After reloading, the entities were unavailable. I had to go back on the version 2024.02.5.

wuschi1983, 2024-02-28:
Same problem. 2024.02.7b2 solved the problem not really. 
aftet around 2-3 minutes the values freeze and the entitys do not update.

james474789, 2024-02-29:
> Just released [2024.02.7b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.7b2) which waits for HA to fully load before establishing the Modbus connection.

When I updated to 2.6 my sensors all went unavailable; HA restart or integration reload did not help. I resolved this by deleting the hub within the integration and recreating (with identical settings). However, this left me with the issue that the sensors would go unavailable every time HA restarted. At this point reloading got it working again.

Updating to 7b2 appears to have resolved the second problem based on a few restarts I tried this morning.

wills106, 2024-03-05:
Just released [2024.03.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.1) which allows the Integration to start reading from the Inverter without the need to restart HA again. But it waits till HA has fully loaded before starting the connection.

matomatusov, 2024-03-05:
@wills106  After 30 minutes, communication stopped.

wills106, 2024-03-06:
What errors do you get when it stops?
https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#detailed-error-log

matomatusov, 2024-03-06:
@wills106 
I am sending a log:

[homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant

2024-03-06 17:35:55.417 INFO (MainThread) [homeassistant.setup] Setting up solax_modbus
2024-03-06 17:35:55.417 INFO (MainThread) [homeassistant.setup] Setup of domain solax_modbus took 0.0 seconds
2024-03-06 17:35:56.150 INFO (MainThread) [custom_components.solax_modbus.config_flow] detected HA core version 2024 2
2024-03-06 17:35:56.150 INFO (MainThread) [custom_components.solax_modbus.config_flow] starting configflow - domain = solax_modbus
2024-03-06 17:35:56.150 INFO (MainThread) [custom_components.solax_modbus] trying to load plugin - plugin_name: solax
2024-03-06 17:36:25.297 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX: trying to determine inverter type
2024-03-06 17:36:25.390 INFO (SyncWorker_9) [homeassistant.loader] Loaded esphome from homeassistant.components.esphome
2024-03-06 17:36:25.470 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] Read SolaX 0x0 serial number before potential swap: H34A10I5402094
2024-03-06 17:36:25.507 INFO (MainThread) [homeassistant.components.button] Setting up solax_modbus.button
2024-03-06 17:36:25.513 INFO (MainThread) [homeassistant.components.number] Setting up solax_modbus.number
2024-03-06 17:36:25.518 INFO (MainThread) [homeassistant.components.select] Setting up solax_modbus.select
2024-03-06 17:36:25.532 INFO (MainThread) [homeassistant.components.sensor] Setting up solax_modbus.sensor
2024-03-06 17:36:25.543 INFO (MainThread) [custom_components.solax_modbus.button] hub.wakeuButton: SolaxModbusButtonEntityDescription(key='battery_awaken', device_class=None, entity_category=None, entity_registry_enabled_default=False, entity_registry_visible_default=True, force_update=False, icon='mdi:battery-alert-variant', has_entity_name=False, name='Battery Awaken', translation_key=None, translation_placeholders=None, unit_of_measurement=None, allowedtypes=6144, register=86, command=1, blacklist=None, write_method=1, value_function=None, autorepeat=None)
2024-03-06 17:36:25.548 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x0 seriesnumber to block with start 0x0
2024-03-06 17:36:25.548 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x1d safety_code to block with start 0x0
2024-03-06 17:36:25.549 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x1e matebox_enabled to block with start 0x0
2024-03-06 17:36:25.549 INFO (MainThread) [custom_components.solax_modbus.sensor] Starting new block at 0x7d
2024-03-06 17:36:25.549 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x7d firmware_DSP_minor_version to block with start 0x7d
2024-03-06 17:36:25.549 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x7e firmware_DSP_hardware_version to block with start 0x7d
2024-03-06 17:36:25.549 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x7f firmware_DSP_major_version to block with start 0x7d
2024-03-06 17:36:25.549 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x80 firmware_ARM_major_version to block with start 0x7d
2024-03-06 17:36:25.549 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x81 firmwareversion_modbustcp_major to block with start 0x7d
2024-03-06 17:36:25.549 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x82 firmwareversion_modbustcp_minor to block with start 0x7d
2024-03-06 17:36:25.549 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x83 firmware_ARM_minor_version to block with start 0x7d
2024-03-06 17:36:25.549 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x84 bootloader_version to block with start 0x7d
2024-03-06 17:36:25.549 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x85 rtc to block with start 0x7d
2024-03-06 17:36:25.549 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x8b charger_use_mode to block with start 0x7d
2024-03-06 17:36:25.549 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x8c manual_mode to block with start 0x7d
2024-03-06 17:36:25.549 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x8d battery_type to block with start 0x7d
2024-03-06 17:36:25.549 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x8e battery_charge_float_voltage to block with start 0x7d
2024-03-06 17:36:25.549 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x8f battery_discharge_cut_off_voltage to block with start 0x7d
2024-03-06 17:36:25.549 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x90 battery_charge_max_current to block with start 0x7d
2024-03-06 17:36:25.549 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x91 battery_discharge_max_current to block with start 0x7d
2024-03-06 17:36:25.549 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x94 selfuse_nightcharge_upper_soc to block with start 0x7d
2024-03-06 17:36:25.549 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x97 charger_start_time_1 to block with start 0x7d
[... cut ...]
2024-03-06 17:36:30.373 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] local data update callback for entity: remotecontrol_import_limit new limit: 15000
2024-03-06 17:36:30.373 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] local data update callback for entity: export_control_user_limit new limit: 15000
2024-03-06 17:36:30.373 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] local data update callback for entity: generator_max_charge new limit: 15000

wills106, 2024-03-30:
If the data is still freezing try [2024.03.8b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.8b1)

matomatusov, 2024-04-02:
@wills106 
Is it necessary to remove the older version of the integration and restart HA?
Then download the new integration and restart HA again.

wills106, 2024-04-02:
You can remove the Integration from the Integration page. Download new version and restart and add it again. You don't need the double restart.

With some code changes you can just download the new version and restart. I don't know at which point you need to delete and readd.

jakobschou, 2024-04-15:
I have the same issue, after updating my HA to 2024.4.3 and HA OS to 12.2, the SolaX integration went unavailable for my Growatt SPH3600 inverter (worked until the update this morning with 9600 baud rate, 10 second interval, and serial to USB).

At first I updated the integration to 2024.04.3 without any luck.

Afterwards I have tried to remove the integration completely and restarting, and adding it again - but it does not generate any entities....

wills106, 2024-08-03:
@matomatusov can this be closed down now?

matomatusov, 2024-08-13:
@wills106 ,
yes it can be closed.

closed 2024-08-13 by matomatusov