wills106/homeassistant-solax-modbus #771: [Bug]: Solis - No more update available

nitronix17, 2024-02-23:
### Describe the bug

hi,
After setting the charge times and pressing update I receive the error "No Response received after 3 retries.
[image]

This does not happen in version 2024.02.05

### Integration Version

2024.02.06

### Homeassistant core version

2024.2.2

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### Serial prefix

???

### Connection Method

RS485-USB

burden42, 2024-02-24:
yes solis user here too updated to 2024.02.6  and had zero connection to my inverter , no reading of any data . reverted back to 2024.02.5  and all came good again using modbus to tcp hardware as described here 
https://github.com/Jumpy07/Solis---SolisCloud-and-Home-Assistant?fbclid=IwAR1acuEvWkuDfk8j-1oICUASdsy4kTpmIFcNezVlOoo62F_Ehpm4CXqgJlc

wills106, 2024-02-26:
Does [2024.02.7b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.7b1) help with the issue?

nitronix17, 2024-03-01:
Here, the issue is not resolved
[image]

wills106, 2024-03-04:
Once the general issues with 2024.2.x are sorted we can look into specific issues.
Hopefully get the connectivity sorted in the next day or two.

wills106, 2024-03-05:
Just released [2024.03.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.1) are you still getting the errors after updating?

luftdieb, 2024-03-17:
Hi, I'm using 2024.03.3 and have just for one function this kind of errormessage. That means, all read parameter works great. But when I'm trying to change the Power Switch from Off to On, I'm getting this error. From On to Off, it works. 

This problem is somehow news since ~3-6 weeks. 
I do also have a nodered flow, which try to switch between ON and OFF and also here, I'm getting also error message on same direction (OFF --> ON):
` "[error] [api-call-service:PowerSwitch on] Call-service error. AsyncModbusTcpClient.close() got an unexpected keyword argument 'intern'" `

I've also  followed up the recommendation from previous changelog comment to uninstall the Solax instance from HA and also from HACS, but this didn't change anything.

With Debug active, I'm getting this exception

`2024-03-17 18:40:31.148 INFO (MainThread) [custom_components.solax_modbus.select] writing solis_modbus_ select register 43007 value 190
2024-03-17 18:40:51.664 INFO (MainThread) [custom_components.solax_modbus] exception Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.3.238:8899] : solis_modbus_ cannot read holding registers at device 1 position 0xa7ff
NoneType: None
2024-03-17 18:40:51.664 DEBUG (MainThread) [custom_components.solax_modbus] assuming sleep mode - slowing down by factor 10
2024-03-17 18:40:54.244 INFO (MainThread) [custom_components.solax_modbus.select] writing solis_modbus_ select register 43007 value 190
2024-03-17 18:41:14.252 INFO (MainThread) [custom_components.solax_modbus] exception Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.3.238:8899] : solis_modbus_ cannot read holding registers at device 1 position 0xa7ff
NoneType: None
2024-03-17 18:41:14.252 DEBUG (MainThread) [custom_components.solax_modbus] assuming sleep mode - slowing down by factor 10
2024-03-17 18:41:14.253 DEBUG (MainThread) [custom_components.solax_modbus] assuming sleep mode - slowing down by factor 10
2024-03-17 18:41:21.385 INFO (MainThread) [custom_components.solax_modbus.select] writing solis_modbus_ select register 43007 value 190`

Hope that helps to identify the problem.

wills106, 2024-03-17:
There is mention of this `'intern'` In the release notes for pyModbus 3.6.6 https://github.com/pymodbus-dev/pymodbus/pull/2080 but I'm not sure if I need to declare something extra?
I need to read into this.

luftdieb, 2024-04-14:
It seems, new version create different problem now. I'm using V2024.04.3 with latest HA version.
Now, NodeRed report "[error] [api-call-service:PowerSwitch on] Call-service error. Modbus Error: [Input/Output] ERROR: No response received after 3 retries".
If I enable debug information, I'm getting this output:
`2024-04-14 20:16:30.544 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2024-04-14 20:16:30.544 DEBUG (MainThread) [custom_components.solax_modbus] Trying to connect to Inverter at 192.168.3.238:8899
2024-04-14 20:16:30.548 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 192.168.3.238:8899
2024-04-14 20:16:33.439 INFO (MainThread) [custom_components.solax_modbus.select] writing solis_modbus_ select register 43007 value 190
2024-04-14 20:16:54.493 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2024-04-14 20:16:54.493 DEBUG (MainThread) [custom_components.solax_modbus] Trying to connect to Inverter at 192.168.3.238:8899
2024-04-14 20:16:54.497 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 192.168.3.238:8899
2024-04-14 20:17:06.249 INFO (MainThread) [custom_components.solax_modbus.select] writing solis_modbus_ select register 43007 value 222
2024-04-14 20:17:28.481 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2024-04-14 20:17:28.481 DEBUG (MainThread) [custom_components.solax_modbus] Trying to connect to Inverter at 192.168.3.238:8899
2024-04-14 20:17:28.483 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 192.168.3.238:8899
`
Now I have the situation, I can't revert back to 2024.02.05, because HACS only allow to select last 5 versions, and so, oldest version I can select is 2024.03.7, which have also a problem with controlling PowerSwitch function :-( .

What can I do ?

wills106, 2024-04-14:
[image]

wills106, 2024-04-14:
Are you trying to talk to the Inverter with this Integration and NodeRed at the same time?

luftdieb, 2024-04-14:
> Are you trying to talk to the Inverter with this Integration and NodeRed at the same time?

This could be the cause, which does not seem to be a problem in the older version. The interval in HA is set to 15 seconds and in NodeRed there is a cyclic check every 5 minutes and to test a manual trigger. If HA and NodeRed clatter together, this can cause simultaneous access to the inverter.

wills106, 2024-04-14:
Modbus is only designed to have a single Mater and multiple slaves.

You have introduced two masters, so you have introduced collisions etc and you are trying to use Modbus in a way it's not designed to be used.

wills106, 2024-04-14:
The Solis Inverters aren't exactly Modbus compliant to begin with!

luftdieb, 2024-04-14:
I did not give you full explaination about my NodeRed setup. It's not using a parallel Modbus communication to inverter.
Instead I'm changing Solax entity, which perform the action over the existing solax-inverter modbus channel.

wills106, 2024-05-18:
Is this issue still relevant or has it been resolved now?

luftdieb, 2024-05-18:
Unfortunatly my NodeRed Flow, where I try to control the HomeAssistant Entitdy still report error. 
Here are some outputs of NodeRed: `sent data : msg :
object
domain: "select"
service: "select_last"
data: object
entity_id: "select.solis_modbus_power_switch"`
`msg : string[92]
"Call-service error. Modbus Error: [Input/Output] ERROR: No response received after 3 retries"`

After enabling debugging I'm getting this output with trigger from NodeRed flow or also same via  HA control instance 
[image]

`2024-05-18 18:28:36.643 INFO (MainThread) [custom_components.solax_modbus.select] writing solis_modbus_ select register 43007 value 190
2024-05-18 18:28:56.651 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2024-05-18 18:28:56.651 DEBUG (MainThread) [custom_components.solax_modbus] Trying to connect to Inverter at 192.168.3.238:8899
2024-05-18 18:28:56.659 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 192.168.3.238:8899
2024-05-18 18:28:57.600 INFO (MainThread) [custom_components.solax_modbus.select] writing solis_modbus_ select register 43007 value 222
2024-05-18 18:29:17.788 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2024-05-18 18:29:17.788 DEBUG (MainThread) [custom_components.solax_modbus] Trying to connect to Inverter at 192.168.3.238:8899
2024-05-18 18:29:17.791 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 192.168.3.238:8899`

That means, I'm really interessted finding a solution :-)

luftdieb, 2025-01-19:
Hi, I need to bring this problem again on the table.
Using this instance "solis_modbus_power_switch" in HA, it still not work reliable, which mean if you have this in automation 1x per week, the switch is not being applied.
When I use this instance manually, it is reporting for the 2nd switch action after restart of HA  this message 

[image]

After enable debug log information, I could see this message:
Logger: homeassistant
Quelle: /usr/src/homeassistant/homeassistant/runner.py:112
Erstmals aufgetreten: 18:17:18 (15 Vorkommnisse)
Zuletzt protokolliert: 18:20:21

Error doing job: Fatal error: protocol.data_received() call failed. (None)
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
    self._protocol.data_received(data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 304, in data_received
    self.datagram_received(data, None)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 178, in callback_data
    used_len, pdu = self.framer.processIncomingFrame(self.trace_packet(False, data))
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/framer/base.py", line 76, in processIncomingFrame
    data_len, pdu = self._processIncomingFrame(data[used_len:])
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/framer/base.py", line 97, in _processIncomingFrame
    if (result := self.decoder.decode(frame_data)) is None:
                  ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/pdu/decoders.py", line 112, in decode
    pdu.decode(frame[1:])
    ~~~~~~~~~~^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/pdu/register_message.py", line 182, in decode
    self.address, register = struct.unpack(">HH", data)
                             ~~~~~~~~~~~~~^^^^^^^^^^^^^
struct.error: unpack requires a buffer of 4 bytes

I can also offer Wireshark stream for further troubleshooting... 

If you tell me perform this and this action on terminal or if you have a test script, which I can perform to give you further details, I'm happy to help out :-)

luftdieb, 2025-01-20:
Hi @wills106 , I've extended the python file __init__.py in line 567 with _LOGGER.warning("address: " + str(address) + " payload: " + str(payload[0]) + f" pyModbus: {pymodbus.__version__}" )

For first execution of changing powermode I'm getting "address: 43007 payload: 190 pyModbus: 3.7.4"

For 2nd execution I'm getting above error message.
For 3rd  execution I'm getting "address: 43007 payload: 222 pyModbus: 3.7.4"

Is the pyModbus version ok ?
How to dig deeper in pymodbus python files? Where are there located in HA filesystem? The path which is mentioned in logfile, I can't find. I do have /usr/local/lib... But no subfolders ./python3.13/site-package....

luftdieb, 2025-01-20:
I''m somehow confused. Now after restarting HA again, it reports pyModbus 3.8.3.  But problem still remain:
Traceback (most recent call last):
  File "/usr/local/lib/python3.13/asyncio/selector_events.py", line 1019, in _read_ready__data_received
    self._protocol.data_received(data)
    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 304, in data_received
    self.datagram_received(data, None)
    ~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transport/transport.py", line 338, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
  File "/usr/local/lib/python3.13/site-packages/pymodbus/transaction/transaction.py", line 178, in callback_data
    used_len, pdu = self.framer.processIncomingFrame(self.trace_packet(False, data))
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/framer/base.py", line 76, in processIncomingFrame
    data_len, pdu = self._processIncomingFrame(data[used_len:])
                    ~~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/framer/base.py", line 97, in _processIncomingFrame
    if (result := self.decoder.decode(frame_data)) is None:
                  ~~~~~~~~~~~~~~~~~~~^^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/pdu/decoders.py", line 112, in decode
    pdu.decode(frame[1:])
    ~~~~~~~~~~^^^^^^^^^^^
  File "/usr/local/lib/python3.13/site-packages/pymodbus/pdu/register_message.py", line 182, in decode
    self.address, register = struct.unpack(">HH", data)
                             ~~~~~~~~~~~~~^^^^^^^^^^^^^
struct.error: unpack requires a buffer of 4 bytes

luftdieb, 2025-01-20:
Now after additional restart of HA, I'm having pyModbus version 3.7.4 again. How is that possible?
I've also add in __init__.py the output of "resp" status, which is " Exception Response(134, 6, None)".
But I'm afraid this will not help.

closed 2025-10-03 by github-actions[bot]