wills106/homeassistant-solax-modbus #773: [Bug]: All entities and sensors disabled after update to 2024.02.6

md091979, 2024-02-24:
### Describe the bug

All entities and sensors disabled after update to 2024.02.6 , they are not more available

### Integration Version

2024.02.6

### Homeassistant core version

20242..3

### Inverter brand

Solax X3 Hybrid

### Plugin used

plugin_solax.py

### Serial prefix

H34A08

### Inverter firmware versions

-

### Connection Method

-

### Dongle firmware

-

### Detailed Error Log

-

### Additional context

-

spetrie13, 2024-02-24:
Same for me with Solis inverter

Alfred99-HA, 2024-02-24:
Same for me.
It seems that also Solax Cloud doesn't show values as a consequence

the-sim87, 2024-02-24:
Same for me for Growatt inverter - reload of solax-modbus plugin helped for now.

mc4wrk, 2024-02-24:
I have the same problem.
I had a integration for my X1-Hybrid-G4 using modbus and after update HA, all my sensors are gone and loose the connection to the USB dongle. I have made a rollback using the backup files and now I can't get any info from the integration. Any advice will be most welcome. Thanks

closed 2024-02-26 by wills106