wills106/homeassistant-solax-modbus #774: [Bug]: 'Sofar' - 'HYD 10KTL-3PH' Corrupt Mobus Data Gets Accepted

fische11, 2024-02-24:
### Describe the bug

It seems that corrupt mobus data gets accepted from the LSE-3 Logger. Yesterday at 2am the all the data in the following category where completely random for one response: Electric Power (0x0680-0x06BF)

I might found a possible solution on how to check the modbus data for errors as it seems that the LSE-3 does not check the CRC itself: 
https://github.com/StephanJoubert/home_assistant_solarman/issues/120#issuecomment-1282318815

### Integration Version

2024.02.6

### Homeassistant core version

2024.2.3

### Inverter brand

Sofar

### Plugin used

plugin_sofar.py

### Serial prefix

SP2ES1

### Connection Method

LSE-3

hruskaexe, 2024-03-04:
Hello, I have same Sofar device and after update to this version 2024.02.6, communication always drops out after about an hour. Please fix it, I'm losing my data.

hruskaexe, 2024-03-05:
UPDATE:

Log of errors in previous message are before connection stack. After is in log this:

-----------------------------------------------------------------------------------------------------
Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:240
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 05:00:21 (7 occurrences)
Last logged: 08:01:18

[281473607444416] Modbus Error: [Input/Output] ERROR: No response received after 3 retries
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 240, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2279, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2316, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 892, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 962, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 195, in async_handle_select_option
    await self.async_select_option(option)
  File "/config/custom_components/solax_modbus/select.py", line 106, in async_select_option
    await self._hub.async_write_registers_single(unit=self._modbus_addr, address=self._register, payload=payload)
  File "/config/custom_components/solax_modbus/__init__.py", line 425, in async_write_registers_single
    return await self._client.write_registers(address, payload, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 175, in async_execute
    raise ModbusIOException(
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: No response received after 3 retries

-------------------------------------------------------------------------------------------------------------------------------------------
Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:240
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 05:40:33 (13 occurrences)
Last logged: 08:11:25

[281473607444416] Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.0.205:8899]
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 240, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2279, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2316, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 892, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 962, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 195, in async_handle_select_option
    await self.async_select_option(option)
  File "/config/custom_components/solax_modbus/select.py", line 106, in async_select_option
    await self._hub.async_write_registers_single(unit=self._modbus_addr, address=self._register, payload=payload)
  File "/config/custom_components/solax_modbus/__init__.py", line 425, in async_write_registers_single
    return await self._client.write_registers(address, payload, **kwargs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/mixin.py", line 426, in write_registers
    return self.execute(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 147, in execute
    raise ConnectionException(f"Not connected[{self!s}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.0.205:8899]

wills106, 2024-03-05:
Is this still with 2024.02.6 or 2024.03.1?

hruskaexe, 2024-03-05:
Now, I updated to 2024.03.1 and communication worked longer then hour, but the same result.

BEFORE:
-----------------------------------------------------------------------------------------------------------------------
Logger: custom_components.solax_modbus
Source: custom_components/solax_modbus/__init__.py:507
Integration: SolaX Inverter Modbus (documentation, issues)
First occurred: 09:35:07 (277 occurrences)
Last logged: 11:47:39

sofar: read failed at 0x41a: heatsink_temperature_1
sofar: read failed at 0x41b: heatsink_temperature_2
sofar: read failed at 0x420: module_temperature_1
sofar: read failed at 0x421: module_temperature_2
sofar: read failed at 0x42c: rtc

-----------------------------------------------------------------------------------------------------------------------
AFTER:
-----------------------------------------------------------------------------------------------------------------------
Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:240
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 12:00:19 (3 occurrences)
Last logged: 12:20:19

[281472915908544] Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.0.205:8899]
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 240, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2279, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2316, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 892, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 962, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 195, in async_handle_select_option
    await self.async_select_option(option)
  File "/config/custom_components/solax_modbus/select.py", line 106, in async_select_option
    await self._hub.async_write_registers_single(unit=self._modbus_addr, address=self._register, payload=payload)
  File "/config/custom_components/solax_modbus/__init__.py", line 428, in async_write_registers_single
    return await self._client.write_registers(address, payload, **kwargs)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/mixin.py", line 426, in write_registers
    return self.execute(
           ^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 147, in execute
    raise ConnectionException(f"Not connected[{self!s}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.0.205:8899]

-----------------------------------------------------------------------------------------------------------------------
Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:240
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 12:30:50 (4 occurrences)
Last logged: 13:01:05

[281472915908544] Modbus Error: [Input/Output] ERROR: No response received after 3 retries
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 240, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2279, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2316, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 892, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 962, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 195, in async_handle_select_option
    await self.async_select_option(option)
  File "/config/custom_components/solax_modbus/select.py", line 106, in async_select_option
    await self._hub.async_write_registers_single(unit=self._modbus_addr, address=self._register, payload=payload)
  File "/config/custom_components/solax_modbus/__init__.py", line 428, in async_write_registers_single
    return await self._client.write_registers(address, payload, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 175, in async_execute
    raise ModbusIOException(
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: No response received after 3 retries

wills106, 2024-03-06:
Do you get this error when you are trying to write to the Inverter then?

hruskaexe, 2024-03-06:
When I restart HASS, before connection error, I can change parameters. I try it with "Energy Storage Mode", but I see, that duration between send start and send done confirmation is about 30 seconds. When a disconnect communication error occurs, I can't change parameters and in log is this:

-----------------------------------------------------------------------------------------------------------------
Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:240
Integration: Home Assistant WebSocket API (documentation, issues)
First occurred: 13:22:38 (2 occurrences)
Last logged: 13:24:23

[281472890835136] Modbus Error: [Input/Output] ERROR: No response received after 3 retries
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 240, in handle_call_service
    response = await hass.services.async_call(
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2279, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2316, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 892, in entity_service_call
    single_response = await _handle_entity_call(
                      ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 962, in _handle_entity_call
    result = await task
             ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 195, in async_handle_select_option
    await self.async_select_option(option)
  File "/config/custom_components/solax_modbus/select.py", line 106, in async_select_option
    await self._hub.async_write_registers_single(unit=self._modbus_addr, address=self._register, payload=payload)
  File "/config/custom_components/solax_modbus/__init__.py", line 428, in async_write_registers_single
    return await self._client.write_registers(address, payload, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 175, in async_execute
    raise ModbusIOException(
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] ERROR: No response received after 3 retries

cschlipf, 2024-03-10:
@hruskaexe Are you also using the LSE-3 logger or have you connected to the RS485 port? In the later, did you properly terminate the bus? See also #644, which seems to be a duplicate of your issue. However your issue does not seem to be related to what @fische11 reports here.

fische11, 2024-03-10:
@cschlipf Yes, but for me it was just not that often. I already posted a potential fix in my first post. As this issue does not seem to be a priority I switched to this RS485 Adapter: https://www.waveshare.com/wiki/2-CH_RS485_TO_ETH_(B)
So far the data is always correct with the modbus adapter but the connection gets lost sometimes and you have to restart the inverter. I think the baudrate was set to high. Now with a lower baudrate it seems to be stable.

cschlipf, 2024-04-03:
@wills106 

What do you think about a simple sanitizer?

- For percentages do not allow values of more than 150% (e.g. my battery got a SoC of more than 6000% reported lately)
- For powers do not allow values of more than a configurable amount of power. (e.g. with my 9.84 kWp everything above 20 kW) is pretty unrealistic for all powers.

This is not a 100% fix of course, but in my case this would filter out about 99% of false values.

cschlipf, 2024-04-03:
I know that I need to use V110051. I have it running since 3 months now.

However I would like to avoid having to buy new hardware - especially the LSW-3 is a standard Sofar hardware and is used by many people today.

hruskaexe, 2024-04-25:
> @hruskaexe Are you also using the LSE-3 logger or have you connected to the RS485 port? In the later, did you properly terminate the bus? See also #644, which seems to be a duplicate of your issue. However your issue does not seem to be related to what @fische11 reports here.

Hello, thanks for your reply, I want to write you result of my problem. I have RS485 port.

I tried some wiring changes. With both resistors (terminators) connected it produced the same errors. So I changed the device to Waveshare (in link) and I connected the common ground (GND) (Sofar with Waveshare) and also I grounded the frame of the device (PE) together with the metal sheath of the cable. Now it seems to be working good and stable.

Waveshare: https://www.waveshare.com/product/iot-communication/wired-comm-converter/ethernet-to-uart-rs232-rs485/rs232-485-to-wifi-eth-b.htm

cschlipf, 2025-05-03:
As you are also using the LSE-3, please test whether it works better in transparency mode. It does work very well for me. Please see the updated documentation here: https://homeassistant-solax-modbus.readthedocs.io/en/latest/sofar-installation/#configuration

wills106, 2025-08-28:
Closing down as no response.