wills106/homeassistant-solax-modbus #780: [Bug]: 2024.02.7b2 fails to load on Gen3 SolaX

wills106, 2024-02-27:
### Describe the bug

Integration fails to look for a Serial Number from a fresh setup.

### Integration Version

2024.02.7b2

### Homeassistant core version

2024.2.2

### Inverter brand

SolaX

### Plugin used

plugin_solax.py

### Serial prefix

H1E5

### Inverter firmware versions

ARM 3.28, DSP 3.32

### Connection Method

Gen3 Ethernet

### Dongle firmware

N/A

### Detailed Error Log

2024-02-27 18:38:44.208 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-02-27 18:39:21.306 WARNING (MainThread) [homeassistant.components.homeassistant.triggers.numeric_state] Error initializing 'Saving Sessions Min SOC' trigger: In 'numeric_state' condition: unknown entity sensor.solax_battery_capacity
2024-02-27 18:39:31.613 INFO (MainThread) [custom_components.solax_modbus.config_flow] detected HA core version 2024 2
2024-02-27 18:39:31.613 INFO (MainThread) [custom_components.solax_modbus.config_flow] starting configflow - domain = solax_modbus
2024-02-27 18:39:35.017 INFO (MainThread) [custom_components.solax_modbus.config_flow] validating base: {'name': 'SolaX', 'interface': 'tcp', 'read_modbus_addr': 1, 'plugin': 'solax', 'scan_interval': 2, 'read_eps': False, 'read_dcb': False, 'read_pm': False}
2024-02-27 18:39:35.017 INFO (MainThread) [custom_components.solax_modbus.config_flow] validating base config for SolaX: pre: {'name': 'SolaX', 'interface': 'tcp', 'read_modbus_addr': 1, 'plugin': 'solax', 'scan_interval': 2, 'read_eps': False, 'read_dcb': False, 'read_pm': False}
2024-02-27 18:39:40.128 INFO (MainThread) [custom_components.solax_modbus.config_flow] validating host: returning data: {'host': '192.168...', 'port': 502, 'tcp_type': 'tcp'}
2024-02-27 18:39:40.128 INFO (MainThread) [custom_components.solax_modbus.config_flow] title configflow solax_modbus name: {'name': 'SolaX', 'interface': 'tcp', 'read_modbus_addr': 1, 'plugin': 'solax', 'scan_interval': 2, 'read_eps': False, 'read_dcb': False, 'read_pm': False, 'host': '192.168.1.250', 'port': 502, 'tcp_type': 'tcp'}
2024-02-27 18:39:40.128 INFO (MainThread) [homeassistant.setup] Setting up solax_modbus
2024-02-27 18:39:40.128 INFO (MainThread) [homeassistant.setup] Setup of domain solax_modbus took 0.0 seconds
2024-02-27 18:39:40.129 INFO (MainThread) [custom_components.solax_modbus] trying to load plugin - plugin_name: solax
2024-02-27 18:40:23.724 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry SolaX for button
2024-02-27 18:40:23.726 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry SolaX for number
2024-02-27 18:40:23.727 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry SolaX for select
2024-02-27 18:40:23.728 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry SolaX for sensor

lukemob, 2024-02-28:
Hello,

same issue here. Inverter Solax Hybrid X3 G4.

Lukas

mrpohoda, 2024-02-28:
I have the same issue with Gen3 SolaX for about 1 week. I always install the most recent version this plugin. Whenever there is a new update of HA (System, Core, other plugin, ...) and HA restarts, SolaX entities become unavailable. I have to restart HA again, then it becomes available again. Also SolaX Lock State changes to Locked and I have to change the select box to Unlocked (I've already made an automation for that).

boarderhype, 2024-03-02:
> Hello,
> 
> same issue here. Inverter Solax Hybrid X3 G4.
> 
> Lukas

Hi,

I do have exactly the same issue and setup. Looking forward to a fix.

Frank

wills106, 2024-03-04:
@wrfz with the changes done in #778 Does the Integration start reading from the Inverter as soon as you add it. Or do you also have to restart HA for the Integration to start reading.

At least on SolaX Inverters when you initially add the Inverter, it hangs at "trying to load plugin"

> 2024-02-27 18:39:40.129 INFO (MainThread) [custom_components.solax_modbus] trying to load plugin - plugin_name: solax

You do get:
[image]
But you don't get the option to assign the device to an area.

If you look in the Hub it lists no devices or entities, as it's not reading the Serial Number.
[image]

If you restart HA it will start reading from the Inverter.

I think this is all related to https://github.com/wills106/homeassistant-solax-modbus/blob/01b81467940074f040175d92992c733c67b2670d/custom_components/solax_modbus/__init__.py#L148

I am guessing it only get triggered once at start-up.
If HA is already started when you initially add the Inverter (which is the only way you can add it) the Integration wont start reading from the Inverter.

I can't really find much documentation on the event handler other than the brief bit mentioned [here](https://developers.home-assistant.io/docs/integration_listen_events#listening-to-the-event-bus-directly) are you aware of any other methods to ensure HA is already started, other than checking on startup?

wrfz, 2024-03-04:
I don't think I have reinstalled Solax since then. 
I had always restarted HA during the tests and checked the start and did not check if I could add Solax without restarting HA.

As you describe, it cannot work at all without restarting HA, 
because the EVENT_HOMEASSISTANT_STARTED event only occurs at startup.
Unfortunately, I have not looked at this case.

Without this event, Solax with Sofar does not work at all on my Raspbery Pi 4.
Since switching to async, HA works much more smoothly and all possible hangs in the UI are gone.
However, the change has also meant that, at least on my Raspbery, the Solax integration only establishes a connection to the inverter if the start of the Solax is delayed by a few seconds or if it waits until the HA has fully started.

The static timout of 5 seconds had worked for me, but was not a nice solution.
I had been looking for a deterministic variant to delay Solax. I had tested 4 variants.
The only one that worked was the EVENT_HOMEASSISTANT_STARTED event.
I think waiting for HA start is basically not wrong, as long as we don't know exactly why it is sometimes not possible to establish a Modbus connection when starting Solax.

You just have to see if HA is already running, then continue the initialization. I'll try something.

wills106, 2024-03-04:
Just tested it on a very basic Modbus simulator and it seems to work ok!
I'll test it on the real installation and Inverter tomorrow when I don't have any automations running.

Where did you learn about `if hass.is_running:`?
I find the HA documentation quite hard to follow as it appears incomplete (at least to me) and only shows / explains the absolute minimum and doesn't always give a full understanding.

wills106, 2024-03-05:
Just released [2024.03.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.1) which allows the Integration to start reading from the Inverter without the need to restart HA again.

wrfz, 2024-03-05:
> Where did you learn about `if hass.is_running:`? 

I searched in Core sources for `EVENT_HOMEASSISTANT_STARTED` and found that it is triggered in core.py
In the same place the state is changed: `self.set_state(CoreState.running)`
I checked where `CoreState.running` is used and found the `is_running` method.
> I find the HA documentation quite hard to follow as it appears incomplete (at least to me) and only shows / explains the absolute minimum and doesn't always give a full understanding.

You are right. It is totally surprising that HA is so relatively stable, even though there is almost no central documentation.
I would have expected it here:
https://developers.home-assistant.io/docs/dev_101_hass/
If you feel like it, feel free to expand the documentation (there's an edit button at the bottom :-). 
Otherwise I might do it when I get the chance.

wills106, 2024-03-30:
Closing down as this issue is sorted.

closed 2024-03-30 by wills106