wills106/homeassistant-solax-modbus #781: [Bug]: Solax EV-Charger Integration shows no entities at all after update

Philipp-X1, 2024-02-27:
### Describe the bug

After updating the Integration to 2024.02.7b2 to check out if the EV_Charger is now working ( changes from PatrikTrestik in (https://github.com/PatrikTrestik)(https://github.com/wills106/homeassistant-solax-modbus/pull/772) Thank you very much! ), It shows no Entities at all now.
Am i to early to try with this beta?

### Integration Version

2024.02.7b2

### Homeassistant core version

2024.2.4

### Inverter brand

X3-EVC-11K

### Plugin used

plugin_solax_ev_charger.py

### Serial prefix

C31103

### Inverter firmware versions

N/A

### Connection Method

Waveshare RS485 to ETH

### Dongle firmware

N/A

### Detailed Error Log

2024-02-27 19:33:24.541 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry SolaX_EV for button
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 605, in async_unload
    result = await component.async_unload_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/button/__init__.py", line 78, in async_unload_entry
    return await component.async_unload_entry(entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 198, in async_unload_entry
    raise ValueError("Config entry was never loaded!")
ValueError: Config entry was never loaded!
2024-02-27 19:33:24.552 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry SolaX_EV for number
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 605, in async_unload
    result = await component.async_unload_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/number/__init__.py", line 125, in async_unload_entry
    return await component.async_unload_entry(entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 198, in async_unload_entry
    raise ValueError("Config entry was never loaded!")
ValueError: Config entry was never loaded!
2024-02-27 19:33:24.559 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry SolaX_EV for select
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 605, in async_unload
    result = await component.async_unload_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/select/__init__.py", line 115, in async_unload_entry
    return await component.async_unload_entry(entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 198, in async_unload_entry
    raise ValueError("Config entry was never loaded!")
ValueError: Config entry was never loaded!
2024-02-27 19:33:24.565 ERROR (MainThread) [homeassistant.config_entries] Error unloading entry SolaX_EV for sensor
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 605, in async_unload
    result = await component.async_unload_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/sensor/__init__.py", line 147, in async_unload_entry
    return await component.async_unload_entry(entry)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/helpers/entity_component.py", line 198, in async_unload_entry
    raise ValueError("Config entry was never loaded!")
ValueError: Config entry was never loaded!

### Additional context

All entities where grey. I deleted the Integration and tryed to load it again but now it shows not even a device anymore.

wills106, 2024-02-27:
There are still some on going issues with the async change from #762 #778 
On my Gen 3 Hybrid I'm not even getting a serial number #780 so further work is needed.

Philipp-X1, 2024-03-14:
Closing this down as it was working again after updating to the latest version

closed 2024-03-14 by Philipp-X1