wills106/homeassistant-solax-modbus #782: [Bug]: no devices no entities

mnh637, 2024-02-29:
### Describe the bug

when it works it's great, but frequently it becomes unavailable. I have 2 solax inverters and a victron. solax ones can just go unavailable, one or the other on their own or after update, sometimes they come back sometimes not, i have been running 2 HA boxes and it can continue to work on one when the other fails, no real pattern. the victron modbus just works any update any reboot, but can't get the solax one to be stable. Can you help please. ? 

### Integration Version

2024.02.1

### Homeassistant core version

2024.2.5

### Inverter brand

Solax

### Plugin used

plugin_solax.py

### Serial prefix

srfq7m

### Inverter firmware versions

9.03

### Connection Method

Solax pocket wifi 3

### Dongle firmware

3.001.02

### Detailed Error Log

would you like access to my HA to play with? I've set up a second HA and  would be ok with you accessing it to help me and others? wont' post details here but will open firewall to you if you can help.
many thanks

LnK-Holly, 2024-03-03:
I seem to be having same issue with Solax inverter, no rhyme nor reason that I can identity. Was working fine until integration update last few days. 

Any help much appreciated 🙏

LnK-Holly, 2024-03-03:
Further info, My connection is LAN ethernet to Modbus, ethernet connection stays, IP the same, integration just seems to stop communicating with the inverter.

RosFol, 2024-03-03:
I have similar issue. After an update it is communicating with an inverter half an hour at maximum and than it shows last values with no update. Modbus serial connection ...

wills106, 2024-03-05:
2024.02.6 Introduced async communication. Can you try [2024.03.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.1) as that further improved the async method.

If it's still doing it on [2024.03.1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.1) Can you get post the [detailed error logs](https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#detailed-error-log)

mnh637, 2024-03-05:
updated, I'll see how it goes, thank you.

LnK-Holly, 2024-03-15:
Just to update, the 2024.03.1 update seems to have fixed it for me, thanks Will!

mnh637, 2024-03-18:
I've not had any more issues either, Thanks too.

closed 2024-03-18 by mnh637

Trevbil, 2024-04-16:
2004.03.01 actually broke mine today.

mnh637, 2024-04-18:
Hi,
I have found that removing the Wi-Fi dongle and putting it back in again, then reloading the int

2024.03.01 actually broke mine today.

Trevbil, 2024-04-18:
Hi @mnh637 

I have been able to solve the issue by changing to a static IP and deleting and reinstalling the integration. The integration has been working fine for over 24 hours. Fingers crossed I don't have to do anything more