wills106/homeassistant-solax-modbus #783: "Platform solax_modbus does not generate unique IDs."

uspino2, 2024-03-01:
### Describe the bug

I've been starting to get this Error on HA restart. I don't think I've renamed any entity: 

```
Log Details (ERROR)
Logger: homeassistant.components.sensor
Source: helpers/entity_platform.py:636
Integration: Sensor 
First occurred: 03:43:50 (1 occurrences)
Last logged: 03:43:50
Platform solax_modbus does not generate unique IDs. ID Growatt_Modbus_today_solar_energy already exists - ignoring sensor.growatt_modbus_today_s_solar_energy
```

Any advise? Thanks 

### Integration Version

2024.02.6

### Homeassistant core version

2024.2.5

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

AL1.0Z

### Connection Method

Waveshare

wills106, 2024-03-16:
There was a duplicate entity. I have removed this in [2024.03.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.2)

wills106, 2024-03-29:
Closing down as duplicate entity has been removed.

closed 2024-03-29 by wills106