wills106/homeassistant-solax-modbus #784: [Bug]: 'sofar' - '3.3-12KTLX-G3' PV Power Total no longer correct after firmware update of inverter

darkrain-nl, 2024-03-02:
### Describe the bug

After updating my inverter to firmware version: V100011-10-14 (sorry don't know the old firmware version) the PV Power Total value is changed from scale 0.01 to 0.1 (like the hybrid models). Another change I noticed is it doesn't show 10-90W values anymore, only 100W and above, so lower then 100W is shown as 0W.
[image]

The scale I can fix locally, and I could do a pull request for it, but that would probably be a breaking change on inverters that are still on the older firmware... Or is there a way to have it be depending on firmware?

What are the options here, is there something that can be done to show the 10-90W values as well, can the scale be adjusted just for this firmware version?

### Integration Version

2024.02.6

### Homeassistant core version

2024.2.5

### Inverter brand

Sofar

### Plugin used

plugin_sofar.py

### Serial prefix

SS2ES1

### Connection Method

Waveshare adapter

wills106, 2024-03-04:
I have seen your PR.

I think we either need to detect Firmware version through the Integration. Or have the option to manually override the scaling.

darkrain-nl, 2024-03-04:
That makes sense, I do not have time to work on this anytime soon... but feel free to close the current PR.

Can come back to it later, or somebody else might have a go...

wills106, 2024-03-04:
I'll leave it open as a reminder.

We have done similar overrides on SolaX. So I can't see it being a  big issue.

d4ystek, 2025-01-05:
I have the same problem only with all entities (data is not correct). Did you manage to solve the problem? Software version 370

closed 2025-10-03 by github-actions[bot]