wills106/homeassistant-solax-modbus #786: [Bug]: Solis - RHI-3P10K-HVES-5G: Setting Engery Control Switch Setting not working

kai-seipp, 2024-03-02:
### Describe the bug

Hi Guys, 

**Motivation / goal:** 
Since I now have an electric car I am very much interested in being able to alter some battery storage settings to avoid draining my home battery when chargeing my car. 
Therefore I now take some interest in these settings. 

**Issue description:** 
_energy_storage_control_switch: 
selecting different setting will not work as expected: 

- OK: Choosing "Selfuse - No Grid Chargeing" is working as expected. 
- Bug: Choosing "Timed Charge/Discharge - No Grid Charging" - is triggering "Timed Charge/Discharge - No Grid Charging" - But is actually allowing grid charging
- Bug: Choosing "Backup/Reserve - No Grid Charging" switches the inverter back to "Selfuse - No Grid Charging" 
- Bug / redundant: Choosing "Selfuse" switches the inverter to "Selfuse - No Grid Charging" 
- Bug / redundant: Choosing "Timed Charge/Discharge" - is triggering "Timed Charge/Discharge - No Grid Charging"
- unclear: "Battery Awaken" will go back to Selfuse - no Grid Charging. I don't know what the expected behavior would be
- Bug: "Choosing Backup/Reserve - No Timed Charge/Discharge" results in "Selfuse - No Grid Charging"
- Bug: Choosing "Backup/Reserve" as value results the Inverter to be set to "Timed Charge/Discharge - No Grid Charging" 

I can't test "Off-Grid Mode" as I don't have my house of grid

**Expected:** 
It would be super helpful being able to switch to Backup/Reserve Mode to prevent battery drainage. 
I would also suggest to strip down the possible options to those we can working. 

**Enviroment**
I am Running on 2024.01.8 again, as 2024.2.x kept messing up my statistics (will have to address within an other issue). 
Running on current HA version 

### Integration Version

2024.1.8

### Homeassistant core version

2024.2.5

### Inverter brand

Solis

### Plugin used

plugin_solis.py

### Serial prefix

110CA2

### Connection Method

Waveshare Adapter

### Detailed Error Log

There are no errors within the log

i-jk, 2024-04-12:
Yes, I'm not sure where all these new values for Energy Control Switch come from? They seemed broadly correct before the recent updates but now there's loads and they don't align with the inverter's actual behaviour. 

I'm on 5G RHI 5 kW... All the "good", working ones seem to end in "no grid charging", which is a misnomer.

I'll look at the history and work out when the regressions happened. It's possible we need two lists of settings for different model ranges.

i-jk, 2024-04-12:
Ok so I've looked through the history and this commit broke the mode: https://github.com/wills106/homeassistant-solax-modbus/commit/0731c6d8e68ff69109bf79998e2f737cc22ae5ef#diff-b7608f0f1474cba38ac0549ffa6731114128b485fad0fd58c13c165e6b44475bL743

The issue is this will make people's inverters discharge when they think they're choosing self use. It's bad, it will potentially cost people money - I just exported all my stored solar due to this mislabeled option and now need to pay for electricity for tonight.

I think we need more rigor around commits that change these.

And we also need to look at the naming because it's pretty crazy now, some names are so long they can't be read on mobile.

i-jk, 2024-04-12:
The correct values for line 775 and 776 of plugin_solis.py are:
```
                33: "Self-Use",
                35: "Timed Charge/Discharge",
```

Can't raise PR, on holiday doing from mobile.

i-jk, 2024-04-12:
Sod it, timed charge/discharge isn't right. I'll look properly as soon as I'm near my laptop. Self use as above is correct, tho.

wills106, 2024-05-18:
Does "Energy Storage Control Switch" still need correcting?

kai-seipp, 2024-05-20:
> Does "Energy Storage Control Switch" still need correcting?

Hi Will, 

I did install the latest Version (did run on January version until yesterday, as I had major issues with newer once) to check. Yes there had been no changes made regarding this issue. So yes, that would be great! 

But I will also open a new issue to make the integration working with Solis again in general. Currently the integration can't be used when an inverter goes offline during nighttime. (Just as heads up).

neilrfuller, 2024-06-09:
Hi Folks,

Think I am experiencing these issues too. Very much a newbie with all this, but tried to switch inverters to "Backup/Reserve No Grid Charging". Inverters immediately started to charge from grid. I'm on Intelligent Octopus go, my motivation was to leave my batteries set to not charge from grid (I have sufficient solar to do that these days), but prevent them from being drained overnight when my car starts smartcharging. Be great if this could be fixed, everything else seems to work great, my thanks and admiration for all your efforts here.

i-jk, 2024-08-11:
This integration is _broken for Solis_, @wills106. Can we get a revert of the problematic commits referenced in this issue post? Then all will be well.

i-jk, 2024-08-11:
How can we mark a bug as critical? This one is

wills106, 2024-08-11:
> This integration is broken for Solis

As in completely doesn't work? Or the incorrectly named mode?

Do you agree with the comment? https://github.com/wills106/homeassistant-solax-modbus/commit/0731c6d8e68ff69109bf79998e2f737cc22ae5ef#commitcomment-143902523

wills106, 2024-09-15:
Closing down as no further response.
If there are still issues either reopen or raise a new issue.

closed 2024-09-15 by wills106

kai-seipp, 2024-09-19:
Hi Will, 

I have taken your last comment as trigger to update the integration the it's current version. 
I can't stress enough: For Solis - at least with my setup - it is 100% broken and unusable. 

I would love to help debugging. But It would be great if we could set a time like a couple of days of a week that works best for everyone as I have to revert to an old version to avoid totally messing up my database. 

Until we get this working I would advice against using the integration for Solis (or at least the hybrid model I use). 
Here some screenshot of the magnatude of wrong information flooding my database. 

[image]
Since updateing our solar plant creates almost 4kW even after midnight (and no, not near the north pole)

[image]
Look at the scale of production even during night hours. Consuption numbers get affected since there is no feed into the grid so HA assumes all that energy must have been used within the house. 

Let me know if you would want to takle this together. I don't have the skills to do so.