wills106/homeassistant-solax-modbus #791: Improvement online offline

iobrokerneu, 2024-03-05:
### Describe the bug

Hello, is there a way to create an entity that reports whether the inverter is online or offline? In iobroker there is this "entity" and it is called "status"

I usw growatt mod 5000TL3-xh with ark hv battery.

Thanks 
Sascha 

### Integration Version

2024.01.1

### Homeassistant core version

2024.2.1

### Inverter brand

Growatt

### Plugin used

plugin_growatt.py

### Serial prefix

JDM0DB

### Connection Method

USR TCP232 - 410S

iobrokerneu, 2024-08-04:
Hello, are there any updates regarding the suggestion for improvement? Regards, Sascha

wills106, 2024-08-04:
"Run Mode" provides the status of the Inverter, or do you mean when it's connected to the Internet?

iobrokerneu, 2024-08-04:
Hi, the inverter goes "offline" once a day for about 1-2 minutes. During this time, it essentially checks itself, and I can monitor this through iobroker to see if it's online or offline.

Unfortunately, I can't find this "run mode." Is there a screenshot of it?

Thanks and regards, Sascha

wills106, 2024-08-04:
I hadn't added in the status for your new Inverter. I'll push an update out later today.

wills106, 2024-08-04:
Just pushed [2024.08.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.3) out

iobrokerneu, 2024-08-04:
Hi, I just installed the new version. Unfortunately, no data is being displayed anymore. All data is no longer available.

wills106, 2024-08-04:
What's displayed in the error logs?

iobrokerneu, 2024-08-04:
sorry, where exactly can I find that?

wills106, 2024-08-04:
https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/

iobrokerneu, 2024-08-04:
Logger: homeassistant.components.sensor
Quelle: helpers/entity_platform.py:764
Integration: Sensor (Dokumentation, Probleme)
Erstmals aufgetreten: 13:37:58 (1 Vorkommnisse)
Zuletzt protokolliert: 13:37:58

Platform solax_modbus does not generate unique IDs. ID GrowattModbus_battery_voltage is already used by sensor.growattmodbus_battery_voltage - ignoring sensor.GrowattModbus_battery_voltage

2024-08-04 13:37:33.793 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant

iobrokerneu, 2024-08-04:
is that sufficient?

wills106, 2024-08-04:
Just released [2024.08.4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.4)

iobrokerneu, 2024-08-04:
unfortunately it still doesn't work. this only shows up in the log when I search for solax

2024-08-04 14:12:56.717 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant

iobrokerneu, 2024-08-04:
2024-08-04 14:12:56.712 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hon which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-08-04 14:12:56.713 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration homematicip_local which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-08-04 14:12:56.713 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration skodaconnect which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-08-04 14:12:56.714 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration growatt_server_api which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-08-04 14:12:56.715 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration landroid_cloud which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-08-04 14:12:56.716 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration iphonedetect which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-08-04 14:12:56.717 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-08-04 14:12:56.718 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration hacs which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-08-04 14:13:03.912 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' calls hass.http.register_static_path which is deprecated because it does blocking I/O in the event loop, instead call `await hass.http.async_register_static_paths([StaticPathConfig("/hacsfiles/frontend", "/config/custom_components/hacs/hacs_frontend_experimental", False)])`; This function will be removed in 2025.7 at custom_components/hacs/frontend.py, line 49: hass.http.register_static_path(, please create a bug report at https://github.com/hacs/integration/issues
2024-08-04 14:13:03.922 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' calls hass.http.register_static_path which is deprecated because it does blocking I/O in the event loop, instead call `await hass.http.async_register_static_paths([StaticPathConfig("/hacsfiles/iconset.js", "/config/custom_components/hacs/iconset.js", True)])`; This function will be removed in 2025.7 at custom_components/hacs/frontend.py, line 57: hass.http.register_static_path(, please create a bug report at https://github.com/hacs/integration/issues
2024-08-04 14:13:03.928 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' accesses hass.components.frontend. This is deprecated and will stop working in Home Assistant 2024.9, it should be updated to import functions used from frontend directly at custom_components/hacs/frontend.py, line 68: hass.components.frontend.async_register_built_in_panel(, please create a bug report at https://github.com/hacs/integration/issues
2024-08-04 14:13:03.937 WARNING (MainThread) [homeassistant.helpers.frame] Detected that custom integration 'hacs' calls hass.http.register_static_path which is deprecated because it does blocking I/O in the event loop, instead call `await hass.http.async_register_static_paths([StaticPathConfig("/hacsfiles", "/config/www/community", True)])`; This function will be removed in 2025.7 at custom_components/hacs/base.py, line 1162: self.hass.http.register_static_path(, please create a bug report at https://github.com/hacs/integration/issues
2024-08-04 14:13:06.234 ERROR (SyncWorker_4) [pyicloud.base] Authentication required for Account. (421)
2024-08-04 14:13:06.362 WARNING (ImportExecutor_0) [homeassistant.components.device_tracker] SOURCE_TYPE_ROUTER was used from iphonedetect, this is a deprecated constant which will be removed in HA Core 2025.1. Use SourceType.ROUTER instead, please report it to the author of the 'iphonedetect' custom integration
2024-08-04 14:13:06.383 WARNING (ImportExecutor_0) [homeassistant.components.device_tracker] SOURCE_TYPE_ROUTER was used from iphonedetect, this is a deprecated constant which will be removed in HA Core 2025.1. Use SourceType.ROUTER instead, please report it to the author of the 'iphonedetect' custom integration
2024-08-04 14:13:07.754 ERROR (SyncWorker_4) [pyicloud.base] Authentication required for Account. (421)
2024-08-04 14:13:07.756 ERROR (SyncWorker_4) [homeassistant.components.icloud.account] Your password for 'wulfpacks@gmx.de' is no longer working; Go to the Integrations menu and click on Configure on the discovered Apple iCloud card to login again
[... cut ...]
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Not connected[AsyncModbusTcpClient 192.168.0.47:1]
2024-08-04 14:25:08.143 WARNING (MainThread) [custom_components.solax_modbus.plugin_growatt] growatt: reading firmware number from address 0xbb9 failed; other address may succeed
2024-08-04 14:25:08.143 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] growatt: cannot find firmware version, even not for other Inverter
2024-08-04 14:25:08.143 ERROR (MainThread) [custom_components.solax_modbus.plugin_growatt] unrecognized growatt inverter type - firmware version : unknown

iobrokerneu, 2024-08-04:
[image]

iobrokerneu, 2024-08-04:
Hi.

Was an error possibly found?

wills106, 2024-08-04:
It's not showing any errors. Have you tried power cycling the USR? Or has the IP address changed if using DHCP?

iobrokerneu, 2024-08-04:
This will be displayed after restarting home assistant
[solax.txt](https://github.com/user-attachments/files/16488329/solax.txt)

wills106, 2024-08-04:
Have you tried deleting the Inverter from the Integration and adding it back in? There has been quite a few changes since 2024.01.1

iobrokerneu, 2024-08-04:
After I did that, there is only 1 device left. Previously there were 3 devices and the entities have also become fewer. but still no data is displayed.

iobrokerneu, 2024-08-04:
[image]

iobrokerneu, 2024-08-04:
[image]

iobrokerneu, 2024-08-20:
Hello.

Are there any new updates? I deleted the device, installed the new update, and re-added the device. Unfortunately, I no longer have any data after that. Greeting

iobrokerneu, 2024-08-25:
@wills106 

Good morning, I just installed version 2024.08.12. Unfortunately, no data is still displayed. Greetings Sasha

iobrokerneu, 2024-08-25:
@wills106 

Dieser Fehler wurde von einer benutzerdefinierten Integration verursacht

Logger: custom_components.solax_modbus.config_flow
Quelle: custom_components/solax_modbus/config_flow.py:163
Integration: SolaX Inverter Modbus (Dokumentation, Probleme)
Erstmals aufgetreten: 09:23:58 (1 Vorkommnisse)
Zuletzt protokolliert: 09:23:58

instance name SolaX already defined or default name for non-default inverter

iobrokerneu, 2024-08-25:
@wills106 

2024-08-25 09:20:46.598 WARNING (SyncWorker_0) [homeassistant.loader] We found a custom integration solax_modbus which has not been tested by Home Assistant. This component might cause stability problems, be sure to disable it if you experience issues with Home Assistant
2024-08-25 09:23:58.866 WARNING (MainThread) [custom_components.solax_modbus.config_flow] instance name SolaX already defined or default name for non-default inverter

wills106, 2024-08-25:
It looks like your either adding the same Inverter in again, or adding a second Inverter without changing the default name on the second Inverter?

iobrokerneu, 2024-08-25:
@wills106 

I removed the inverter and then added the same inverter again. I did that because no data was displayed after the update and I thought it would work again afterwards. unfortunately not.

iobrokerneu, 2024-08-25:
@wills106 

I tried everything. no data after update. Data deleted and re-added, no data. a new one added no data.

iobrokerneu, 2024-08-28:
@wills106 

hi, I have just installed the update 2024.08.13 and now the data is displayed again. Thanks

but unfortunately the status has been removed, whether the changer is online or offline. greetings Sascha

TonyMathiesen, 2024-08-28:
Regarding the missing run mode (register 3000) which currently stops the integration (assumes sleep mode) for Growatt MOD. I believe that is due to the overlapping registers for the MOD inverters, as holding register address 3000 is 8-bit and input register address 3000 is 16-bit devided into high and low bit for state and run mode. I only removed the GEN3 to get the plugin working again. 

I'm not sure how to modify the plugin to support the state and run mode for GEN3. But I will gladly support with testing/debugging if needed. 

Holding reg (8-bit):
[image]

Input reg (16-bit high/low):
[image]

wills106, 2024-08-29:
Just had another look at it, I have forgotten to add the input_register part

```
# TL-X TL-XH
    GrowattModbusSensorEntityDescription(
        name = "Inverter State",
        key = "inverter_state",
        register = 3000,
        unit = REGISTER_U8H,
        scale = { 0: "Waiting", 3: "Fault", 4: "Flash", 5: "PV Bat Online", 6: "Bat Online", },
        allowedtypes = GEN3 | GEN4,
    ),
    GrowattModbusSensorEntityDescription(
        name = "Run Mode",
        key = "run_mode",
        register = 3000,
        unit = REGISTER_U8L,
        scale = { 0: "Standby", 1: "Normal", 3: "Fault", 4: "Flash", },
        allowedtypes = GEN3 | GEN4,
    ),
```

Should be:

```
# TL-X TL-XH
    GrowattModbusSensorEntityDescription(
        name = "Inverter State",
        key = "inverter_state",
        register = 3000,
        register_type = REG_INPUT,
        unit = REGISTER_U8H,
        scale = { 0: "Waiting", 3: "Fault", 4: "Flash", 5: "PV Bat Online", 6: "Bat Online", },
        allowedtypes = GEN3 | GEN4,
    ),
    GrowattModbusSensorEntityDescription(
        name = "Run Mode",
        key = "run_mode",
        register = 3000,
        register_type = REG_INPUT,
        unit = REGISTER_U8L,
        scale = { 0: "Standby", 1: "Normal", 3: "Fault", 4: "Flash", },
        allowedtypes = GEN3 | GEN4,
    ),
```

wills106, 2024-08-29:
I haven't pushed it out as a release yet but 2024.08.14 of https://github.com/wills106/homeassistant-solax-modbus/blob/main/custom_components/solax_modbus/plugin_growatt.py includes the above changes, if someone want's to test it?

TonyMathiesen, 2024-08-29:
I'm currently testing it, as there are some conflicts with incorrect "GEN3" on some other sensor. 
After cleaning that it fails with reading this group of input registers, however 1-124 is 8bit registers and 3000-3249 are 16bit. 
Any suggestion on what to add to seperate the two reading groups?

[image]

TonyMathiesen, 2024-08-29:
Arh believe I found the issue.. Have to completely change all the input 1-124 input registers from GEN2 and switch to 3000-3249 GEN4 structure. 

Holding register range: 0-124 + 3000-3124
Input register range: 3000-3124 + 3125-3249

Let me test if that helps sorts out the registers for the MOD TL-XH inverters.

TonyMathiesen, 2024-08-29:
It seems we cannot avoid using mixture of two ranges 0-124 and 3000-3249 in order to cover the functionality. 
@wills106 Is there any way to tell the plugin to split the input block in two groups?

wills106, 2024-08-29:
> 1-124 is 8bit registers and 3000-3249 are 16bit

I think the majority of the 1-124 registers are a mixture of 16 and 32bit ones.

Some of the 3,000 range are 8bit and also 16 & 32bit.

> Is there any way to tell the plugin to split the input block in two groups?

It's meant to split them into 100 count register blocks.

https://github.com/wills106/homeassistant-solax-modbus/blob/97cf9df93b7d3e1a5bc86784cde54479b465d545/custom_components/solax_modbus/plugin_growatt.py#L4974

Something must be broken on the Growatt plugin?

wills106, 2024-08-29:
You can also force a newblock, change:

https://github.com/wills106/homeassistant-solax-modbus/blob/97cf9df93b7d3e1a5bc86784cde54479b465d545/custom_components/solax_modbus/plugin_growatt.py#L3334-L3342

to

```
    GrowattModbusSensorEntityDescription(
        name = "Inverter State",
        key = "inverter_state",
        newblock = True,
        register = 3000,
        register_type = REG_INPUT,
        unit = REGISTER_U8H,
        scale = { 0: "Waiting", 3: "Fault", 4: "Flash", 5: "PV Bat Online", 6: "Bat Online", },
        allowedtypes = GEN3 | GEN4,
    ),
```

TonyMathiesen, 2024-08-30:
Found that the plugin_growatt.py is missing the "auto_block_ignore_readerror = True" in the plugin_instance, which is why it stops for all the registers and not only the block group. 

Adding the newblock did not do any difference. Despite the plugin splitting up the other groups.
[image]

Added some debug logger info to see what is going on. And it seems for some reason the spliting is not working for register 3000 as it is "dict" which according to the logic is then skipping the splitting. 
[image]
Splitting logic:
[image]

@wills106 Any clues to why that is and if this can explain why? 😊

wills106, 2024-08-31:
I have released [2024.08.14](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.08.14) with the change, I'll leave the issue open until confirmed it works.

TonyMathiesen, 2024-08-31:
2024.08.14 will not close this with "Inverter State" and "Run Mode" (register 3000), as it is currenly commented out. 
Adding the newblock is not working when added to the plugin_growatt.py for some reason. 
[https://github.com/wills106/homeassistant-solax-modbus/issues/791#issuecomment-2322114347](url)

iobrokerneu, 2024-09-06:
@wills106 

hi, I have installed version 2024.09.1 ​​and generally everything works.

but unfortunately the entities where I can see whether the inverter is online or offline are still missing. Greetings Sascha

iobrokerneu, 2024-09-24:
@wills106 

Hello, are there any updates on this topic? Is it possible to create an entity where you can see if the inverter is online or offline?  
Best regards,  
Sascha

iobrokerneu, 2024-10-13:
@infradom @wills106 

Good morning, I wanted to ask if it’s even possible to implement this? Best regards.

TonyMathiesen, 2024-10-15:
@wills106 I made a workaround the issue with register 3000 in PR #1095. It should solve this issue with run mode :)

closed 2024-10-21 by wills106

iobrokerneu, 2024-10-22:
@TonyMathiesen @wills106 

Thanks

PR #1025 (2024-08-31): Growatt: Fix sensors unknown due to issue #791. Add EMS / Battery control and other fixes
Fixes: add “auto_block_ignore_readerror = True” to growatt_plugin (fixes #1022)

Adds:
EMS Discharging Rate
EMS Discharging Stop SOC
EMS Charging Rate
EMS Charging Stop SOC
EPS Switch (for MOD series)
Total Grid Export (for MOD series)

Removes:
"Battery First Maximum SOC"  & "Charger Upper SOC" (Replaced by EMS controls above) 
"Limit Grid Export" (Priority mode select) as this is not for changing run mode for TL-XH. 
"Total Load" as not for TL-XH
Comments out register 3000 for “Inverter State” and “Run  [... cut]
commit 52887f7a (2024-08-31): Update plugin_growatt.py

Fixes: add “auto_block_ignore_readerror = True” to growatt_plugin (fixes #1022)

Adds:
EMS Discharging Rate
EMS Discharging Stop SOC
EMS Charging Rate
EMS Charging Stop SOC
EPS Switch (for MOD series)
Total Grid Export (for MOD series)

Removes:
Battery First Maximum SOC  & Charger Upper SOC (Replaced by EMS control above)
Limit Grid Export / Priority mode select as this is not for changing run mode for TL-XH
R
Total Load as not for TL-XH
Comments out register 3000 for “Inverter State” and “Run mode” until issue is fixed #791

Corrects:
Typo in naming of: bettery_comb [... cut]
PR #1095 (2024-10-21): Growatt: Fix run mode #791
Work around the register 3000 not being split up. Read out the value as internal and then use value_function to translate.
To fix #791