wills106/homeassistant-solax-modbus #793: [Bug]: 'Sofar' - 'HYD15KTL' Connection lost since v2024.02.6

sahomm, 2024-03-06:
### Describe the bug

First of all, thank you very much for the great work done here!

Since the update to 2024.03.1, Solax no longer receives data via Modbus after a few minutes. Everything is ok for the first few minutes after start/restart Solax, then it stops. I was also able to make this observation with version 2024.06.2. With version 2024.05.X the connection remains stable (see screenshot).
No evaluable data can be found in the log using the search term “solax” or “modbus”.
There are entries here, but they have to do with writing registers, which is a different problem and I would describe it in a new issue after checking.

I suspect that this is related to this #762 

[image]

### Integration Version

2024.03.1

### Homeassistant core version

2024.2.5

### Inverter brand

Sofar

### Plugin used

plugin_sofar.py

### Serial prefix

000000

### Connection Method

LSE3

rcourtman, 2024-03-06:
I've noticed a similar issue and it's been happening for me for about two weeks. Unfortunately, I can't be useful and provide version numbers, but I've noticed that all my data just flatlines and I can't find a rhyme or reason as to why. 

[image]

Only when I reboot Home Assistant does it pick up again.

cschlipf, 2024-03-10:
A workaround: Create an automation that checks whether a value does not change for e.g. 1 hour, which then would restart the Solax integration.

wills106, 2024-05-18:
Closing down as no further response in past couple of months.

Feel free to reopen if needed.

closed 2024-05-18 by wills106