wills106/homeassistant-solax-modbus #794: [Bug]: No data after upgrade from 2024.02.2 to 2024.03.1

Kars-de-Jong, 2024-03-07:
### Describe the bug

I have a SolaX X3-MIC. It was not entirely clear to me whether the change announced in release 2024.02.5 was applicable to me, so I decided to remove the integration before upgrading and then adding it again, just to be sure.

After restarting and re-adding the integration, the entities remained unavailable.

The log file shows this info which seems relevant:

`2024-03-07 10:03:35.039 INFO (MainThread) [custom_components.solax_modbus] read_error : SolaX X3-MIC cannot read holding registers at device 1 position 0x300
NoneType: None`

### Integration Version

2024.03.1

### Homeassistant core version

2024.1.6

### Inverter brand

SolaX Power

### Plugin used

plugin_solax.py

### Serial prefix

MU802T

### Inverter firmware versions

N/A

### Connection Method

USR-N510

### Dongle firmware

N/A

### Detailed Error Log

Logs from version 2024.03.1:
```
2024-03-07 10:02:43.850 INFO (MainThread) [custom_components.solax_modbus.config_flow] detected HA core version 2024 1
2024-03-07 10:02:43.850 INFO (MainThread) [custom_components.solax_modbus.config_flow] starting configflow - domain = solax_modbus
2024-03-07 10:03:14.983 INFO (MainThread) [custom_components.solax_modbus.config_flow] validating base: {'name': 'SolaX X3-MIC', 'interface': 'tcp', 'read_modbus_addr': 1, 'plugin': 'solax', 'scan_interval': 10, 'read_eps': False, 'read_dcb': False, 'read_pm': False}
2024-03-07 10:03:14.984 INFO (MainThread) [custom_components.solax_modbus.config_flow] validating base config for SolaX X3-MIC: pre: {'name': 'SolaX X3-MIC', 'interface': 'tcp', 'read_modbus_addr': 1, 'plugin': 'solax', 'scan_interval': 10, 'read_eps': False, 'read_dcb': False, 'read_pm': False}
2024-03-07 10:03:24.454 INFO (MainThread) [custom_components.solax_modbus.config_flow] validating host: returning data: {'host': '10.0.0.10', 'port': 502, 'tcp_type': 'tcp'}
2024-03-07 10:03:24.454 INFO (MainThread) [custom_components.solax_modbus.config_flow] title configflow solax_modbus name: {'name': 'SolaX X3-MIC', 'interface': 'tcp', 'read_modbus_addr': 1, 'plugin': 'solax', 'scan_interval': 10, 'read_eps': False, 'read_dcb': False, 'read_pm': False, 'host': '10.0.0.10', 'port': 502, 'tcp_type': 'tcp'}
2024-03-07 10:03:24.457 INFO (MainThread) [custom_components.solax_modbus] trying to load plugin - plugin_name: solax
2024-03-07 10:03:24.517 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX X3-MIC: trying to determine inverter type
2024-03-07 10:03:24.677 WARNING (MainThread) [custom_components.solax_modbus.plugin_solax] SolaX X3-MIC: reading serial number from address 0x0 failed; other address may succeed
2024-03-07 10:03:24.677 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] Read SolaX X3-MIC 0x0 serial number before potential swap: None
2024-03-07 10:03:24.847 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] Read SolaX X3-MIC 0x300 serial number before potential swap: UM08T2xxxxxxxx
2024-03-07 10:03:24.865 INFO (MainThread) [custom_components.solax_modbus.button] hub.wakeuButton: None
2024-03-07 10:03:24.866 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x300 seriesnumber to block with start 0x300
2024-03-07 10:03:24.866 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x318 rtc to block with start 0x300
2024-03-07 10:03:24.866 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x33d firmwareversion_dsp to block with start 0x300
2024-03-07 10:03:24.866 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x33e firmwareversion_arm to block with start 0x300
2024-03-07 10:03:24.866 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x400 pv_voltage_1 to block with start 0x400
2024-03-07 10:03:24.866 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x401 pv_voltage_2 to block with start 0x400
2024-03-07 10:03:24.866 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x402 pv_current_1 to block with start 0x400
2024-03-07 10:03:24.866 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x403 pv_current_2 to block with start 0x400
2024-03-07 10:03:24.866 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x404 grid_voltage_l1 to block with start 0x400
2024-03-07 10:03:24.866 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x405 grid_voltage_l2 to block with start 0x400
2024-03-07 10:03:24.866 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x406 grid_voltage_l3 to block with start 0x400
[... 56 lines cut ...]
2024-03-07 10:11:06.204 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x409 grid_frequency_l3 to block with start 0x400
2024-03-07 10:11:06.204 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x40a grid_current_l1 to block with start 0x400
2024-03-07 10:11:06.204 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x40b grid_current_l2 to block with start 0x400
2024-03-07 10:11:06.204 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x40c grid_current_l3 to block with start 0x400
2024-03-07 10:11:06.204 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x40d inverter_temperature to block with start 0x400
2024-03-07 10:11:06.204 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x40e measured_power to block with start 0x400
2024-03-07 10:11:06.204 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x40f run_mode to block with start 0x400
2024-03-07 10:11:06.204 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x410 measured_power_l1 to block with start 0x400
2024-03-07 10:11:06.204 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x411 measured_power_l2 to block with start 0x400
2024-03-07 10:11:06.204 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x412 measured_power_l3 to block with start 0x400
2024-03-07 10:11:06.204 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x414 pv_power_1 to block with start 0x400
2024-03-07 10:11:06.204 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x415 pv_power_2 to block with start 0x400
2024-03-07 10:11:06.204 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x423 total_yield to block with start 0x400
2024-03-07 10:11:06.204 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x425 today_yield to block with start 0x400
2024-03-07 10:11:06.204 INFO (MainThread) [custom_components.solax_modbus.sensor] Starting new block at 0x435 
2024-03-07 10:11:06.204 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x435 inverter_power to block with start 0x435
2024-03-07 10:11:06.204 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x437 total_grid_export to block with start 0x435
2024-03-07 10:11:06.204 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x439 total_grid_import to block with start 0x435
2024-03-07 10:11:06.204 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x43b today_grid_export to block with start 0x435
2024-03-07 10:11:06.204 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x43c today_grid_import to block with start 0x435
2024-03-07 10:11:06.204 INFO (MainThread) [custom_components.solax_modbus.sensor] SolaX X3-MIC returning input block: 0x400 0x427 [1024, 1025, 1026, 1027, 1028, 1029, 1030, 1031, 1032, 1033, 1034, 1035, 1036, 1037, 1038, 1039, 1040, 1041, 1042, 1044, 1045, 1059, 1061]
2024-03-07 10:11:06.204 INFO (MainThread) [custom_components.solax_modbus.sensor] SolaX X3-MIC returning input block: 0x435 0x43d [1077, 1079, 1081, 1083, 1084]
2024-03-07 10:11:06.205 INFO (MainThread) [custom_components.solax_modbus.sensor] computedRegs: {'pv_power_total': SolaXMicModbusSensorEntityDescription(key='pv_power_total', device_class=<SensorDeviceClass.POWER: 'power'>, entity_category=None, entity_registry_enabled_default=True, entity_registry_visible_default=True, force_update=False, icon='mdi:solar-power-variant', has_entity_name=False, name='PV Power Total', translation_key=None, unit_of_measurement=None, last_reset=None, native_unit_of_measurement=<U [... cut ...] unit_of_measurement=None, allowedtypes=8192, scale=1, read_scale_exceptions=None, read_scale=1, blacklist=None, register=-1, rounding=1, register_type=1, unit='_uint16', newblock=False, value_function=<function value_function_grid_import at 0x7f5807eb3ba0>, wordcount=None, sleepmode=1, ignore_readerror=False)}
2024-03-07 10:12:06.628 INFO (MainThread) [custom_components.solax_modbus] no local data file found after 5 tries - is this a first time run? or didnt you modify any DATA_LOCAL entity?
```

tsunami-ad, 2024-03-14:
I have an issue too after upgrading home assistant core to 2024.3.0
The solax integration doesnt finds any devices or entities.. before updationg, it worked fine.

Chawiby, 2024-03-15:
Same here. I see these error messages in the log:

[image]

tom3606, 2024-03-15:
All of my entities are unavailable since the update. I've tried to roll back to a backup and they're unavailable there too.

tsunami-ad, 2024-03-15:
I had the same problem.
Go to the ethernet cable, disconnect it and plug it back in.
Than reload the integration. It worked for me

tom3606, 2024-03-15:
> I had the same problem. Go to the ethernet cable, disconnect it and plug it back in. Than reload the integration. It worked for me

Unfortunately disconnecting and reconnecting hasn't worked for me, rolling back to the older version didn't work,. Restarting Home Assistant and checking the configuration of the integration made no difference. I've checked that it is pointing at the right USB device as sometimes it's the case that it switched from USB1 to USB0. It was on the correct USB port.

cewbdex, 2024-03-15:
Encountered this too. Restarting HA did not help, had to restart the whole VM with HAOS.

Kars-de-Jong, 2024-03-18:
@tsunami-ad @Chawiby @tom3606 @cewbdex Please create another issue for this. It does not match this issue. My inverter is detected correctly. The issue is that the integration is reading registers that are not present on my inverter / with my firmware version.

cewbdex, 2024-03-18:
Is it? I don't think so, 0x300 is serial number, I would say it's your custom name.

Kars-de-Jong, 2024-03-18:
I don't agree, 0x300 just happens to be the first register in the block that is defined which includes the registers for rtc, firmwareversion_dsp and/or firmwareversion_arm:

```
2024-03-07 10:03:24.866 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x300 seriesnumber to block with start 0x300
2024-03-07 10:03:24.866 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x318 rtc to block with start 0x300
2024-03-07 10:03:24.866 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x33d firmwareversion_dsp to block with start 0x300
2024-03-07 10:03:24.866 INFO (MainThread) [custom_components.solax_modbus.sensor] adding register 0x33e firmwareversion_arm to block with start 0x300
```

Register 0x300 itself can be read just fine:
```
2024-03-07 10:03:24.847 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] Read SolaX X3-MIC 0x300 serial number before potential swap: UM08T2xxxxxxxx
```

jimbaxter, 2024-03-19:
I started seeing this yesterday after I tried reinstalling the Solax integration to fix the problems with 2024.03.4 before it was updated.

Now I am on version 2024.03.6 but getting the following error, I have tried power cycling the inverter and the Raspberry PI:

Logger: custom_components.solax_modbus.plugin_solax
Source: custom_components/solax_modbus/plugin_solax.py:68
integration: SolaX Inverter Modbus ([documentation](https://github.com/wills106/homsassistant-solax-modbus), [issues](https://github.com/wills106/homsassistant-solax-modbus/issues))
First occurred: 15:00:30 (1 occurrences)
Last logged: 15:00:30

SolaX: attempt to read serialnumber failed at 0x0 data: None
Traceback (most recent call last):
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 68, in async_read_serialnr
    inverter_data = await hub.async_read_holding_registers(unit=hub._modbus_addr, address=address, count=7)
                    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 401, in async_read_holding_registers
    return await self._client.read_holding_registers(address, count, **kwargs)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/site-packages/pymodbus/client/base.py", line 167, in async_execute
    self.send(packet)
  File "/usr/local/lib/python3.12/site-packages/pymodbus/transport/transport.py", line 391, in send
    self.transport.write(data)  # type: ignore[attr-defined]
    ^^^^^^^^^^^^^^^^^^^^
AttributeError: 'NoneType' object has no attribute 'write'

Kars-de-Jong, 2024-03-19:
@jimbaxter That is not the same as the issue I reported here. I haven't tried any newer version yet because the change logs don't indicate any fixes yet.

jimbaxter, 2024-03-19:
I have created a new ticket for my issue at https://github.com/wills106/homeassistant-solax-modbus/issues/814

wills106, 2024-03-21:
I have just released [#794b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/%23794b1) which tries to prevent the Integration from reading 0x33D and 0x33E
I'm not sure if I need to blacklist your Inverter as UM08T2 due to the byteswap bug in some X3 MIC's

Kars-de-Jong, 2024-03-21:
@wills106 Probably yes, I also saw the serial "swapped" in the device info page when I ran 2024.03.1.
I will test this tomorrow when the sun is back up. Thanks!

wills106, 2024-03-22:
The changes are also in [2024.03.7b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.7b1) which may be easier to install through HACS as it's using a conventional numbering system for the release name.

Kars-de-Jong, 2024-03-22:
@wills106 I tested 2024.03.7b1, it still has the same problem. I then manually updated the two blacklist entries to UM08T2 and now it works.

But like I said, the serial number is also shown in the wrong order on the device info page, so I think you should correct it at the source when reading, then it should also be fine in the blacklist.

wills106, 2024-03-22:
I can't properly test it but does [2024.03.7b2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.7b2) swap the serial for you?

Kars-de-Jong, 2024-03-22:
@wills106 I will check this again tomorrow, when the sun is up.

wills106, 2024-03-22:
I have found a way to better test it and I believe [2024.03.7b3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.7b3) should work better.

kajik2, 2024-03-22:
Hello. I have the same problem in version 2024.03.7b3

wills106, 2024-03-22:
@kajik2 do you still need to invert the serial number in the blacklists like Kars mentions?

kajik2, 2024-03-22:
> @kajik2 do you still need to invert the serial number in the blacklists like Kars mentions?
> 
> Edit: Does your Inverter have the serial number starting MU802T? Or a different X3 MIC Gen1?

i made a mistake, it doesn't work for me, but i have solax pocket wifi

wills106, 2024-03-22:
@kajik2 Which Inverter do you own including the first 4-6 digits of the serial?

kajik2, 2024-03-22:
> @kajik2 Which Inverter do you own including the first 4-6 digits of the serial?

X3-Hybrid-G4 - H34A10

wills106, 2024-03-22:
@kajik2 Your issue isn't related to this issue then.

This issue revolves around the fact some early X3 MIC's are missing certain registers.

The X3 MIC uses a completely different register set to the Hybrids.

Try HA 2024.03.2 and Integration 2024.03.7bx

kajik2, 2024-03-22:
> @kajik2 Your issue isn't related to this issue then.
> 
> This issue revolves around the fact some early X3 MIC's are missing certain registers.
> 
> The X3 MIC uses a completely different register set to the Hybrids.
> 
> Try HA 2024.03.2 and Integration 2024.03.7bx

Should i start a new issue? It hasn't worked for about a week. Through several versions. Last I had HA 2024.3.2 and 2024.03.7b3

Kars-de-Jong, 2024-03-23:
@

> I have found a way to better test it and I believe [2024.03.7b3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.7b3) should work better.

@wills106 Thanks, this version works for me and shows the serial number correctly on the device info page.

The only minor thing I noticed is that the log message now also shows the serial number correctly, but still says "before potential swap":
`2024-03-23 06:55:13.567 INFO (MainThread) [custom_components.solax_modbus.plugin_solax] Read SolaX X3-MIC 0x300 serial number before potential swap: MU802Txxxxxxxx`

wills106, 2024-03-23:
I'll reword that in the next release as I moved the swap code to a different location.

I'll close this issue down then.

closed 2024-03-23 by wills106

wills106, 2024-03-25:
Can you try [2024.03.7b4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.7b4) to make sure it loads correctly and still displays your Serial correctly?

2024.03.7b3 incorrectly byte swapped AC & Hybrid Inverters by mistake. So I want to make sure 2024.03.7b4 still works correctly on the MIC's

Kars-de-Jong, 2024-03-25:
@wills106 Checking it now.