wills106/homeassistant-solax-modbus #804: [Bug]: 'Sofar' - 'HYD KTL10-3PH' Some connection error - from time to time integration stops.

hubertozog, 2024-03-13:
### Describe the bug

From time to time the integrtion stops. All data freez wont get updated. There is no any real error in logs. Integration just hangs... To make it work even reboot of HA wont help. Also on the inverter there is no RS485 icon. If i try to send some request by RS485 usb adapter via modpool all works. I dont have any idea what its the issue. Sometimes it work for 5 days, sometimes issue will occure after one day...

### Integration Version

2024.03.1

### Homeassistant core version

2024.3.0

### Inverter brand

Sofar

### Plugin used

plugin_sofar.py

### Serial prefix

SP2ES110N

### Connection Method

Waveshare adapter RS485 to ETH

### Detailed Error Log

2024-03-13 17:44:30.798 ERROR (MainThread) [homeassistant] Error doing job: Future exception was never retrieved
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected

024-03-13 15:29:18.277 DEBUG (MainThread) [custom_components.solax_modbus] connect modbus
2024-03-13 15:29:24.535 INFO (MainThread) [custom_components.solax_modbus] exception 'NoneType' object has no attribute 'write' : Sofar cannot read holding registers at device 1 position 0x445
NoneType: None
2024-03-13 15:29:24.536 DEBUG (MainThread) [custom_components.solax_modbus] assuming sleep mode - slowing down by factor 10
2024-03-13 15:29:24.536 ERROR (MainThread) [homeassistant] Error doing job: Future exception was never retrieved

cschlipf, 2024-03-14:
@hubertozog is this the same as reported with #793 ?

Other question: Did you properly terminate the RS485 bus on the inverter side?

hubertozog, 2024-03-15:
@cschlipf No, today i changed the wire and also made the termination. Will see if that helps.

cschlipf, 2024-03-15:
Thanks for the update. A proper termination is absolutely necessary for reliable operation

ppiwowar, 2024-03-15:
Same here. It's been working for months ... and after recent integration update >2024.03.1 it needed restart once per day. Yesterday after HA update >2024.3.1 it needs restart every 5 mins. I see Modbus reading registers error in the log (same as above)

ppiwowar, 2024-03-16:
I did quick&dirty automation that reloads config of this automation every 5 mins. So I keep data consistency of my Sofar inverter. I hope this problem goes away with new version ... soon,  P L E A S E.

alias, 2024-03-16:
perhaps related: I use a LSE3 LAN Stick Logger on a SofarSolar HYD. Operation stops after a while (less 24h) with timeput messages: 

```
2024-03-16 11:34:26.152 INFO (MainThread) [custom_components.solax_modbus] exception AsyncModbusTcpClient.close() got an unexpected keyword argument 'intern' : SofarSolar cannot read holding registers at device 1 position 0x404
NoneType: None
2024-03-16 11:34:26.154 DEBUG (MainThread) [custom_components.solax_modbus] assuming sleep mode - slowing down by factor 10
2024-03-16 11:34:36.148 DEBUG (MainThread) [custom_components.solax_modbus] connect modbus
2024-03-16 11:34:46.160 DEBUG (MainThread) [custom_components.solax_modbus] assuming sleep mode - slowing down by factor 10
2024-03-16 11:34:51.150 DEBUG (MainThread) [custom_components.solax_modbus] connect modbus
2024-03-16 11:35:06.151 DEBUG (MainThread) [custom_components.solax_modbus] connect modbus
2024-03-16 11:35:06.166 DEBUG (MainThread) [custom_components.solax_modbus] assuming sleep mode - slowing down by factor 10
202
```

Perhaps this repeated "slowing down" multipy up?

hubertozog, 2024-03-16:
@cschlipf @ppiwowar @alias 

Ok, got some new info. I am not sure, but after update of ha, and correct termination with new good wire, its working even worst.... But one goodside its that if i reload integration its gonna work again all the time. @ppiwowar coudl you provide to me your automation?

Its works about 2-5 mins and link is down, on waveshare adapter link icon tunrs yellow. here are the error msgs

Traceback (most recent call last):
  File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 1181, in _call_connection_lost
    super()._call_connection_lost(exc)
  File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 901, in _call_connection_lost
    self._protocol.connection_lost(exc)
  File "/usr/local/lib/python3.12/site-packages/pymodbus/transport/transport.py", line 288, in connection_lost
    self.close(intern=True)
TypeError: AsyncModbusTcpClient.close() got an unexpected keyword argument 'intern'

2024-03-16 12:44:24.273 WARNING (MainThread) [asyncio] socket.send() raised exception.
2024-03-16 12:44:29.274 WARNING (MainThread) [asyncio] socket.send() raised exception.
2024-03-16 12:44:34.276 WARNING (MainThread) [asyncio] socket.send() raised exception.
2024-03-16 12:47:21.182 WARNING (MainThread) [asyncio] socket.send() raised exception.
2024-03-16 12:47:26.184 WARNING (MainThread) [asyncio] socket.send() raised exception.

Maybe that will help a little.

I also lookend and checked wireing with diffrent adapter - waveshare RS485 to usb and using modpoll - tried to read register and for 30 mint didnt have any communication error.

vitjurasek, 2024-03-16:
I have the same problem 
https://github.com/wills106/homeassistant-solax-modbus/issues/807
after recent update
it is necessary to restart several times a day because the communication freezes

cschlipf, 2024-03-16:
Which Firmware version do you guys have? With V110022 for example there is a known Modbus Problem that causes frequent hangs. With V110051 this is fixed.

BTW: From time from time I also see that values are not updated, but it always recovers itself. At the same time however EVCC does still update its values.

hubertozog, 2024-03-16:
Inverter:V110044
PCU:V010019
BDU:V010010
BMS:V010019

So i need to go with update right now...

hubertozog, 2024-03-16:
Ok even after update of sofar:

2024-03-16 15:24:29.344 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback _SelectorSocketTransport._call_connection_lost(None)
Traceback (most recent call last):
File "/usr/local/lib/python3.12/asyncio/events.py", line 88, in _run
self._context.run(self._callback, *self._args)
File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 1181, in _call_connection_lost
super()._call_connection_lost(exc)
File "/usr/local/lib/python3.12/asyncio/selector_events.py", line 901, in _call_connection_lost
self._protocol.connection_lost(exc)
File "/usr/local/lib/python3.12/site-packages/pymodbus/transport/transport.py", line 288, in connection_lost
self.close(intern=True)
TypeError: AsyncModbusTcpClient.close() got an unexpected keyword argument 'intern'
2024-03-16 15:24:54.768 WARNING (MainThread) [asyncio] socket.send() raised exception.
2024-03-16 15:24:59.769 WARNING (MainThread) [asyncio] socket.send() raised exception.
2024-03-16 15:25:04.771 WARNING (MainThread) [asyncio] socket.send() raised exception.
2024-03-16 15:25:09.771 WARNING (MainThread) [asyncio] socket.send() raised exception.

Also is its possible to see in integraton data of 2 batteries connected to one BTS?

cschlipf, 2024-03-16:
With #797 you will be able to see the data of up to 8 batteries. Note however that in case of the BTS each BDU counts as one battery not matter how many BTS-5K modules are attached to that BDU. If you have connected two BDUs, then the second BDU will be shown as battery two.

In case of AMASS GTX3000, each module is counted as individual battery.

wills106, 2024-03-16:
I have just released [2024.03.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.2) which forces pyModbus to v3.6.6
v3.6.5 and v3.6.6 have quite a few fixes for various communication related issues by the look.
I'm not getting any of these errors people are encounting so I don't know how much of an improvement v3.6.6 will make.

hubertozog, 2024-03-16:
@wills106 

Thanks for support! For me right now it looks a lot better. 

I got:

024-03-16 18:32:49.916 ERROR (MainThread) [homeassistant] Error doing job: Future exception was never retrieved
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
2024-03-16 18:33:00.318 WARNING (MainThread) [custom_components.solax_modbus.sensor] holding register already used: 0x104d bat_config_depth_of_discharge
2024-03-16 18:33:00.318 WARNING (MainThread) [custom_components.solax_modbus.sensor] holding register already used: 0x104e bat_config_end_of_discharge
2024-03-16 18:33:00.319 WARNING (MainThread) [custom_components.solax_modbus.sensor] holding register already used: 0x1055 bat_config_address_4

But that dont kill or freez the integration, it will come to live by its own so its a lot better than it was before. If its possible to gather any data that could be helpfull for you - feel free to ask.

cschlipf, 2024-03-16:
> @wills106
>
> Thanks for support! For me right now it looks a lot better.
>
> I got:
>
> 024-03-16 18:32:49.916 ERROR (MainThread) [homeassistant] Error doing job: Future exception was never retrieved [...]

That is another unrelated issue I just fixed with #810

ppiwowar, 2024-03-17:
@wills106 , thank you a great deal! Your release 2024.03.2 works here like a charm for almost 12 hours.

cschlipf, 2024-03-17:
Can you give some feedback what happens, when you try to set values? E.g. try to change the Energy Storage Mode or try to change Battery Charge/Discharge powers. Does this work without errors for you?

In my case I am receiving errors since 2024.03. And I have problems setting some values like the Battery power values.

hubertozog, 2024-03-17:
@ppiwowar Do you have any errors like this? i got around 100 from yesterday till today :/

2024-03-17 14:40:28.774 ERROR (MainThread) [homeassistant] Error doing job: Future exception was never retrieved
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected

@cschlipf I got same issues, cant change Energy storage mode etc.

wills106, 2024-03-17:
Can you try [2024.03.3](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.3) I have added a check to see if the Inverter is connected and if it isn't then it try's and connects.

hubertozog, 2024-03-17:
Sure, update in progress, gonna check is that help. Thanks for your hard work!!!

ppiwowar, 2024-03-17:
@hubertozog , I am a newbie of HA. Where do I need to look for these errors, pls?

hubertozog, 2024-03-17:
@wills106 
Unfortunatly still:

2024-03-17 15:33:41.225 ERROR (MainThread) [homeassistant] Error doing job: Future exception was never retrieved
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected

Gonna see how often it will appear.

@ppiwowar Settings -> System -> Logs

wills106, 2024-03-17:
https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#detailed-error-log

wills106, 2024-03-17:
@hubertozog are you getting anything like:

```
2024-03-17 14:41:02.230 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2024-03-17 14:41:02.370 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 192.168.x.x:502
```

In your logs? Or `Unable to connect to Inverter at...`

hubertozog, 2024-03-17:
Thats all i got related to Solax integration:

```
2024-03-17 15:33:41.225 ERROR (MainThread) [homeassistant] Error doing job: Future exception was never retrieved
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
2024-03-17 15:37:45.806 ERROR (MainThread) [homeassistant] Error doing job: Future exception was never retrieved
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
```

wills106, 2024-03-17:
Do you have your logger set with `info` like in the list I posted a couple of posts ago?

hubertozog, 2024-03-17:
Just setet it up like you told, gonna paste detailed log only when i got error.

ppiwowar, 2024-03-17:
@hubertozog , any errors in the log were by the time of installing 2024.03.2. No errors thenafter.

hubertozog, 2024-03-17:
@wills106 Heres error log. Hope that it will help.

```
2024-03-17 16:02:20.866 ERROR (MainThread) [homeassistant] Error doing job: Future exception was never retrieved
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
2024-03-17 16:14:38.466 INFO (MainThread) [custom_components.solax_modbus] exception 'NoneType' object has no attribute 'write' : Sofar cannot read holding registers at device 1 position 0x60b
NoneType: None
2024-03-17 16:14:38.466 ERROR (MainThread) [homeassistant] Error doing job: Future exception was never retrieved
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
2024-03-17 16:18:43.073 INFO (MainThread) [custom_components.solax_modbus] exception 'NoneType' object has no attribute 'write' : Sofar cannot read holding registers at device 1 position 0x404
NoneType: None
2024-03-17 16:18:43.073 ERROR (MainThread) [homeassistant] Error doing job: Future exception was never retrieved
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
2024-03-17 16:22:48.499 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2024-03-17 16:22:55.440 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.100.x.x:502
2024-03-17 16:31:00.680 INFO (MainThread) [custom_components.solax_modbus] exception 'NoneType' object has no attribute 'write' : Sofar cannot read holding registers at device 1 position 0x404
NoneType: None
2024-03-17 16:31:00.680 ERROR (MainThread) [homeassistant] Error doing job: Future exception was never retrieved
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
2024-03-17 16:35:05.290 INFO (MainThread) [custom_components.solax_modbus] exception 'NoneType' object has no attribute 'write' : Sofar cannot read holding registers at device 1 position 0x1029
NoneType: None
2024-03-17 16:35:05.291 ERROR (MainThread) [homeassistant] Error doing job: Future exception was never retrieved
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
2024-03-17 16:55:33.161 INFO (MainThread) [custom_components.solax_modbus] exception 'NoneType' object has no attribute 'write' : Sofar cannot read holding registers at device 1 position 0x5c4
NoneType: None
2024-03-17 16:55:33.161 ERROR (MainThread) [homeassistant] Error doing job: Future exception was never retrieved
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
2024-03-17 16:55:33.632 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2024-03-17 16:55:34.855 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.100.x.x:502
2024-03-17 17:16:00.978 INFO (MainThread) [custom_components.solax_modbus] exception 'NoneType' object has no attribute 'write' : Sofar cannot read holding registers at device 1 position 0x484
NoneType: None
2024-03-17 17:16:00.978 ERROR (MainThread) [homeassistant] Error doing job: Future exception was never retrieved
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
2024-03-17 17:20:08.130 INFO (MainThread) [custom_components.solax_modbus] exception 'NoneType' object has no attribute 'write' : Sofar cannot read holding registers at device 1 position 0x404
NoneType: None
2024-03-17 17:20:08.131 ERROR (MainThread) [homeassistant] Error doing job: Future exception was never retrieved
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
2024-03-17 17:28:18.457 INFO (MainThread) [custom_components.solax_modbus] exception 'NoneType' object has no attribute 'write' : Sofar cannot read holding registers at device 1 position 0x60b
NoneType: None
2024-03-17 17:28:18.457 ERROR (MainThread) [homeassistant] Error doing job: Future exception was never retrieved
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
2024-03-17 17:28:18.755 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2024-03-17 17:28:19.789 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.100.x.x:502
2024-03-17 17:44:40.482 INFO (MainThread) [custom_components.solax_modbus] exception 'NoneType' object has no attribute 'write' : Sofar cannot read holding registers at device 1 position 0x504
NoneType: None
2024-03-17 17:44:40.482 ERROR (MainThread) [homeassistant] Error doing job: Future exception was never retrieved
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
2024-03-17 17:52:53.414 INFO (MainThread) [custom_components.solax_modbus] exception 'NoneType' object has no attribute 'write' : Sofar cannot read holding registers at device 1 position 0x484
NoneType: None
2024-03-17 17:52:53.414 ERROR (MainThread) [homeassistant] Error doing job: Future exception was never retrieved
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
2024-03-17 18:01:03.887 INFO (MainThread) [custom_components.solax_modbus] Inverter is not connected, trying to connect
2024-03-17 18:01:25.842 INFO (MainThread) [custom_components.solax_modbus] Inverter connected at 10.100.x.x:502
2024-03-17 18:09:15.420 INFO (MainThread) [custom_components.solax_modbus] exception 'NoneType' object has no attribute 'write' : Sofar cannot read holding registers at device 1 position 0x100a
NoneType: None
2024-03-17 18:09:15.421 ERROR (MainThread) [homeassistant] Error doing job: Future exception was never retrieved
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Client is not connected
```

Zwer2k, 2024-03-17:
Those who have problems. Can you please show your settings of Waveshare adapter RS485 to ETH?

hubertozog, 2024-03-18:
I found a copy of HA without solax integration, made recovery from it. Reinstalled integration again. Setted up all again and i will see how it will work.
[image]
[image]

Zwer2k, 2024-03-18:
[image]
My settings are slightly different. I have no problems at all. You can perhaps test my settings

hubertozog, 2024-03-19:
For right now after about 10 hours, after fresh install and Zwer2k Waveshare settings - no error or any messeage right now. Will check if anything will happen in next few days - but i hope so that everything its ok right now :)

cschlipf, 2024-03-19:
What about changing values like the Energy Storage Mode?

hubertozog, 2024-03-19:
Unfortuneatly not working, checket it few seconds ago.

```
2024-03-19 07:20:32.801 INFO (MainThread) [custom_components.solax_modbus.select] writing Sofar select register 4368 value 1
2024-03-19 07:20:58.157 INFO (MainThread) [custom_components.solax_modbus.select] *** local data written tou_charge_start_time: 1536
2024-03-19 07:21:03.181 INFO (MainThread) [custom_components.solax_modbus] saved modified persistent date: {'_version': 1, 'passive_mode_grid_power': 0, 'passive_mode_battery_power_min': 0, 'passive_mode_battery_power_max': 0, 'feedin_max_power': 0, 'timing_charge_power': 1000, 'timing_discharge_power': 1000, 'timing_id': 0.0, 'tou_id': 0, 'tou_target_soc': 0, 'tou_charge_power': 0, 'feedin_limitation_mode': 'Disabled', 'timing_charge': 'Disabled', 'timing_charge_start_time': None, 'timing_charge_end_time': None, 'timing_discharge_start_time': None, 'timing_discharge_end_time': None, 'tou_control': 'Enabled - Charging & Discharging', 'tou_charge_start_time': '06:00', 'tou_charge_end_time': '16:45', 'passive_mode_timeout': None, 'passive_mode_timeout_action': None}
2024-03-19 07:22:45.315 INFO (MainThread) [custom_components.solax_modbus.number] *** local data written tou_target_soc: 60.0
2024-03-19 07:22:48.165 INFO (MainThread) [custom_components.solax_modbus] saved modified persistent date: {'_version': 1, 'passive_mode_grid_power': 0, 'passive_mode_battery_power_min': 0, 'passive_mode_battery_power_max': 0, 'feedin_max_power': 0, 'timing_charge_power': 1000, 'timing_discharge_power': 1000, 'timing_id': 0.0, 'tou_id': 0, 'tou_target_soc': 60.0, 'tou_charge_power': 0, 'feedin_limitation_mode': 'Disabled', 'timing_charge': 'Disabled', 'timing_charge_start_time': None, 'timing_charge_end_time': None, 'timing_discharge_start_time': None, 'timing_discharge_end_time': None, 'tou_control': 'Enabled - Charging & Discharging', 'tou_charge_start_time': '06:00', 'tou_charge_end_time': '16:45', 'passive_mode_timeout': None, 'passive_mode_timeout_action': None}
2024-03-19 07:23:13.040 INFO (MainThread) [custom_components.solax_modbus.number] *** local data written tou_charge_power: 1300.0
2024-03-19 07:23:18.183 INFO (MainThread) [custom_components.solax_modbus] saved modified persistent date: {'_version': 1, 'passive_mode_grid_power': 0, 'passive_mode_battery_power_min': 0, 'passive_mode_battery_power_max': 0, 'feedin_max_power': 0, 'timing_charge_power': 1000, 'timing_discharge_power': 1000, 'timing_id': 0.0, 'tou_id': 0, 'tou_target_soc': 60.0, 'tou_charge_power': 1300.0, 'feedin_limitation_mode': 'Disabled', 'timing_charge': 'Disabled', 'timing_charge_start_time': None, 'timing_charge_end_time': None, 'timing_discharge_start_time': None, 'timing_discharge_end_time': None, 'tou_control': 'Enabled - Charging & Discharging', 'tou_charge_start_time': '06:00', 'tou_charge_end_time': '16:45', 'passive_mode_timeout': None, 'passive_mode_timeout_action': None}
2024-03-19 07:23:19.743 ERROR (MainThread) [homeassistant.components.websocket_api.http.connection] [140667539567040] None
Traceback (most recent call last):
File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 239, in handle_call_service
response = await hass.services.async_call(
^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/core.py", line 2319, in async_call
response_data = await coro
^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/core.py", line 2356, in _execute_service
return await target(service_call)
^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 905, in entity_service_call
single_response = await _handle_entity_call(
^^^^^^^^^^^^^^^^^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/helpers/service.py", line 975, in _handle_entity_call
result = await task
^^^^^^^^^^
File "/usr/src/homeassistant/homeassistant/components/button/__init__.py", line 139, in _async_press_action
await self.async_press()
File "/config/custom_components/solax_modbus/button.py", line 91, in async_press
await self._hub.async_write_registers_multi(
File "/config/custom_components/solax_modbus/__init__.py", line 476, in async_write_registers_multi
if hasattr(descr, 'reverse_option_dict'): value = descr.reverse_option_dict[value] # string to int
~~~~~~~~~~~~~~~~~~~~~~~~~^^^^^^^
KeyError: None
2024-03-19 07:24:32.476 INFO (MainThread) [custom_components.solax_modbus.select] writing Sofar select register 4368 value 0
2024-03-19 07:25:03.191 INFO (MainThread) [custom_components.solax_modbus] saved modified persistent date: {'_version': 1, 'passive_mode_grid_power': 0, 'passive_mode_battery_power_min': 0, 'passive_mode_battery_power_max': 0, 'feedin_max_power': 0, 'timing_charge_power': 1000, 'timing_discharge_power': 1000, 'timing_id': 0.0, 'tou_id': 0, 'tou_target_soc': 0, 'tou_charge_power': 1300.0, 'feedin_limitation_mode': 'Disabled', 'timing_charge': 'Disabled', 'timing_charge_start_time': None, 'timing_charge_end_time': None, 'timing_discharge_start_time': None, 'timing_discharge_end_time': None, 'tou_control': 'Enabled - Charging & Discharging', 'tou_charge_start_time': '06:00', 'tou_charge_end_time': '16:45', 'passive_mode_timeout': None, 'passive_mode_timeout_action': None}
2024-03-19 07:25:33.172 INFO (MainThread) [custom_components.solax_modbus] saved modified persistent date: {'_version': 1, 'passive_mode_grid_power': 0, 'passive_mode_battery_power_min': 0, 'passive_mode_battery_power_max': 0, 'feedin_max_power': 0, 'timing_charge_power':  [... cut ...]  'timing_discharge_power': 1000, 'timing_id': 0.0, 'tou_id': 0, 'tou_target_soc': 0, 'tou_charge_power': 0, 'feedin_limitation_mode': 'Disabled', 'timing_charge': 'Disabled', 'timing_charge_start_time': None, 'timing_charge_end_time': None, 'timing_discharge_start_time': None, 'timing_discharge_end_time': None, 'tou_control': 'Enabled - Charging & Discharging', 'tou_charge_start_time': '06:00', 'tou_charge_end_time': '16:45', 'passive_mode_timeout': None, 'passive_mode_timeout_action': None}
```

andbad, 2024-03-21:
After upgrading HA to version 2023.03.1, the integration crashed after a few minutes. To resolve this I went back to 2023.03.0.
This morning I deleted the integration (deleting the directory under custom_components), downloaded and installed version 2023.03.6 of the component.
I then updated HA to version 2023.03.1, and it has currently been working for about 30 minutes.
After the startup, the only logs are the following:
```
2024-03-21 10:00:03.940 INFO (MainThread) [custom_components.solax_modbus] read_error : InverterZCS cannot read holding registers at device 1 position 0x1044
2024-03-21 10:00:04.811 WARNING (MainThread) [custom_components.solax_modbus] InverterZCS: read failed at 0x69a: battery_Output_energy_total 
2024-03-21 10:05:10.612 INFO (MainThread) [custom_components.solax_modbus] exception AsyncModbusTcpClient.close() got an unexpected keyword argument 'intern' : InverterZCS cannot read holding registers at device 1 position 0x620

```

By(t)e

andbad, 2024-03-21:
:(
While I was sending the previous message, the integration stopped updating the data.
The last update are on 10.00
This is the last logs:

```
2024-03-21 10:04:50.728 ERROR (MainThread) [pymodbus.logging] Unable to decode response Modbus Error: Unknown response 0
2024-03-21 10:04:50.729 ERROR (MainThread) [homeassistant] Error doing job: Fatal error: protocol.data_received() call failed.
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/asyncio/selector_events.py", line 1009, in _read_ready__data_received
    self._protocol.data_received(data)
  File ".homeassistant/homeassistant/lib/python3.11/site-packages/pymodbus/transport/transport.py", line 303, in data_received
    self.datagram_received(data, None)
  File ".homeassistant/homeassistant/lib/python3.11/site-packages/pymodbus/transport/transport.py", line 337, in datagram_received
    cut = self.callback_data(self.recv_buffer, addr=addr)
          ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File ".homeassistant/homeassistant/lib/python3.11/site-packages/pymodbus/client/base.py", line 201, in callback_data
    self.framer.processIncomingPacket(data, self._handle_response, slave=0)
  File ".homeassistant/homeassistant/lib/python3.11/site-packages/pymodbus/framer/base.py", line 139, in processIncomingPacket
    self.frameProcessIncomingPacket(single, callback, slave, **kwargs)
  File ".homeassistant/homeassistant/lib/python3.11/site-packages/pymodbus/framer/socket_framer.py", line 104, in frameProcessIncomingPacket
    if (result := self.decoder.decode(data)) is None:
        ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
pymodbus.exceptions.ModbusIOException: Modbus Error: [Input/Output] Unable to decode request
2024-03-21 10:04:50.739 ERROR (MainThread) [homeassistant] Error doing job: Exception in callback _SelectorSocketTransport._call_connection_lost(ModbusIOExcep...code request'))
Traceback (most recent call last):
  File "/usr/local/lib/python3.11/asyncio/events.py", line 80, in _run
    self._context.run(self._callback, *self._args)
  File "/usr/local/lib/python3.11/asyncio/selector_events.py", line 1117, in _call_connection_lost
    super()._call_connection_lost(exc)
  File "/usr/local/lib/python3.11/asyncio/selector_events.py", line 896, in _call_connection_lost
    self._protocol.connection_lost(exc)
  File ".homeassistant/homeassistant/lib/python3.11/site-packages/pymodbus/transport/transport.py", line 288, in connection_lost
    self.__close()
TypeError: AsyncModbusTcpClient.close() got an unexpected keyword argument 'intern'
2024-03-21 10:05:10.612 INFO (MainThread) [custom_components.solax_modbus] exception AsyncModbusTcpClient.close() got an unexpected keyword argument 'intern' : InverterZCS cannot read holding registers at device 1 position 0x620
NoneType: None
```

andbad, 2024-03-22:
Updated to 2023.3.7 Beta 1 and then Home assistant to 2023.3.1
The integration stop to update data after some minutes.

By(t)e

andbad, 2024-03-22:
In the Home Assistant 2023.3.1 [changelog](https://www.home-assistant.io/blog/2024/03/06/release-20243/#release-202431---march-14) I noticed that some changes have been made to modbus.

- Issue warning modbus configuration when modbus configuration is empty ([@janiversen](https://github.com/janiversen) - [#112618](https://github.com/home-assistant/core/pull/112618))
- modbus scan_interval: 0 is correct configuration ([@janiversen](https://github.com/janiversen) - [#112619](https://github.com/home-assistant/core/pull/112619))
- Bump pymodbus to v3.6.5 ([@janiversen](https://github.com/janiversen) - [#112629](https://github.com/home-assistant/core/pull/112629))

Considering that with the HA version 2023.3.0 (before these changes) I don't encounter the problem while with the 2023.3.1 version I do, is it possible that the cause is the changes introduced to modbus?

By(t)e

wills106, 2024-03-22:
[2024.02.6](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.02.6) Introduced the main change of using async for communications.

Introduction of pyModbus v3.6.6 was trying to address #807 
HA is updating to pyModbus v3.6.6 as well [2024.3.2](https://github.com/home-assistant/core/releases/tag/2024.3.2)

andbad, 2024-03-22:
I updated HA to version 2023.3.2 about 1 hour and a half ago and it's still working for now.
Thank you

By(t)e

wills106, 2024-03-22:
What version of the Integration are you using with HA 2024.3.2?

andbad, 2024-03-22:
2023.3.7 Beta 1

By(t)e

hubertozog, 2024-03-23:
Ok, for me for 5 days any issues for last days - all working fine. Critical was correct configuration of waveshare and clean installation of integration. Thanks for your hard work!

wills106, 2024-03-23:
Did you use the settings Zwer2k shared?

hubertozog, 2024-03-23:
Yes, the difference was with wait time and iddle time. After that tweaks - i didnt receaived any error. So IMO that was main issue. Maybe it could be good idea to add that information to documentation for Sofar users?

hubertozog, 2024-03-26:
Unfortunetl yesterday the integration stops again. With that msg. Gonna update to the newest version and will try again. The biggest issue is that it even didnt try to reconect. Tried remove integration, restart of Waveshare, HA. Nothing helps. Probably if i would connect Inverter to the other adapter it would work again. But it was faster to reboot inverter itself.

`2024-03-25 03:32:55.375 INFO (MainThread) [custom_components.solax_modbus] exception Modbus Error: [Input/Output] ERROR: No response received after 3 retries : Sofar cannot read holding registers at device 1 position 0x404`

Any other error. Onlu that information and nothing else in logs :/

wills106, 2024-03-29:
Try [2024.03.8b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.8b1) if your still having issues.

ppiwowar, 2024-03-29:
I had same issues yesterday after upgrading integration to version ... .7. After I reversed to ... .3 it is all fine for more then 24h.

hubertozog, 2024-03-29:
Ok update done, gonna test. Thanks for your hard work!

ppiwowar, 2024-04-01:
After 3 days (of downgrading to ....3) it stopped again. Testing 2024.03.8b1 now ....

hubertozog, 2024-04-06:
For now on newest version - any issue, but in meanwhile many reboots of the HA. Gonna update in the end of the next week.

hubertozog, 2024-04-16:
OK after 10 days, I didnt noticed any issue with the integration. Gonna try pasive mode right now. But normal oparation in self use works perfect without any issue on newest version. One more time thanks for your hard work!

ppiwowar, 2024-04-26:
Unfortunately lost the connection to inverter again. Do you have any idea how to reset the inverter 
                        AVOIDING COMPLETE SWITCH OFF? 
I am now more hesitant for switch-off as I have just installed a battery. To switch it off I need also to shut down battery (all DC). Of course I tried everything else - integration reload, Waveshare RS485 to ETH adapter reset. 
Like before these do not work. Only inverter switch off helped then.

lieglw, 2024-04-26:
I am in a similar situation with my second inverter.
I did a system restart and then waited a few miniatures and then did a HA restart and back it was.
But I have Solax with WLAN connected.

ppiwowar, 2024-04-26:
> One more question for those who have problems: How many devices do you use to read the data? I use the HA-Solax integration and also EVCC. If I also connect my test HA with Solax integration (i.e. read out with 3 devices), the error seems to occur even more frequently

My inverter sends data to Solarman portal through Wi-Fi logger. HA-Solax Integration connects through WaveShare ETH to RS485 directly to RS485 inverter port. No other devices on this RS485 bus. Polling frequency is 1/15 seconds.
(Strangely: I recall cases when I discovered hang-up within, say 10-20 minutes, reloading the integration was enough to bring back to normal operation.)

ppiwowar, 2024-05-07:
Do I need termination resistors at both ends of RS485 bus - at Waveshare adapter side and Sofar inverter side?
If yes, then is below the right way o install them.

"My RS485 Adapter does not work reliably

Have you double checked that the termination resistor is installed on both ends? On the adapter side make sure that it has a termination resistor in the adapter. If not add it. On the inverter side Sofar inverters also require a termination adaptor on the last connected inverter. Not that on the COM Port the PINs 1 and 2 (A+) and PINs 3 and 4 (B-) are internally connected with each other. So if your signal cable is connected to PINs 1 and 4 you can use the PINs 2 and 3 to connect your termination resistor."

cschlipf, 2024-05-07:
> Do I need termination resistors at both ends of RS485 bus - at Waveshare adapter side and Sofar inverter side? If yes, then is below the right way o install them.
>
> "My RS485 Adapter does not work [...]

Most adapters have a terminator resistor already built in. If in doubt check the manual of your adapter. So typically a resistor is only needed on the inverter side.

wills106, 2024-05-18:
Are people still having issues or can this be closed now?

ppiwowar, 2024-05-19:
Oh, I have forgot to mention, I have also updated system time ... Precisely:
10:19 Upgrade HA Core 2024.5.3 -> 2024.5.4, restart
10:26 Update System Time command sent to inverter
11:26 Integration stopped sending data
So the STOP happened exactly 1h after I pressed Update System Time. BTW, I have never updated system time before....

[6 more replies]

closed 2024-08-03 by wills106