wills106/homeassistant-solax-modbus #807: [Bug]: socket.send() raised exception.

vitjurasek, 2024-03-16:
### Describe the bug

After updating to the new version of core HA (2024.3.1), a problem with this integration started to appear - on the desktop, after a while the update of values from the inverter freezes and resumes only after restarting HA.

In log:

Logger: asyncio
Zdroj: custom_components/solax_modbus/__init__.py:380
integrace: SolaX Inverter Modbus ([dokumentace](https://github.com/wills106/homsassistant-solax-modbus), [problémy](https://github.com/wills106/homsassistant-solax-modbus/issues))
První výskyt: 11:00:52 (192 výskyty)
Naposledy logováno: 12:57:32

socket.send() raised exception.

### Integration Version

2024.03.1

### Homeassistant core version

2024.3.1

### Inverter brand

solax g4 x3-hybrid

### Plugin used

plugin_solax.py

### Serial prefix

H34A10

### Inverter firmware versions

?

### Connection Method

Socket

### Dongle firmware

?

Joerivanrokeghem, 2024-03-16:
I have the exact same problem, but with a solax x1 hybrid 5.0

Karssi, 2024-03-16:
Same here.
But you can do a reload of the integration (iso complete HA reboot). But it will only function for a short while before it hangs again.

wills106, 2024-03-16:
I have just released [2024.03.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.2) which forces pyModbus to v3.6.6
v3.6.5 and v3.6.6 have quite a few fixes for various communication related issues by the look.
I'm not getting any of these errors people are encounting so I don't know how much of an improvement v3.6.6 will make.

Joerivanrokeghem, 2024-03-16:
Hi @wills106 

I juist dit the update 15 minutes ago and it is seems to work now. 

Thanks!

WebberXX, 2024-03-17:
I still have the problem
I have deleted Solax, then restart HA and download Solax again, but its still frezees after few min
[image]

https://github.com/wills106/homeassistant-solax-modbus/assets/127571142/5c53d7e2-7b16-488d-afb1-8811d08cbe75

WebberXX, 2024-03-17:
[image]

vitjurasek, 2024-03-17:
New version not work for me. The integration still freeze

Zwer2k, 2024-03-17:
The error sounds to me like incorrect use of the lock() function

PutoPunko, 2024-03-20:
Hi,

I experienced as well communicaiton problems with my SolaX inverter. I am using Core 2024.3.1 Supervisor 2024.03.0 and OS 12.1. I found that latest version without issue for me is 2024.02.5. I tested removing prior installing (with rebootings) without sucess.

Maybe it is unrelated but my other modbus sensors in configuration.yaml get freeze as well if I update SolaX component.

Not sure if this will help but SolaX component manifest states pymodbus>=3.6.6 however my installation reports pymodbus 3.6.5

I followed these commands to get pymodbus version:

docker exec -it homeassistant /bin/bash
pip show pymodbus

Is it possible that there is some incompatibility around here?

Best regards,

Zwer2k, 2024-03-20:
By forcing version 3.6.6 for the Solax integration, other integrations that use pymodbus and have not yet been adapted for 3.6.6 have problems. 
Here, for example, huawei_solar integration. However, this has now been adapted
https://github.com/wlcrs/huawei_solar/pull/658

wills106, 2024-03-22:
HA is updating to pyModbus v3.6.6 as well [2024.3.2](https://github.com/home-assistant/core/releases/tag/2024.3.2)

wills106, 2024-03-22:
> The error sounds to me like incorrect use of the lock() function

I have been looking at other integrations using async and they don't seem to be using `async with self._lock:` as much as this Integration. So perhaps we are over using it?

wills106, 2024-03-22:
I have released [2024.03.7b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.7b1) which only uses `async with self._lock:` on the refresh of the data and not each `async_read` element.

Tested ok on my SolaX Gen3 Hybrid, but I'm not having any issues with any of these releases?

PutoPunko, 2024-03-22:
Hi,

Updated to 2024.3.2 and SolaX 2024.03.07b1. It looks OK for now :) let's wait some hours to confirm.

@wills106 thanks for your great work.

simondrake, 2024-03-23:
This looks to be the same issue as #816. I've updated to the latest version but I'm still getting frozen data.

wills106, 2024-03-23:
@vitjurasek how are you connecting to your Gen4? Waveshare or PocketWiFi?

PutoPunko, 2024-03-23:
Hi,

I had to go back again to 2024.02.05 after testing beta1. It got frozen again after few hours. Good news is that It was no longer freezing my other modbus connections.

wills106, 2024-03-23:
@PutoPunko How are you connecting to your SolaX Inverter? Waveshare or PocketWiFi?

PutoPunko, 2024-03-23:
Modbus TCP over CAT5E cable

wills106, 2024-03-23:
So your not using a Waveshare adaptor then?

PutoPunko, 2024-03-23:
no adaptor direct ETH connection

wills106, 2024-03-23:
O so you have an older Gen3 like me then.
I'm surprised you are having an issue then, I haven't experienced a single issue with mine.

PutoPunko, 2024-03-23:
yes SolaX gen3 hybrid. I am experiencing freezes with anything newer than 2024.02.05. I tried uninstalling+rebooting and then installing again unsuccesfully. Maybe It is related to having several modbus TCP devices on my config. None of them are integracions,they are directly configured from my configuration.yaml

WebberXX, 2024-03-23:
@wills106 
I still have problems!? And it doesn't matter if I use Solax 2024.03.2-2024.03.6 and even beta 2024.03.7b1/2/3

Other systems are
Core = 2024.3.3
Supervisor = 2024.03.0
Operating System = 12.1
User interface = 20240307.0

When I reload it works anywhere from a few minutes to 10 minutes then it freezes and has to reload again

No changes were made to my setup between it working perfectly and it suddenly stopping
If you're up for it, I'd be happy to create a user so you can log in yourself and look in my system

I'm saddened by this, I've loved your integration, but now I'm just losing my temper

[image]

wills106, 2024-03-23:
If yourself or anyone else is using a Waveshare have a look at this post https://github.com/wills106/homeassistant-solax-modbus/issues/804#issuecomment-2004783893

Either the "Maximum wait time of RS485" or "RS485 bus conflict" settings has helped others.

wills106, 2024-03-30:
If people are still having the data freeze after trying the settings in the previous post then try [2024.03.8b1](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.8b1)

PutoPunko, 2024-04-04:
> yes SolaX gen3 hybrid. I am experiencing freezes with anything newer than 2024.02.05. I tried uninstalling+rebooting and then installing again unsuccesfully. Maybe It is related to having several [...]

Hi,

I can confirm it works OK for me (modbus TCP) with: 

- SolaX 2024.04.1
- Core 2024.4.0
- Supervisor 2024.03.1
- OS 12.1

Best regards,

wills106, 2024-05-18:
Are people still having issue or can this be closed now?

wills106, 2024-05-26:
Closing down as no further response.
If people are still having an issue feel free to reopen.

closed 2024-05-26 by wills106