wills106/homeassistant-solax-modbus #811: [Bug]:

dz1984sg, 2024-03-17:
### Describe the bug

Seriennummer vom Ultra beginnen auch mit H3BD!
 SolaXModbusSensorEntityDescription(
        name = "Battery 2 Power Charge",
        key = "battery2_power_charge",
        native_unit_of_measurement = UnitOfPower.WATT,
        device_class = SensorDeviceClass.POWER,
        state_class = SensorStateClass.MEASUREMENT,
        register = 0x129,
        register_type = REG_INPUT,
        unit = REGISTER_S16,
        allowedtypes = AC | HYBRID | GEN2 | GEN3 | GEN4,
        icon = "mdi:battery-charging",
 SolaXModbusSensorEntityDescription(
        name = "Battery 2 Capacity",
        key = "battery2_capacity_charge",
        native_unit_of_measurement = PERCENTAGE,
        device_class = SensorDeviceClass.BATTERY,
        register = 0x12D,
        register_type = REG_INPUT,
        allowedtypes = AC | HYBRID | GEN2 | GEN3 | GEN4,

had only added power and soc so far 

### Integration Version

2024.03.3

### Homeassistant core version

2024.3.1

### Inverter brand

SolaX

### Plugin used

plugin_solax.py

### Serial prefix

H3BD20

### Inverter firmware versions

ARM10.0 / DSP7.63

### Connection Method

WiFi + LAN

### Dongle firmware

2.09

wills106, 2024-03-18:
I have added in more support to [2024.03.4](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.4)

dz1984sg, 2024-03-18:
now it no longer works .... even after a reinstallation :(

wills106, 2024-03-18:
Are you able to provided the error log so I can see what it's failing on.
https://homeassistant-solax-modbus.readthedocs.io/en/latest/faq/#detailed-error-log

wills106, 2024-03-18:
Just released [2024.03.5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.5) try this version and let me know if it's still not working. Don't forget the error logs if it still won't work.

dz1984sg, 2024-03-18:
I have converted 2024.03.3 for myself .... so I won't be testing it.

wills106, 2024-03-18:
I have added in more than just "Battery 2 Power Charge" and "Battery 2 Capacity"

dz1984sg, 2024-03-18:
I have just looked at the new file and it is still incomplete.

wills106, 2024-03-18:
Have you added in more than just "Battery 2 Power Charge" and "Battery 2 Capacity" which you mentioned in the first post?

wills106, 2024-03-18:
I am going to close this down as I have opened #813 to properly look into supporting these new Inverters.

closed 2024-03-18 by wills106

wills106, 2025-10-23:
@dz1984sg Does you Inverter have 2 or 3 MPPT's? Does your Serial have a K after H3BD20? I'm trying to find out how to detect the 3rd MPPT on some X3-Ultra20K models as per #1668