wills106/homeassistant-solax-modbus #812: [Bug]: Error on setting up entry fo Solax Modbus

JohnCannon87, 2024-03-18:
### Describe the bug

Hi, Getting the following error, was all working fine recently then suddenly broke, tried updating to latest version of plugin but no difference it seems, this is the error I'm getting on latest version:

Logger: homeassistant.config_entries
Source: config_entries.py:504
First occurred: 12:09:11 (1 occurrences)
Last logged: 12:09:11

Error setting up entry SolaX for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 504, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 120, in async_setup_entry
    plugin = importlib.import_module(f".plugin_{plugin_name}", 'custom_components.solax_modbus')
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 1136, in <module>
    option_dict = TIME_OPTIONS_GEN4 | GEN5,
                  ~~~~~~~~~~~~~~~~~~^~~~~~
TypeError: unsupported operand type(s) for |: 'dict' and 'int'

### Integration Version

2024.03.4

### Homeassistant core version

2024.3.1

### Inverter brand

Solax 

### Plugin used

plugin_solax.py

### Serial prefix

H4602A

### Inverter firmware versions

Can't find this ?

### Connection Method

Waveshare Adapter via Ethernet

### Dongle firmware

V1. 452 on the Waveshare

### Detailed Error Log

Logger: homeassistant.config_entries
Source: config_entries.py:504
First occurred: 12:09:11 (1 occurrences)
Last logged: 12:09:11

Error setting up entry SolaX for solax_modbus
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/config_entries.py", line 504, in async_setup
    result = await component.async_setup_entry(hass, self)
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/config/custom_components/solax_modbus/__init__.py", line 120, in async_setup_entry
    plugin = importlib.import_module(f".plugin_{plugin_name}", 'custom_components.solax_modbus')
             ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/local/lib/python3.12/importlib/__init__.py", line 90, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "<frozen importlib._bootstrap>", line 1387, in _gcd_import
  File "<frozen importlib._bootstrap>", line 1360, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1331, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 935, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 995, in exec_module
  File "<frozen importlib._bootstrap>", line 488, in _call_with_frames_removed
  File "/config/custom_components/solax_modbus/plugin_solax.py", line 1136, in <module>
    option_dict = TIME_OPTIONS_GEN4 | GEN5,
                  ~~~~~~~~~~~~~~~~~~^~~~~~
TypeError: unsupported operand type(s) for |: 'dict' and 'int'

tretabyte, 2024-03-18:
had to rollback to version .2 because with this morning update from 2 to 4 nothing on modbus can't be picked up by the integration

tscholz79, 2024-03-18:
Hi, 

I have exactly the same issue. This morning I made this update and afterwards the connection was broken. Did also the uninstall and reinstall but without success. Log message are the same as above.

davebarrett1959, 2024-03-18:
Also getting the same error after updating this morning. Can anyone tell me how to roll back please ?

wills106, 2024-03-18:
Just releasing a fix

wills106, 2024-03-18:
[2024.03.5](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.5) should fix this.

davebarrett1959, 2024-03-18:
Back on line 😄. Many thanks 👍

tretabyte, 2024-03-18:
can confirm it's working perfectly now, thank you for the quick fix :)

wills106, 2024-03-18:
I'll close this issue down then.

closed 2024-03-18 by wills106

JohnCannon87, 2024-03-18:
Works perfectly now thank you for the quick fix :)