wills106/homeassistant-solax-modbus #813: [FR] Add support for SolaX Gen5 X1-IES / X3-IES / X3-Ultra Hybrid Inverters

wills106, 2024-03-18:
Solax X3 Ultra 25k > `H3BC25...`

_Originally posted by @simowNgithub in https://github.com/wills106/homeassistant-solax-modbus/discussions/520#discussioncomment-8790368_

Add support for SolaX Gen5 X1-IES / X3-IES / X3-Ultra Hybrid Inverters

2024.03.5 adds support for Battery 2 and MPPT 3

It looks like quite a few of the Selects and Number entities have moved to different registers on the Gen5 Inverters.
So I wouldn't make any changes to those yet.

_Originally posted by @wills106 in https://github.com/wills106/homeassistant-solax-modbus/discussions/520#discussioncomment-8829590_

wills106, 2024-03-18:
I have just released [2024.03.6](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.03.6) which corrects the Generator Control as the registers have moved to a different location.

I have removed the Following from the Gen5:
https://github.com/wills106/homeassistant-solax-modbus/blob/176d15b889b29635aa1980ecf5c03690c03ee733/custom_components/solax_modbus/plugin_solax.py#L3259-L3382
I'm not sure if they are present on the Gen5.

simowNgithub, 2024-03-18:
@wills106 i updated and checked your new implementation, following entities getting wrong values: 

The example with the battery capcatiy is really confusing, because you map the same register to solax_battery_1_capacaty and solax_battery_capacaty. One mapping is working and the other mapping is not working.

[image]

wills106, 2024-03-19:
It looks like you have a mix of new and old entities there.
Did you manually add your Inverter like https://github.com/wills106/homeassistant-solax-modbus/discussions/520#discussioncomment-8814618
I would remove the Inverter from the Integration page, restart HA and add it back in again.
it should then configure itself correctly. We can then look at which entities need scaling correctly.

simowNgithub, 2024-03-19:
I always add a second Inverter„SolaxTest“ with the same IP address. I think then everything is on a clean setup, correct?

Krawuzzi, 2024-03-20:
removed inverter, restarted and added ti back again, 

my battery is at 20% but shows 158%
[image]

Krawuzzi, 2024-03-20:
now restored from backup to old version again

so battery input/output energy show 5.3 and 5.4 kWh with the old version which is correct
new version shows 33.0 and 11,7kWh

[image]

wills106, 2024-03-21:
I don't understand how Battery 1 Capacity can be different from Battery Capacity in the old version.

They are the same register just renamed in the new version.

Although Battery Total Capacity is returning the correct 20%

https://github.com/wills106/homeassistant-solax-modbus/blob/176d15b889b29635aa1980ecf5c03690c03ee733/custom_components/solax_modbus/plugin_solax.py#L4142-4159

Same with Battery Input / Output Energy Today they are the same register on both versions of the Integration

simowNgithub, 2024-04-03:
@wills106 what input do you need to fix this issues, at the moment im testing every release, but still same issue.

- battery 1 capacity shows wrong value
- house load and solax grid export show huge numbers
- battery 1 power shows constant "1 W"

Krawuzzi, 2024-04-04:
I updated to 2024.04.1 because the older version did not work with the April update of HomeAssistant.
I tried a few things but ended up changing it back to Gen4 by editing this line:

elif seriesnumber.startswith('H3BC25'): invertertype = HYBRID | GEN4 | X3

Whenever i change versions of this addon, or just changing from GEN5 to GEN4 I need to restart Homeassistant and also the Solax inverter. Is this normal behavior?
My guess is that something is messing up the Solax inverter when GEN5 is activated.

wills106, 2024-04-05:
Restarting HA is normal if changes to the code takes place.

I can't test reading registers that don't exist. My Inverter just return 0.

I have released [2024.04.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.04.2) does it now give any ModbusIO errors when it tries to read registers that don't exist?

simowNgithub, 2024-04-05:
> Restarting HA is normal if changes to the code takes place.
>
> I can't test reading registers that don't exist. My Inverter just return 0.
>
> I have released [...]

Does it help if i open you the modbus port for some hours?

Krawuzzi, 2024-04-06:
with 2024.04.2

[image]

[image]

[image]

[image]

[image]
[image]
[image]

KOPRajs, 2024-04-19:
I'm trying to get 2x SolaX X3-ULT-30K (master-slave) to work with HA and I can provide another information if needed.

Serial: H3BC30JC......
HA: 2024.4.2

Also I've got a PDF with the Modbus documentation v3.29 from the SolaX support, but I'm not sure if it can be uploaded directly here. Let me know where should I send it, if needed.

Krawuzzi, 2024-05-02:
I now have all my PV panels installed and I'm using all three MPPTs.
The workaround of disguising the Ultra as a Gen4 is no longer working as it only uses the first two MPPTS for PV power total value.

I have some of the most important parameters working now, except battery energy input/output and battery charge.

@KOPRajs I would be very interested in the Modbus documentation, can you upload it here or send it to the following email: modbussolaxultra@magspam.net

Thank you

simowNgithub, 2024-05-02:
> with 2024.04.2
> 
> ![image](https://private-user-images.githubusercontent.com/146495311/320210938-c39a4857-fe31-4182-b2af-1e66ddb8fb43.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ2ODYwOTAsIm5iZiI6MTcxNDY4NTc5MCwicGF0aCI6Ii8xNDY0OTUzMTEvMzIwMjEwOTM4LWMzOWE0ODU3LWZlMzEtNDE4Mi1iMmFmLTFlNjZkZGI4ZmI0My5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwMlQyMTM2MzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1hNjA3ZjFmOTZlYmI1ZjAxZjFlOTgxODlkZWJiZDI4NmU0NjZkZjIzYmE2MzU2ZDE2YTZhYjNlNDIyMjJiYmY1JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.sdH8E22GAK7NKLTRRTj63x9UDyg9ijBTVJ4YlnkFbTg)
> 
> ![image](https://private-user-images.githubusercontent.com/146495311/320210953-c776b28d-bc42-427c-97d6-ffef6e9e2083.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ2ODYwOTAsIm5iZiI6MTcxNDY4NTc5MCwicGF0aCI6Ii8xNDY0OTUzMTEvMzIwMjEwOTUzLWM3NzZiMjhkLWJjNDItNDI3Yy05N2Q2LWZmZWY2ZTllMjA4My5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwMlQyMTM2MzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wYWZjZmJlY2QxMTRkMjk4NjYzMTZjOGZmNzU2NDEwMmRhMjIzNDA4YjZhZmVlYTMyODgxYzFiNTI1OTRlNWIzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.GEYxwsHyXF6_S1gAswULResvJ4Dkkq36EWgvmdfjPp0)
> 
> ![image](https://private-user-images.githubusercontent.com/146495311/320210957-9e9047c2-d8b9-4565-a85d-8a3314232136.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ2ODYwOTAsIm5iZiI6MTcxNDY4NTc5MCwicGF0aCI6Ii8xNDY0OTUzMTEvMzIwMjEwOTU3LTllOTA0N2MyLWQ4YjktNDU2NS1hODVkLThhMzMxNDIzMjEzNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwMlQyMTM2MzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jNTQ0ODU2YTc0MmQ5YmQxNWU1YjI0ZWYyYTRlNDc2ZTVlN2I3YzliNWJmOThjNzQxZjJjYzYxYjAwODdhMzg4JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.WuBBhy-8P5q0imEcL3Zd7Gv8cKyMREOGUgBYGnTYFcI)
> 
> ![image](https://private-user-images.githubusercontent.com/146495311/320210961-2d480103-0ae8-43a5-b4bf-42dbad728906.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ2ODYwOTAsIm5iZiI6MTcxNDY4NTc5MCwicGF0aCI6Ii8xNDY0OTUzMTEvMzIwMjEwOTYxLTJkNDgwMTAzLTBhZTgtNDNhNS1iNGJmLTQyZGJhZDcyODkwNi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwMlQyMTM2MzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02ZWVkMGMxYjM0NGRlMWEwMjY1MmM1NDJmN2UzOGVmYzk1NmRhMGE2OGYwNjQxZWNiMjUzM2FmZDRhZTE1MGQzJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.QJqgCGqQ5MORVpGQJEnc_pYgeX5yZxVxH7GWXR--x7g)
> 
> <img alt="image" width="168" src="https://private-user-images.githubusercontent.com/146495311/320211103-c3f577fe-be32-49fc-a12f-bd5378a96560.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ2ODYwOTAsIm5iZiI6MTcxNDY4NTc5MCwicGF0aCI6Ii8xNDY0OTUzMTEvMzIwMjExMTAzLWMzZjU3N2ZlLWJlMzItNDlmYy1hMTJmLWJkNTM3OGE5NjU2MC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwMlQyMTM2MzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT02Nzg1NTFmNmYxNjQ5NzNjYThjZjZmYTIxODRkNDJiMzhjZTNmZWRkZjQxY2I4OTY1ODM2NGQ2YmQ3OTEzYmY5JlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.BVg3S7EkryUuXekfZdvNvk_MLBLkFnObHmgrqtx20GE"> <img alt="image" width="175" src="https://private-user-images.githubusercontent.com/146495311/320211219-fb536142-5b14-41ce-8b02-f67e6013179f.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ2ODYwOTAsIm5iZiI6MTcxNDY4NTc5MCwicGF0aCI6Ii8xNDY0OTUzMTEvMzIwMjExMjE5LWZiNTM2MTQyLTViMTQtNDFjZS04YjAyLWY2N2U2MDEzMTc5Zi5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTAyJTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUwMlQyMTM2MzBaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0wZmI0NzM2YTNhMDhiYzJiMGExYzlhMWYyNzViOTczZmU3YzQ0NjA2MjU3Y2Y0MGVmOTQ3M2I2ZjI4N2Q1MDgxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.im2Nb-UoHXLVjks8B8yELRrYoz7qLtuv6HdfLcZK2k4"> <img alt="image" width="170" src="https://private-user-images.githubusercontent.com/146495311/320211269-35e1b4d7-2297-49c8-a85c-34508e3d2245.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTQ2ODYwOTAsIm5iZiI6MTcxNDY4NTc5MCwicGF0aCI6Ii8xNDY0OTUzMTEvMzIwMjExMjY5LTM1ZTFiNGQ3LTIyOTctNDljOC1hODVjLTM0NTA4ZTNkMjI0NS5wbmc_WC1BbX [... cut ...] 16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1jZGU3YzY3N2M0ZWE1ZDU0OWE4YmVjYTVjNWQ0ZDY1MjkxYTQ1ODFjMzdmYjE3MDBlNjQ2YjRiNTVlYjY2YjMyJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.GqmxPbBFrp9poCGzE7L4osRAKtI4X4T2QwwX3O89pCY">

I created a pull request which fixed the issue on my HA 2024.04.4. https://github.com/wills106/homeassistant-solax-modbus/pull/877 

Could you test it by changing the file on your installation? to give @wills106 feedback that the changes are correct?

Krawuzzi, 2024-05-03:
@simowNgithub 
Which values have been corrected by your changes?
Just tested and still had wrong values.
Maybe i did something wrong, will try again tomorrow.

simowNgithub, 2024-05-03:
> @simowNgithub Which values have been corrected by your changes? Just tested and still had wrong values. Maybe i did something wrong, will try again tomorrow.

`Measured_Power`, `House_Load`, `Battery_1_*`
dropped values are: `Battery_*`

wills106, 2024-05-03:
I have just released [2024.05.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2024.05.2) containing these changes.

Krawuzzi, 2024-05-03:
Just tested 2024.05.2, looking very good now.
Thank you very much!

The only thing i found is that house load has a value of 0 W.

Krawuzzi, 2024-05-14:
Not sure if that is what you are asking for:
[image]
[image]

wills106, 2024-05-14:
That's the version I mentioned fixed at an old number. I have the same on my older Gen3. I think it must be for the boot loader.

There will be sensors disabled by default under Diagnostic.
They will look something like:
[image]

Are you able to enable them and provide a screenshot of what they show.

Krawuzzi, 2024-05-14:
[image]

wills106, 2024-05-14:
It looks like you could be on ARM 11.03, but DSP doesn't match 9.6. It would appear to be 4.6?
Will have to wait and see if we get more feedback from others.

@simowNgithub Does your Inverter report "Inverter Temperature" in the hundred degrees region as well like @Krawuzzi ?

I can scale it down if it's the same for both of you on the Gen5.

Could you all let me know what "Inverter Rated Power" returns in the Integration and what your Inverter is rated at.
For example if your Inverter is a SolaX X3-ULT-30K does it return 30,000 or just 30?

simowNgithub, 2024-05-14:
> It looks like you could be on ARM 11.03, but DSP doesn't match 9.6. It would appear to be 4.6? Will have to wait and see if we get more feedback from others.
>
> @simowNgithub Does your Inverter report [...]

Yes, same issue on my side

wills106, 2024-05-14:
> "Inverter Temperature"

I can easily split that off for a Gen5 specific one to scale it to 34.5oC in Krawuzzi case above.

wills106, 2024-05-14:
> You mean: Enable the entity by changing your the custom script and give you the value of this entity? At the moment the entity is deactivated.

If you click on it, there is an option to enable it when you press the cog / gear icon.

Krawuzzi, 2024-05-14:
[image]
i have the Ultra with 25kW

wills106, 2024-05-14:
> ![image](https://private-user-images.githubusercontent.com/146495311/330576555-513da311-7bba-4d34-ab9a-3bd875f3dcc0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF [...]

Thanks, I can start to replicate https://github.com/wills106/homeassistant-solax-modbus/issues/813#issuecomment-2109473775 for the Gen5 (Device Info)

I'll also fix "Inverter Temperature"

Krawuzzi, 2024-06-10:
I think the firmware of my inverter has been updated today.
Now, the "charger upper limit" setting is working as intended, this never worked until today.

[image]
[image]

simowNgithub, 2024-08-21:
@wills106 today I tested a lot and found some unclear values over the solax inverter modbus interface....

The solax cloud tells me, that i have a at night sleep power on the house of 200-300w. I think this is / could be a correct value.

[image]

The modbus interface show me the real power value of 500-1000w in the night. This is not correct I think? 

[image]

What could be the correct value is the sum of the three Inverter Power Sensors, but the "Inverter Power" is always set to 0 W?

[image]

spn91, 2025-01-22:
I'm also an user of the x3-ultra 25k and have the same 0 W reading for the inverter power. Is this the expected behavior, or could we somehow help to fix this?

Krawuzzi, 2025-02-09:
I'm using the X3-ultra 25k too, and have the inverter power showing the correct value.

[image]

But i have another issue:
I upgraded my PV setup and am using all three MPPTs now, but now it seems that "Today's Solar Energy" is incorrect. I think it is calculated by using just the first two MPPTS and ignoring MPPT3.
This results in weird energy graphs in Homeassistant.

[image]

I think i have read about that issue here before but cannot find anything right now.
Does anyone else have the same problem or maybe a fix?

One solution would be to have Homeassistant calculate this value using the Riemann sum (integration) of SolaX Inverter PV Power 1 + 2 + 3.

spn91, 2025-02-10:
There must be something wrong on your side. Using the X3-Ultra-25K with all three MPPTs, the values are correct (at least I think they are correct). 

Comparing with Yield value of the App, `Today's Solar Energy` is 3 kWh off. 62,5 kWh in App and 59,9 kWh in HA. But I think you have to compare it to `Today's Yield`. This value aligns with the one in the App.

Krawuzzi, 2025-02-10:
@spn91 Thanks for the response.
I did more testing today and can definitely say that MPPT3 is not used for "Today's Solar Energy".
Do you also have the problem that sometimes the green bar (battery output) shows at the bottom of the energy dashboard (day view) instead of the top?

Maybe this is also firmware related, I currently have:
Firmware: DSP 1 ARM 13.0

For now, my solution is to let homeassistant calculate the daily energy of all three MPPTs and I added the third one to the energy dashboard as a separate source.

[image]

[image]

[image]

```
sensor:
  # Sensor for Riemann sum of Solax MPPT1 (W -> Wh)
  - platform: integration
    source: sensor.solax_pv_power_1
    name: energy_solax_pv_power_1_sum
    unit_prefix: k
    round: 2
    method: left
  # Sensor for Riemann sum of Solax MPPT2 (W -> Wh)
  - platform: integration
    source: sensor.solax_pv_power_2
    name: energy_solax_pv_power_2_sum
    unit_prefix: k
    round: 2
    method: left
  # Sensor for Riemann sum of Solax MPPT3 (W -> Wh)
  - platform: integration
    source: sensor.solax_pv_power_3
    name: energy_solax_pv_power_3_sum
    unit_prefix: k
    round: 2
    method: left
utility_meter:
  energy_solax_pv_power_1_daily:
    source: sensor.energy_solax_pv_power_1_sum
    name: solax_pv_power_1_daily
    cycle: daily  
  energy_solax_pv_power_2_daily:
    source: sensor.energy_solax_pv_power_2_sum
    name: solax_pv_power_2_daily
    cycle: daily  
  energy_solax_pv_power_3_daily:
    source: sensor.energy_solax_pv_power_3_sum
    name: solax_pv_power_3_daily
    cycle: daily  
```

In case anyone wants to try this code, it is added in configuration.yaml. The fidelity of this approach is dependent on the polling interval of the Solax integration, I have it updating every 2 seconds and the result is good enough for me. (The naming is a bit off, it should use energy instead of power.)

simowNgithub, 2025-02-11:
@Krawuzzi you are right. I have the same problem but didn‘t recognize it because my MPPT3 does have the smallest impact to my total energy consumption. 

**MPPT3 is missing on the sum for todays total solar energy…**

spn91, 2025-02-11:
@Krawuzzi you are using a pretty old firmware. I have never heard of DSP 1 🤔

And I can only repeat again, for my device (firmware was updated a few weeks ago) the third mppt is not missing. As all of my strings have the same slope and azimuth, placed on the same roof, it would be noticeable if one of them is missing in the sum.

spn91, 2025-02-11:
This should be the beginning of the relevant code https://github.com/wills106/homeassistant-solax-modbus/blob/c889aa77137fd181252676c34b1a29a24efde1b4/custom_components/solax_modbus/plugin_solax.py#L4996

as you can see, Today's solar energy is not calculated by the integration, instead it is read from the modbus register. So it seems that the firmware you are using has a bug. This would also explain, why it is perfectly working for me and not working for you.

wills106, 2025-02-11:
I don't own a Gen5 with 3 MPPT's but it does sound like an issue with the Firmware.

Krawuzzi, 2025-02-11:
@spn91 which firmware do you have installed?

I think I'm on DSP 11.1, the DSP major version is not shown in the device-info box yet.

[image]

[image]

spn91, 2025-02-11:
use the cloud interface to get the firmware versions

[image]

Krawuzzi, 2025-02-11:
[image]

I will contact Solax to update my inverter and will report back.

Krawuzzi, 2025-02-12:
With the new Firmware update "Today's Solar Energy" now accounts for all three MPPTS.

[image]

The inverter power stills shows correctly for me:

[image]

TCWORLD, 2025-06-08:
I've just had an X1-IES Hybrid Gen 5 unit installed and getting weird values for "Inverter Power" and then also "House Load".

It seems inverter power is stuck at 0, and so house load is being calculated as some negative value.

[image]

I've had a look at which firmware versions are installed:

[image]

Do you recon this is just because of outdated firmware? I'll contact SolaX anyway and see if there are newer versions available.

TCWORLD, 2025-06-10:
SolaX say they've updated it to the latest versions remotely:

[image]

However the issue persists. Inverter power shows as zero and house load is incorrect.

* The "Inverter Power" sensor always reads zero. 
   * The "Inverter Voltage" and "Inverter Current" are both giving vaguely sensible values.
* The "Inverter Active Power Real" sensor shows a negative value with a magnitude roughly matching what the inverter power should be reading.

* The "Inverter Measured Power" sensor seems to read either the grid export value when positive, or grid import value when negative.

---

Furthermore, while there are two batteries installed, only "Battery 1" reads any values. I presume this is because the two batteries are in the same "tower" so are technically treated as both being "Battery 1" - and presumably "Battery 2" would then be the second unused "tower". The only problem here is presumably because "Battery 2 Capacity" is always zero, the "Battery Total Capacity" always reads as zero. Not the end of the world as I can just use "Battery 1 Capacity" as the sensor to monitor, but its a tad confusing.

TCWORLD, 2025-07-10:
> > ![image](https://private-user-images.githubusercontent.com/146495311/330576555-513da311-7bba-4d34-ab9a-3bd875f3dcc0.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTU3MTcyMjksIm5iZiI6MTcxNTcxNjkyOSwicGF0aCI6Ii8xNDY0OTUzMTEvMzMwNTc2NTU1LTUxM2RhMzExLTdiYmEtNGQzNC1hYjlhLTNiZDg3NWYzZGNjMC5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwNTE0JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDUxNFQyMDAyMDlaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT1kYzRiODg2MDhkODRkNThjMzRlMThkNDQxM2I0YzhhMGY4NGY5ZTRmMjliNzE1YTczNThmNTAxZmJmM2ZlNjkxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.x2W_Y1KInEc2UlPLeEBNRzp9tsQ0wYX8VKN3FYVulUA)
> > i have the Ultra with 25kW
> 
> Thanks, I can start to replicate [#813 (comment)](https://github.com/wills106/homeassistant-solax-modbus/issues/813#issuecomment-2109473775) for the Gen5 (Device Info)
> 
> I'll also fix "Inverter Temperature"

It seems that patch for the inverter temperature that scales it by 1/10th doesn't work for my X1-IES inverter. On mine the value before scaling was already correct.

I'm guessing this is another difference between the X1 and X3 variants of the Gen5 inverters. Looking back the ones that were mis-scaled were X3 variants. I've added a draft PR (#1428) that splits the two inverters so that X3 is scaled and X1 isn't. Would be useful if anyone else with an X1 could confirm whether this is needed, or if it is instead a difference in firmware versions.

Mine before and after the patch:

[image]

wills106, 2025-07-17:
Just released [2025.07.2](https://github.com/wills106/homeassistant-solax-modbus/releases/tag/2025.07.2) with this change.

closed 2025-10-03 by github-actions[bot]